MagicMirrorOrg/MagicMirror #1969: Unable to post pictures on forum topics.

sdetweil, 2020-04-01:
Topic says it all..

We have asked for help in forum for over a month w no response

MichMich, 2020-04-01:
@roramirez Do you have any idea?

sdetweil, 2020-04-05:
@roramirez   @MichMich anything?

MichMich, 2020-04-06:
I tried to find a solution past Friday, but couldn't find a solution. I just got a word from @roramirez. Unfortunately he can't take a look because of the Covid-19 situation. I will dive into it again end of this week.

sdetweil, 2020-04-06:
@MichMich   thanks... we are also seeing more and more postings fail with Askimet.. really need this to go away..

MichMich, 2020-04-06:
Will check that as well. Thanks for your patience.

MichMich, 2020-04-09:
Image uploading is solved. Let's see if the Askimet improves as well. 

Closing this for now, feel free to contact me if there are any issues.

sdetweil, 2020-04-09:
sorry, new problems

closed 2020-06-08 by sdetweil