MagicMirrorOrg/MagicMirror #1970: Issue after update

vxgsiturbo, 2020-04-01:
I have noticed that at the top of the screen it said there was an update available to I updated and now the compliments have gone back to the original and also my calender is no longer working 

any help would be great

thanks

sdetweil, 2020-04-05:
@vxgsiturbo   can u explain
compliments have gone back to the original 

- did u have some modifications ?

my google calenderext is no longer working (are u using calendarExt module? 

- did u npm install in that folder?

vxgsiturbo, 2020-04-05:
I did yes

sdetweil, 2020-04-05:
@vxgsiturbo 

did what ??  

have modifications and/or run npm install??

vxgsiturbo, 2020-04-05:
Original I changed the compliments and had google calendar working but since I did the update none of it is now working and yes I believe I did do the npm install to update it

sdetweil, 2020-04-05:
ok, the compliments.js file belongs to MM, so you should ONLY update thru the config.js

can u show me the config.js entry for the calendar?
again only update the config.js

vxgsiturbo, 2020-04-05:
var DEFAULT_CONFIG = {
	system: {
		useProfileConfig: 0,
		startProfile: "",
		showEmptyView: 1,
		show: ["daily"],
		fullDayEventLocalize: 1,
		redrawInterval: 1*60*1000, //minimum 60000
		locale: null,
	},
	views: {
		daily: {
			direction: "row",
			counts: 5,
			titleFormat: "D",
			overTitleFormat: "MMM D",
			subtitleFormat: "ddd",
		},
		weekly: {
			direction: "row",
			counts: 4,
			titleFormat: "wo",
			overTitleFormat: "gggg wo",
			subtitleFormat: "MMM Do",
		},
		monthly: {
			direction: "row",
			counts: 3,
			titleFormat: "MMMM",
			overTitleFormat: "YYYY MMM",
			subtitleFormat: "YYYY",
		},
		upcoming: {
			title: "Upcoming",
			useRelative: 1
		},
		current: {
			title: "Current",
			useRelative: 1
		},
		month: {
			titleFormat : "D",
			overTitleFormat : "MMM D",
			monthTitleFormat: "MMMM",
			weekdayFormat: "ddd",
			showWeeks: 1,
			weeksTitle: "weeks",
			weeksFormat: "wo",
		},
		weeks: {
			showWeeks: 1,
			weeksTitle: "weeks",
			weeksFormat: "wo",
			weekdayFormat: "ddd",
			titleFormat: "MMM D",
			overTitleFormat: "MMM D",
			counts: 4,
		},
	},
	calendars: [],
	defaultView: {
		position: "bottom_bar",
		positionOrder: -1,
		overflowRolling: 0,
		overflowHeight: 0,
		overflowDuration: 2,
		onlyStartingTime: 0,
		timeFormat: "HH:mm",
		dateFormat: "MMM Do",
		fullDayEventDateFormat: "MMM Do",
		ellipsis: 0,
		limit:0,
		oneLineEvent:0,
		replaceTitle: [],
		classPattern: [],
		classPatternWhere: ["title"],
		symbolPattern: [],
		symbolPatternWhere: ["title"],
	},
	defaultCalendar: {
		name: null,
		profiles: [],
		views: [],
		styleName: "",
		replaceTitle: [],
		classPattern: [],
		classPatternWhere: ["title"],
		symbolPattern: [],
		symbolPatternWhere: ["title"],
		ellipsis: 0,
		symbol: "",
		maxEntries:50,
		maxDays:180,
		interval: 1*5*1000,
		url: null,
		auth: {
			user:"",
			pass:"",
			method:""
		}
	}
}

function Configs(
	systemConfig={},
	viewsConfig={},
	calendarsConfig={},
	profileConfigs={},
	defaultViewConfig={},
	defaultCalendarConfig={}
) {
	this.system = systemConfig
	this.views = viewsConfig
	this.calendars = calendarsConfig
	this.profileConfigs = profileConfigs
	this.defaultView = defaultViewConfig
	this.defaultCalendar = defaultCalendarConfig

	return this;
}

Configs.prototype.mix = function(tmpl, ncfg, includeCalendars=1) {
	var cfg = {}

	//system

	var tcfg = (typeof ncfg.system !== "undefined") ? ncfg.system : {}
	cfg.system = this.assignment({}, tmpl.system, tcfg)
	if(!cfg.system.locale) {
		cfg.system.locale = moment.locale()
	}

	//views
	if(ncfg.views !== "undefined") {
		cfg.views = this.assignment({}, tmpl.views, ncfg.views)
	}

	//defaultView
	if(ncfg.defaultView !== "undefined") {
		cfg.defaultView = this.assignment({}, tmpl.defaultView, ncfg.defaultView)
	}
	//defaultCalendar
	if(ncfg.defaultCalendar !== "undefined") {
		cfg.defaultCalendar = this.assignment(
			{},
			tmpl.defaultCalendar,
			ncfg.defaultCalendar
		)
	}

	//profileConfig
	if (ncfg.profileConfigs !== "undefined") {
		cfg.profileConfigs = this.assignment({}, ncfg.profileConfigs)
	}
	//calendars
	if (!includeCalendars || typeof ncfg.calendars == "undefined") {
		cfg.calendars = this.assignment({}, tmpl.calendars)
	} else {
		cfg.calendars = this.assignment({}, ncfg.calendars)
	}
	return cfg
}

Configs.prototype.getCalConfig = function(idx) {
	return this.assignment(
		{},
		DEFAULT_CONFIG.defaultCalendar,
		this.defaultCalendar,
		this.calendars[idx]
	)
}

Configs.prototype.getViewConfig = function (idx) {
	var ret = this.assignment(
		{},
		DEFAULT_CONFIG.defaultView,
		this.defaultView,
		this.views[idx]
	)
	if (idx == "upcoming" || idx == "current" || idx == "month") {
		ret.counts = 1
	}
	if (typeof ret.locale == "undefined") {
		ret.locale = this.system.locale
	} else {
		if (ret.locale == "") {
			ret.locale = this.system.locale
		}
	}
	ret.mode = idx
	return ret
}

Configs.prototype.make = function(newCfg) {
	var ret = this.mix(DEFAULT_CONFIG, newCfg)
	return new Configs(
		ret.system,
		ret.views,
		ret.calendars,
		ret.profileConfigs,
		ret.defaultView,
		ret.defaultCalendar
	)
}
Configs.prototype.modify = function(newCfg) {
	var ret = this.mix(this, newCfg)
	this.system = ret.system
	this.views = ret.views
	this.calendars = ret.calendars
	this.defaultView = ret.defaultView
	this.defaultCalendar = ret.defaultCalendar
	this.profileConfigs = ret.profileConfigs
	return this;
}
Configs.prototype.assignment = function (result) {
	var stack = Array.prototype.slice.call(arguments, 1);
	var item;
	var key;
	while (stack.length) {
		item = stack.shift();
		for (key in item) {
			if (item.hasOwnProperty(key)) {
				if (
					typeof result[key] === "object"
					&& result[key]
					&& Object.prototype.toString.call(result[key]) !== "[object Array]"
				) {
					if (typeof item[key] === "object" && item[key] !== null) {
						result[key] = this.assignment({}, result[key], item[key]);
					} else {
						result[key] = item[key];
					}
				} else {
					result[key] = item[key];
				}
			}
		}
	}
	return result;
}

sdetweil, 2020-04-05:
where is the calendar url?

vxgsiturbo, 2020-04-05:
in the main config file

vxgsiturbo, 2020-04-05:
sorry I'm very very new to all this so please forgive me

vxgsiturbo, 2020-04-05:
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "", "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1","::ffff:192.168.1.212"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	language: "en",
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
			     // local for armv6l processors, default 
			     //   starts serveronly and then starts chrome browser
			     // false, default for all  NON-armv6l devices
			     // true, force serveronly mode, because you want to.. no UI on this device
	
	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "Calendar",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: ""					}
				]
			}
		},
		
{
      //disabled:true,
      module: 'MMM-CalendarExt',
      position: "top_center",
      config: {
        system: {
          show: ["daily", "upcoming"],
          locale: 'en',
          redrawInterval:60000,
        },
        views: {
          daily: {
            position:'upper_third',
            counts:7,
          },
          upcoming: {
            position:'top_left',
            limit:5
          },
        },
        defaultCalendar: {
          maxEntries:50,
          maxDays:180,
          interval: 1000*60*1,
        },
        calendars :[
          {
            name: "Calendar",
            symbol: "calendar-o",
            url: ""
          },
	  {
		name: "Work",
                styleName:"style5",
		symbol: "calendar-o",
		url: ""
	},
	

        ],
      }
    },
{
			module: "compliments",
			position: "bottom_bar"
		},
{
			module: "compliments 2",
			position: "bottom_bar"
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "Slough",
				locationID: "3333194",  //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "64481ac9ba9a3a11641abc1723af0b9b"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "Slough",
				locationID: "3333194",  //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "64481ac9ba9a3a11641abc1723af0b9b"
			}
		},
		{
			module: "newsfeed",
			position: "",
			config: {
				feeds: [
					{
						title: "",
						url: ""
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
		{
		module: 'MMM-ImageSlideshow',
		position: 'top_center',
		config: {
			imagePaths: ['modules/MMM-ImageSlideshow/exampleImages']
		}
	},

	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

vxgsiturbo, 2020-04-05:
I have removed the url's in the above post for privacy reasons

sdetweil, 2020-04-05:
ok, so I am guessing the same update problem

do
```
cd ~/MagicMirror/modules/MMM-CalendarExt2
rm package-lock.json
rm -rf node_modules
npm install
```

sdetweil, 2020-04-05:
for compliments, u need to add the new ones to config.js per the readme

sdetweil, 2020-04-05:
on mirror, open developers window, crtl-shift-i, select the tab labeled console, and scroll up to find any errors, usually red text

also can put part of module name in filter field 
ext
and the shows just message from that module

may also have to look at the terminal window console.. if pm2 started then

pm2 logs --lines=100

vxgsiturbo, 2020-04-05:
Electron Security Warning (Insecure Resources) This renderer process loads resources using insecure
  protocols.This exposes users of this app to unnecessary security risks.
  Consider loading the following resources over HTTPS or FTPS. 
 - http://localhost:8080/css/main.css
- http://localhost:8080/fonts/roboto.css
- http://localhost:8080/socket.io/socket.io.js
- http://localhost:8080/vendor/node_modules/nunjucks/browser/nunjucks.min.js
- http://localhost:8080/js/defaults.js
- http://localhost:8080/config/config.js
- http://localhost:8080/vendor/vendor.js
- http://localhost:8080/modules/default/defaultmodules.js
- http://localhost:8080/js/logger.js
- http://localhost:8080/translations/translations.js
- http://localhost:8080/js/translator.js
- http://localhost:8080/js/class.js
- http://localhost:8080/js/module.js
- http://localhost:8080/js/loader.js
- http://localhost:8080/js/socketclient.js
- http://localhost:8080/js/main.js
- http://localhost:8080/fonts/node_modules/roboto-fontface/fonts/roboto-condensed/Roboto-Condensed-Regular.woff2
- http://localhost:8080/translations/en.json
- http://localhost:8080/translations/en.json
- http://localhost:8080/modules/default/alert//alert.js
- http://localhost:8080/modules/default/alert/classie.js
- http://localhost:8080/modules/default/alert/modernizr.custom.js
- http://localhost:8080/modules/default/alert/notificationFx.js
- http://localhost:8080/modules/default/alert/ns-default.css
- http://localhost:8080/vendor/css/font-awesome.css
- http://localhost:8080/vendor/node_modules/@fortawesome/fontawesome-free/css/all.min.css
- http://localhost:8080/vendor/node_modules/@fortawesome/fontawesome-free/css/v4-shims.min.css
  
 
For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.
/home/pi/MagicMirror…ity-warnings.js:145 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
    Policy set or a policy with "unsafe-eval" enabled. This exposes users of
    this app to unnecessary security risks.
 
For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.
:8080/modules/defaul…//compliments.js:33 Uncaught SyntaxError: Invalid or unexpected token
:8080/modules/compli…ompliments%202.js:1 Failed to load resource: the server responded with a status of 404 (Not Found)
loader.js:187 Error on loading script: modules/compliments 2//compliments 2.js
localhost/:1 Refused to apply style from 'http://localhost:8080/css/custom.css' because its MIME type ('text/html') is not a supported stylesheet MIME type, and strict MIME checking is enabled.
loader.js:203 Error on loading stylesheet: css/custom.css
main.js:39 Uncaught TypeError: Cannot read property 'appendChild' of undefined
    at main.js:39
    at Array.forEach (<anonymous>)
    at createDomObjects (main.js:23)
    at Object.modulesStarted (main.js:477)
    at startModules (loader.js:59)
    at loader.js:40
    at HTMLLinkElement.stylesheet.onerror (loader.js:204)
api.openweathermap.o…1641abc1723af0b9b:1 Failed to load resource: the server responded with a status of 401 (Unauthorized)
weatherforecast.js:266 weatherforecast: Your AppID does not support long term forecasts. Switching to fallback endpoint.
localhost/:1 Refused to apply style from 'http://localhost:8080/css/custom.css' because its MIME type ('text/html') is not a supported stylesheet MIME type, and strict MIME checking is enabled.
﻿
​

sdetweil, 2020-04-05:
there you go.. 

:8080/modules/defaul…//compliments.js:33 Uncaught SyntaxError: Invalid or unexpected token
no idea

for custom.css do
(not installed by default)
touch ~/magicMirror/css/coustom.css

vxgsiturbo, 2020-04-05:
Ok so sorry what do I need to do ?

sdetweil, 2020-04-05:
figure our what those errors are ...

loader.js:187 Error on loading script: modules/compliments 2//compliments 2.js

this sorta looks like a new install, but old config.js

:8080/modules/defaul…//compliments.js:33 Uncaught SyntaxError: Invalid or unexpected token

in the original compliments.js, this is in the middle of the compliments definition
```
			],
			"....-01-01": [
				"Happy new year!"
			]
		},
```
line Happy new year is line 33

vxgsiturbo, 2020-04-05:
ok so would this be the reason the calendar isn't working too?

sdetweil, 2020-04-05:
i do not know why this is 
```
main.js:39 Uncaught TypeError: Cannot read property 'appendChild' of undefined
at main.js:39
at Array.forEach ()
at createDomObjects (main.js:23)
at Object.modulesStarted (main.js:477)
at startModules (loader.js:59)
at loader.js:40
at HTMLLinkElement.stylesheet.onerror (loader.js:204)
```

vxgsiturbo, 2020-04-05:
ok so have fix the compliments thank you

the calendar still is not working properly, it only shows 1 all day event in the Today part at the top but no list of the 7 days under neath

vxgsiturbo, 2020-04-05:
update: the one all day event is from the default calendar not the 3rd party module so looks like calendarEXT is not working at all

sdetweil, 2020-04-05:
Could be that error I posted above.

vxgsiturbo, 2020-04-05:
sorry io really don't no much about this so don't know how to fix I'm tempted to start all over. any ideas how to completely reinstall magic mirror fresh?

sdetweil, 2020-04-05:
Use the install script from my repo here
https://github.com/sdetweil/MagicMirror_scripts

closed 2020-06-12 by stale[bot]