MagicMirrorOrg/MagicMirror #1971: problem with npm start

Teenshali, 2020-04-02:
[npm.txt](https://github.com/MichMich/MagicMirror/files/4421322/npm.txt)
## I'm not sure if this is a bug
If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

## I found a bug in the MagicMirror installer
If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image
If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror
Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line. 
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file. 

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

sdetweil, 2020-04-02:
not supported on Cygwin/MinGW ..  

u need to run in serverOnly mode (as there is no graphical UI on Mingw)
execute 
node serverOnly
then use a browser to connect to the http://localhost:8080 (default port)
or replace the start line with 
"start":"./run_start.sh"
and get runs-start.sh from my repo (removed in 2.11)
curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/run-start.sh >run-start.sh
chmod+x run-start.sh

then the script will figure it out.. 
(but you still have to access via browser on window side)

Teenshali10, 2020-04-04:
Thank you so much for your response! That really worked for me and now I can run in browser...
But with the browser I am facing one more issue.. can't understand what to do...!
The resource from “http://localhost:8080/vendor/node_modules/nunjucks/browser/nunjucks.min.js” was blocked due to MIME type (“text/html”) mismatch (X-Content-Type-Options: nosniff).
localhost:8080
downloadable font: download failed (font-family: "Roboto Condensed" style:normal weight:400 stretch:100 src index:2): status=2147746065 source: http://localhost:8080/fonts/node_modules/roboto-fontface/fonts/roboto-condensed/Roboto-Condensed-Regular.woff2
The resource from “http://localhost:8080/vendor/node_modules/nunjucks/browser/nunjucks.min.js” was blocked due to MIME type (“text/html”) mismatch (X-Content-Type-Options: nosniff).
localhost:8080
Loading failed for the <script> with source “http://localhost:8080/vendor/node_modules/nunjucks/browser/nunjucks.min.js”. localhost:8080:41:1

sdetweil, 2020-04-04:
@Teenshali10 

Usually this error
>blocked due to MIME type (“text/html”) mismatch (X-Content-Type-Options: nosniff).
is because the file was not found. 
a .js file should be script, not text

so, that means the install didn't complete correctly..

you did an npm install in the MagicMirror folder, right? 
it should have run a step to 

cd to vendor and then npm install there

sdetweil, 2020-04-05:
so, I think this is a newly discovered roboto-fontface install problem..

do this

edit vendor/package.json 
and change the version of  roboto-fontface like this 
from
```
   "dependencies": {
    "roboto-fontface": "^0.8.0"
  }
```
to
```
  "dependencies": {
    "roboto-fontface": "latest"
  }
```
them in the vendor folder do npm install

then try to restart mm..  

please advise your results

rhortal, 2020-04-07:
I have serverOnly: true in my config, since updating to the latest version it seems npm start ignores that setting and tries to open a UI. Can start with node serveronly.

MichMich, 2020-04-07:
Try `node serveronly`

rhortal, 2020-04-07:
I tried that and it works. However node keeps the program in the foreground so had to start using pm2 as a workaround. Has the config option serverOnly been deprecated?

sdetweil, 2020-04-08:
@rhortal  I guess deprecated is a term you could use..   all the scripts were removed from the base repo in 2.11

u can add the capability back, by 

changing package.json start script back to unstart
change to
```
"start":"./run-start.sh $1"
```
and if u need to run-start again
```
curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/run-start.sh >~/MagicMirror/run-start.sh
chmod +x ~/MagicMirror/run-start.sh
```

closed 2020-06-15 by stale[bot]