MagicMirrorOrg/MagicMirror #1972: Error in npm install while installing vendor files please help

PoojaAmin29, 2020-04-03:
pi@raspberrypi:~/MagicMirror $ npm install
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm WARN deprecated @types/vfile-message@2.0.0: This is a stub types definition. vfile-message provides its own type definitions, so you do not need this installed.

> electron-chromedriver@6.0.0 install /home/pi/MagicMirror/node_modules/electron-chromedriver
> node ./download-chromedriver.js

> core-js@2.6.11 postinstall /home/pi/MagicMirror/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
> https://opencollective.com/core-js 
> https://www.patreon.com/zloirock 

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> console-stamp@0.2.9 postinstall /home/pi/MagicMirror/node_modules/console-stamp
> node ./msg.js

Attention
    Console-stamp version 3.0.0 Release Candidate is out. Install by using the tag '@next'. NB: Breaking changes.
    For more details goto: https://www.npmjs.com/package/console-stamp/v/next

> electron@6.1.7 postinstall /home/pi/MagicMirror/node_modules/electron
> node install.js

> magicmirror@2.11.0 install /home/pi/MagicMirror
> echo "Installing vendor files ...
" && cd vendor && npm install --loglevel=error

Installing vendor files ...

npm ERR! Unexpected end of JSON input while parsing near '...5vp6b8jEE6P9iUqepRgrK'

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-03T14_15_02_623Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.11.0 install: `echo "Installing vendor files ...
npm ERR! " && cd vendor && npm install --loglevel=error`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.11.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-03T14_15_02_963Z-debug.log

please help i am new to this

WynandL, 2020-04-04:
I am getting the same errors on RPI3 - used manual and automatic install, but getting exactly the same set of npm ERR! messages...

sdetweil, 2020-04-05:
are you upgrading or installing?

WynandL, 2020-04-05:
Hi, I was installing...used the app a while ago but this was on a fresh
install...

sdetweil, 2020-04-05:
ok.. it would be interesting if my automated install script handles this..
see https://github.com/sdetweil/MagicMirror_scripts
have to rename MM folder out of the way 1st..

i just fixed a problem in vendor npm install for the roboto-fontface install being downlevel. 

I think the problem was caused by the package-lock.json file, which should be erased for new installs.

PoojaAmin29, 2020-04-08:
Rename MM folder to ?

sdetweil, 2020-04-08:
Anything not 'MagicMirror'

PoojaAmin29, 2020-04-08:
Actually somehow i found previous version of magic mirror 2.10.1 and i tried to do npm install on my laptop and it shows me this

E:\MagicMirror-master>npm install
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically un
maintained. Move on to something better. This package will continue to work with
 Grunt v1, but it will not receive any updates.
npm WARN deprecated @types/vfile-message@2.0.0: This is a stub types definition.
 vfile-message provides its own type definitions, so you do not need this instal
led.

> electron-chromedriver@1.8.0 install E:\MagicMirror-master\node_modules\electro
n-chromedriver
> node ./download-chromedriver.js

successfully dowloaded and extracted!

> core-js@2.6.11 postinstall E:\MagicMirror-master\node_modules\core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfill
ing JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Colle
ctive or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a goo
d job -)

> console-stamp@0.2.9 postinstall E:\MagicMirror-master\node_modules\console-sta
mp
> node ./msg.js

Attention
    Console-stamp version 3.0.0 Release Candidate is out. Install by using the t
ag '@next'. NB: Breaking changes.
    For more details goto: https://www.npmjs.com/package/console-stamp/v/next

> electron@3.1.13 postinstall E:\MagicMirror-master\node_modules\electron
> node install.js

Downloading tmp-3328-1-SHASUMS256.txt-3.1.13
[============================================>] 100.0% of 4.79 kB (3.83 kB/s)
E:\MagicMirror-master\node_modules\electron\install.js:49
  throw err
  ^

ChecksumMismatchError: Generated checksum for "electron-v3.1.13-win32-x64.zip" d
id not match expected checksum.
    at Hash.<anonymous> (E:\MagicMirror-master\node_modules\←[4msumchecker←[24m\
index.js:151:20)
←[90m    at Hash.emit (events.js:311:20)←[39m
←[90m    at emitReadable_ (_stream_readable.js:562:12)←[39m
←[90m    at onEofChunk (_stream_readable.js:540:5)←[39m
←[90m    at readableAddChunk (_stream_readable.js:242:5)←[39m
←[90m    at Hash.Readable.push (_stream_readable.js:209:10)←[39m
←[90m    at Hash.Transform.push (_stream_transform.js:152:32)←[39m
←[90m    at done (_stream_transform.js:220:17)←[39m
←[90m    at _stream_transform.js:143:7←[39m
←[90m    at Hash._flush (internal/crypto/hash.js:73:3)←[39m {
  filename: ←[32m'electron-v3.1.13-win32-x64.zip'←[39m,
  message: ←[32m'Generated checksum for "electron-v3.1.13-win32-x64.zip" did not
 match expected checksum.'←[39m
}

> magicmirror@2.10.1 install E:\MagicMirror-master
> cd vendor && npm install

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules\fse
vents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@
1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"}
)

added 82 packages from 57 contributors and audited 220 packages in 6.789s
found 8 vulnerabilities (7 low, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details

> magicmirror@2.10.1 postinstall E:\MagicMirror-master
> sh untrack-css.sh && sh installers/postinstall/postinstall.sh && npm run insta
ll-fonts

'sh' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.10.1 postinstall: `sh untrack-css.sh && sh installers/pos
tinstall/postinstall.sh && npm run install-fonts`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.10.1 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional log
ging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Pooja Amin\AppData\Roaming\npm-cache\_logs\2020-04-08T04_3
6_45_347Z-debug.log

PoojaAmin29, 2020-04-08:
Please help

sdetweil, 2020-04-08:
you are installing on windows.. not very tested...

first, always do npm install --only=prod
this will stop installing the code used for the testing cycle
second sh is not a command, true..    you will have to examine the .sh files and run those commands manually

last  ChecksumMismatchError: Generated checksum for "electron-v3.1.13-win32-x64.zip" d
id not match expected checksum.

don't know... that sounds like network error, or disk error

PoojaAmin29, 2020-04-08:
I tried to install Magic Mirror v2.10.1 on my raspberry pi 3 and npm install successfully installed magic mirror but when i did npm start it gives this error
pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.10.1 start /home/pi/MagicMirror
> ./run-start.sh

sh: 1: ./run-start.sh: Permission denied
npm ERR! code ELIFECYCLE
npm ERR! errno 126
npm ERR! magicmirror@2.10.1 start: `./run-start.sh`
npm ERR! Exit status 126
npm ERR! 
npm ERR! Failed at the magicmirror@2.10.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-08T12_59_27_220Z-debug.log

Please help

PoojaAmin29, 2020-04-08:
> you are installing on windows.. not very tested...
>
> first, always do npm install --only=prod
> this will stop installing the code used for the testing cycle
> second sh is not a command, true.. you will [...]

Thanks sir for your reply

sdetweil, 2020-04-08:
yes, you also have to change the  package.json start command 

from 
```
"start":"./run-start.sh",
```
to
```
"start":"npm start",
```

the MM is assumed to be running on some linux system.

PoojaAmin29, 2020-04-08:
I am running MM on rpi 3

sdetweil, 2020-04-08:
>E:\MagicMirror-master>npm install
>electron-v3.1.13-win32-x64.zip

messages say different

PoojaAmin29, 2020-04-08:
its the previous one which i was trying on windows

PoojaAmin29, 2020-04-08:
I did the changes in package.json file but it still gives error

pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.10.1 start /home/pi/MagicMirror
> node-modules/.bin/electron %1

sh: 1: node-modules/.bin/electron: not found
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! magicmirror@2.10.1 start: `node-modules/.bin/electron %1`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the magicmirror@2.10.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-08T13_21_52_731Z-debug.log

sdetweil, 2020-04-08:
sorryy, I missed that change

please tun my install script.. 

in electron is not installed, you have trouble

rename MM folder out of the way, then run

bash -c "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/raspberry.sh)"

PoojaAmin29, 2020-04-08:
i did 
bash -c "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/raspberry.sh)"
i wanted to ask does it take time as it as stuck on a line for around 5 minutes, sorry for this silly question

sdetweil, 2020-04-08:
it can take some time, as it is upgrading the system, as we need components that have been updated.

PoojaAmin29, 2020-04-08:
ok thanks sir once again for your reply i will notify once done

PoojaAmin29, 2020-04-08:
pi@raspberrypi:~ $ bash -c "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/raspberry.sh)"

$$\      $$\                     $$\           $$\      $$\ $$\                                          $$$$$$\
$$$\    $$$ |                    \__|          $$$\    $$$ |\__|                                        $$  __$$\
$$$$\  $$$$ | $$$$$$\   $$$$$$\  $$\  $$$$$$$\ $$$$\  $$$$ |$$\  $$$$$$\   $$$$$$\   $$$$$$\   $$$$$$\  \__/  $$ |
$$\$$\$$ $$ | \____$$\ $$  __$$\ $$ |$$  _____|$$\$$\$$ $$ |$$ |$$  __$$\ $$  __$$\ $$  __$$\ $$  __$$\  $$$$$$  |
$$ \$$$  $$ | $$$$$$$ |$$ /  $$ |$$ |$$ /      $$ \$$$  $$ |$$ |$$ |  \__|$$ |  \__|$$ /  $$ |$$ |  \__|$$  ____/
$$ |\$  /$$ |$$  __$$ |$$ |  $$ |$$ |$$ |      $$ |\$  /$$ |$$ |$$ |      $$ |      $$ |  $$ |$$ |      $$ |
$$ | \_/ $$ |\$$$$$$$ |\$$$$$$$ |$$ |\$$$$$$$\ $$ | \_/ $$ |$$ |$$ |      $$ |      \$$$$$$  |$$ |      $$$$$$$$\
\__|     \__| \_______| \____$$ |\__| \_______|\__|     \__|\__|\__|      \__|       \______/ \__|      \________|
                       $$\   $$ |
                       \$$$$$$  |
                        \______/

install log being saved to /home/pi/install.log
Updating packages ...
Installing helper tools ...
Check current Node installation ...
Node currently installed. Checking version number.
Minimum Node version: v10.1.0
Installed Node version: v10.19.0
No Node.js upgrade necessary.
Check current NPM installation ...
NPM currently installed. Checking version number.
Minimum npm version: V6.0.0
Installed npm version: V6.13.4
No npm upgrade necessary.
Cloning MagicMirror ...
Cloning into 'MagicMirror'...
remote: Enumerating objects: 316, done.
remote: Counting objects: 100% (316/316), done.
remote: Compressing objects: 100% (276/276), done.
remote: Total 316 (delta 43), reused 179 (delta 27), pack-reused 0
Receiving objects: 100% (316/316), 437.96 KiB | 368.00 KiB/s, done.
Resolving deltas: 100% (43/43), done.
Cloning MagicMirror Done!
Installing dependencies ...
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated rrule-alt@2.2.8: Please use rrule instead
npm WARN deprecated left-pad@1.3.0: use String.prototype.padStart()
npm WARN deprecated mkdirp@0.5.3: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated webdriverio@4.14.4: outdated version, please use @next
npm WARN deprecated mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm WARN deprecated @types/vfile-message@2.0.0: This is a stub types definition. vfile-message provides its own type definitions, so you do not need this installed.

This is going on correctly just confirming

sdetweil, 2020-04-08:
yes

PoojaAmin29, 2020-04-08:
ok sir

PoojaAmin29, 2020-04-08:
> pi@raspberrypi:~ $ bash -c "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/raspberry.sh)"
> 
> $$\ $$\ $$\ $$\ $$\ $$\ $$$$$$
> $$$\ $$$ | __| $$$\ $$$ |__| $$ **$$ $$$$\ $$$$ | $$$$$$\ $$$$$$\ $$\ $$$$$$$\ $$$$\ $$$$ |$$\ $$$$$$\ $$$$$$\ $$$$$$\ $$$$$$\ __/ $$ | $$$$$$ $$ | __**$$\ $$ **$$\ $$ |$$ **_**|$$$$$$ $$ |$$ |$$ **$$\ $$ **$$\ $$ __$$\ $$ **$$\ $$$$$$ | $$ $$$ $$ | $$$$$$$ |$$ / $$ |$$ |$$ / $$ $$$ $$ |$$ |$$ | __|$$ | __|$$ / $$ |$$ | __|$$ **__/ $$ |$ /$$ |$$ __$$ |$$ | $$ |$$ |$$ | $$ |$ /$$ |$$ |$$ | $$ | $$ | $$ |$$ | $$ | $$ | _/ $$ |$$$$$$$ |$$$$$$$ |$$ |$$$$$$$\ $$ | _/ $$ |$$ |$$ | $$ | $$$$$$ |$$ | $$$$$$$$ __| __| _**_**_| __**$$ |__| __****_|__| __|__|__| __| _****_/ __| ________| $$\ $$ | $$$$$$ | ______/
> 
> install log being saved to /home/pi/install.log
> Updating packages ...
> Installing helper tools ...
> Check current Node installation ...
> Node currently installed. Checking version number.
> Minimum Node version: v10.1.0
> Installed Node version: v10.19.0
> No Node.js upgrade necessary.
> Check current NPM installation ...
> NPM currently installed. Checking version number.
> Minimum npm version: V6.0.0
> Installed npm version: V6.13.4
> No npm upgrade necessary.
> Cloning MagicMirror ...
> Cloning into 'MagicMirror'...
> remote: Enumerating objects: 316, done.
> remote: Counting objects: 100% (316/316), done.
> remote: Compressing objects: 100% (276/276), done.
> remote: Total 316 (delta 43), reused 179 (delta 27), pack-reused 0
> Receiving objects: 100% (316/316), 437.96 KiB | 368.00 KiB/s, done.
> Resolving deltas: 100% (43/43), done.
> Cloning MagicMirror Done!
> Installing dependencies ...
> npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
> npm WARN deprecated request@2.88.2: request has been deprecated, see [request/request#3142](https://github.com/request/request/issues/3142)
> npm WARN deprecated rrule-alt@2.2.8: Please use rrule instead
> npm WARN deprecated left-pad@1.3.0: use String.prototype.padStart()
> npm WARN deprecated mkdirp@0.5.3: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
> npm WARN deprecated webdriverio@4.14.4: outdated version, please use @next
> npm WARN deprecated mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
> npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
> npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
> npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
> npm WARN deprecated @types/vfile-message@2.0.0: This is a stub types definition. vfile-message provides its own type definitions, so you do not need this installed.
> 
> This is going on correctly just confirming

Sir its still on that same line till now is there any problem? how much time does it take to install dependencies

sdetweil, 2020-04-08:
how long?  no idea, as long as it takes..   sometimes long, sometimes not...

let it finish

PoojaAmin29, 2020-04-08:
oh ok

PoojaAmin29, 2020-04-08:
After installation it automatically starts magic mirror ?

sdetweil, 2020-04-08:
If u use pm2, yes

PoojaAmin29, 2020-04-08:
sorry sir again a silly question how will i know installation is completed

PoojaAmin29, 2020-04-08:
As of now still it is stuck and there is no new line thus was a little worried

sdetweil, 2020-04-08:
@PoojaAmin29   any update?

PoojaAmin29, 2020-04-09:
It was still stuck over that line for more than 5 hrs so I shutdown the
system now I am going to start again

PoojaAmin29, 2020-04-09:
But how do I get to know that installation is completed?

PoojaAmin29, 2020-04-09:
sir its asking me 
Splashscreen: Changed theme to MagicMirror successfully.
Do you want use pm2 for auto starting of your MagicMirror (y/N)? 
if yes then whenever i need to start magic mirror instead of npm start i need to type pm2?
and if no then what will it be

PoojaAmin29, 2020-04-09:
pi@raspberrypi:~ $ bash -c "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/raspberry.sh)"

$$\      $$\                     $$\           $$\      $$\ $$\                                          $$$$$$\
$$$\    $$$ |                    \__|          $$$\    $$$ |\__|                                        $$  __$$\
$$$$\  $$$$ | $$$$$$\   $$$$$$\  $$\  $$$$$$$\ $$$$\  $$$$ |$$\  $$$$$$\   $$$$$$\   $$$$$$\   $$$$$$\  \__/  $$ |
$$\$$\$$ $$ | \____$$\ $$  __$$\ $$ |$$  _____|$$\$$\$$ $$ |$$ |$$  __$$\ $$  __$$\ $$  __$$\ $$  __$$\  $$$$$$  |
$$ \$$$  $$ | $$$$$$$ |$$ /  $$ |$$ |$$ /      $$ \$$$  $$ |$$ |$$ |  \__|$$ |  \__|$$ /  $$ |$$ |  \__|$$  ____/
$$ |\$  /$$ |$$  __$$ |$$ |  $$ |$$ |$$ |      $$ |\$  /$$ |$$ |$$ |      $$ |      $$ |  $$ |$$ |      $$ |
$$ | \_/ $$ |\$$$$$$$ |\$$$$$$$ |$$ |\$$$$$$$\ $$ | \_/ $$ |$$ |$$ |      $$ |      \$$$$$$  |$$ |      $$$$$$$$\
\__|     \__| \_______| \____$$ |\__| \_______|\__|     \__|\__|\__|      \__|       \______/ \__|      \________|
                       $$\   $$ |
                       \$$$$$$  |
                        \______/

install log being saved to /home/pi/install.log
Updating packages ...
Installing helper tools ...
Check current Node installation ...
Node currently installed. Checking version number.
Minimum Node version: v10.1.0
Installed Node version: v10.20.0
No Node.js upgrade necessary.
Check current NPM installation ...
NPM currently installed. Checking version number.
Minimum npm version: V6.0.0
Installed npm version: V6.14.4
No npm upgrade necessary.
Cloning MagicMirror ...
Cloning into 'MagicMirror'...
remote: Enumerating objects: 316, done.
remote: Counting objects: 100% (316/316), done.
remote: Compressing objects: 100% (276/276), done.
remote: Total 316 (delta 43), reused 179 (delta 27), pack-reused 0
Receiving objects: 100% (316/316), 437.96 KiB | 257.00 KiB/s, done.
Resolving deltas: 100% (43/43), done.
Cloning MagicMirror Done!
Installing dependencies ...
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated rrule-alt@2.2.8: Please use rrule instead
npm WARN deprecated left-pad@1.3.0: use String.prototype.padStart()
npm WARN deprecated mkdirp@0.5.3: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated webdriverio@4.14.4: outdated version, please use @next
npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
npm WARN deprecated mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm WARN deprecated @types/vfile-message@2.0.0: This is a stub types definition. vfile-message provides its own type definitions, so you do not need this installed.
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.1.1 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

Dependencies installation Done!
setting up initial config.js
Check plymouth installation ...
Splashscreen: Checking themes directory.
Splashscreen: Create theme directory if not exists.

Splashscreen: Changed theme to MagicMirror successfully.
Do you want use pm2 for auto starting of your MagicMirror (y/N)? 

Is it correct?

PoojaAmin29, 2020-04-09:
thanks sir it works now 
I did this 
Do you want use pm2 for auto starting of your MagicMirror (y/N)?n
Do you want to disable the screen saver? (y/N)?y
 
We're ready! Run DISPLAY=:0 npm start from the ~/MagicMirror directory to start your MagicMirror.
 
 
pi@raspberrypi:~ $ cd MagicMirror/
pi@raspberrypi:~/MagicMirror $ DISPLAY=:0 npm start
> magicmirror@2.11.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-04-09 08:08:02.828] [LOG]    Starting MagicMirror: v2.11.0
[2020-04-09 08:08:02.842] [LOG]    Loading config ...
[2020-04-09 08:08:02.855] [LOG]    Loading module helpers ...
[2020-04-09 08:08:02.859] [LOG]    No helper found for module: alert.
[2020-04-09 08:08:02.984] [LOG]    Initializing new module helper ...
[2020-04-09 08:08:02.985] [LOG]    Module helper loaded: updatenotification
[2020-04-09 08:08:02.987] [LOG]    No helper found for module: clock.
[2020-04-09 08:08:06.085] [LOG]    Initializing new module helper ...
[2020-04-09 08:08:06.087] [LOG]    Module helper loaded: calendar
[2020-04-09 08:08:06.092] [LOG]    No helper found for module: compliments.
[2020-04-09 08:08:06.096] [LOG]    No helper found for module: currentweather.
[2020-04-09 08:08:06.106] [LOG]    No helper found for module: weatherforecast.
[2020-04-09 08:08:06.175] [LOG]    Initializing new module helper ...
[2020-04-09 08:08:06.176] [LOG]    Module helper loaded: newsfeed
[2020-04-09 08:08:06.177] [LOG]    All module helpers loaded.
[2020-04-09 08:08:06.477] [LOG]    Starting server on port 8080 ... 
[2020-04-09 08:08:06.514] [LOG]    Server started ...
[2020-04-09 08:08:06.515] [LOG]    Connecting socket for: updatenotification
[2020-04-09 08:08:06.517] [LOG]    Connecting socket for: calendar
[2020-04-09 08:08:06.517] [LOG]    Starting node helper for: calendar
[2020-04-09 08:08:06.518] [LOG]    Connecting socket for: newsfeed
[2020-04-09 08:08:06.519] [LOG]    Starting module: newsfeed
[2020-04-09 08:08:06.520] [LOG]    Sockets connected & modules started ...
[2020-04-09 08:08:07.073] [LOG]    Launching application.
[2020-04-09 08:08:11.552] [LOG]    Create new calendar fetcher for url: http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics - Interval: 300000
[2020-04-09 08:08:11.624] [LOG]    Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
[2020-04-09 08:09:00.073] [LOG]    Shutting down server...
[2020-04-09 08:09:00.076] [LOG]    Stopping module helper: updatenotification
[2020-04-09 08:09:00.077] [LOG]    Stopping module helper: calendar
[2020-04-09 08:09:00.078] [LOG]    Stopping module helper: newsfeed

WynandL, 2020-04-09:
I guys, I have been following the conversation. I also renamed the MagicMirror folder and ran (on RPI3)

bash -c "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/raspberry.sh)"

but I still get this (after which the folder is recreated and I navigate there, command in bold):

**pi@raspberrypi:~/MagicMirror $ DISPLAY=:0 npm start**                                                                             
> magicmirror@2.11.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

/home/pi/MagicMirror/node_modules/electron/dist/electron: error while loading shared libraries: libgdk_pixbuf-2.0.so.0: cannot open shared object file: No such file or directory
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! magicmirror@2.11.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the magicmirror@2.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-09T06_06_20_415Z-debug.log

sdetweil, 2020-04-09:
@WynandL   is this a pi 0 or pi 2 (armv6l)?

electron doesn't exist for that device anymore..

sadly they took lout my startup script that handled this

do 

change package.json
from
```
 "start": "DISPLAY=\"${DISPLAY:=:0}\" ./node_modules/.bin/electron js/electron.js",
```
to 
```
 "start":"./run-start.sh",
```
and then 
```
curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/run-start.sh >~/MagicMirror/run-start.sh
chmod +x ~/MagicMirror/run-start.sh
```

sdetweil, 2020-04-09:
@PoojaAmin29 

pm2 will allows you to startr your MM at boot time (without editing any system files) and also allow you to 
start/stop/restart... etc   from the command shell anytime

if u want to add that now, from my repo of the scripts removed from the base
https://github.com/sdetweil/MagicMirror_scripts
do, 
```
bash -c "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/fixuppm2.sh)"
```

PoojaAmin29, 2020-04-09:
> @PoojaAmin29
>
> pm2 will allows you to startr your MM at boot time (without editing any system files) and also allow you to
> start/stop/restart... etc from the command shell anytime
>
> if u want to add [...]

thanks sir

PoojaAmin29, 2020-04-10:
@sdetweil i had one more issue when i start MM with npm start it first shows a white screen and then the magic mirror appears

sdetweil, 2020-04-10:
yes, there is a small time when the white screen shows thru, before the black is applied.. this is an electron bug..for which there is no fix

PoojaAmin29, 2020-04-10:
Ok thanks

WynandL, 2020-04-10:
> @WynandL looks like a missing library
> 
> apt-get install libgtk2.0-0
> 
> the system requires a graphical desktop version of the OS.. not commandline (cli)

Okay - will have to reinstall Raspbian with GUI - will take some time but will get back to you (will first install that library but I assume I'll have extra problems...so reinstall will be after that.

developer0777, 2020-04-16:
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm ERR! Unexpected end of JSON input while parsing near '...P SIGNATURE-----\r\n"' 

I'm facing this issue can anyone tell me what to do

sdetweil, 2020-04-16:
developer0777

how are u doing the install? 
what level of node and npm are you running
node -v
npm -v

otherwide, how old is the SD card you are using? I have been seeing users report quite a few failures lately

pawarnilesh90, 2020-04-17:
I also have the same errors while npm i
npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please,
upgrade your dependencies to the actual version of core-js@3.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated

node version: 10.16.3
npm version: 6.14.4
@angular/cli: 8.3.14
@angular/core: 8.2.12

Please help.

sdetweil, 2020-04-17:
@pawarnilesh90  those are warnings.. nothing u can do about them. some developer will have to change his code.   everything still works

PoojaAmin29, 2020-04-19:
I am facing issue actually due some issue i had to again start from scratch and install raspbian but when i install MM from your repository as earlier it is giving me this error please help 

pi@raspberrypi:~ $ DISPLAY=:0 npm start
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /home/pi/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent 

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-19T10_54_29_287Z-debug.log

sdetweil, 2020-04-19:
~PoojaAmin29

pi@raspberrypi:~ $ DISPLAY=:0 npm start

you must be in the MagicMirror folder

cd ~/MagiMirror
npm start

PoojaAmin29, 2020-04-19:
sorry my mistake, i forgot to do cd MagicMirror now its working, again thanks

sdetweil, 2020-04-19:
one reason to add pm2 , it doesn't care where u are, restarts on boot.. etc

use the fixuppm2 script from my repo.. to add it now. (don't HAVE to use it)

dcthienit1997, 2020-04-25:
OMG, my error
`> magicmirror@2.11.0 start C:\Users\thuho\MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

'DISPLAY' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.11.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.11.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\thuho\AppData\Roaming\npm-cache\_logs\2020-04-25T09_33_56_419Z-debug.log
`

How to fix them ?! I run on windows 10

sdetweil, 2020-04-25:
@dcthienit1997  technically not supported on Windows but

edit the MagicMirror/package.json file and change the line 
```
    "start": "DISPLAY=\"${DISPLAY:=:0}\" ./node_modules/.bin/electron js/electron.js",
```
to 
```
    "start": "node_modules/.bin/electron js/electron.js",
```
also, before doing npm start, u have to run the last two steps of install, as they are written for linux
```
cd MagcMirror/vendor
npm install
cd ..\fonts
npm install
```
then npm start in the MagicMirror folder should work.

dcthienit1997, 2020-04-25:
> @dcthienit1997 technically not supported on Windows but
>
> edit the MagicMirror/package.json file and change the line
>
> ```
>     "start": "DISPLAY=\"${DISPLAY:=:0}\" ./node_modules/.bin/electron [...]

OMG, thank u so much
It worked. But the screen was dark, nothing was displayed. Is it an error?

dcthienit1997, 2020-04-25:
Oh, I deleted all and re-started from the beginning carefully. Finally it works and displays very well. Thank you

sarahselby, 2020-04-28:
Hello, I'm also experiencing problems! 

`0 info it worked if it ends with ok

1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]

2 info using npm@6.14.4

3 info using node@v13.13.0

4 verbose run-script [ 'prestart', 'start', 'poststart' ]

5 info lifecycle magicmirror@2.11.0~prestart: magicmirror@2.11.0

6 info lifecycle magicmirror@2.11.0~start: magicmirror@2.11.0

7 verbose lifecycle magicmirror@2.11.0~start: unsafe-perm in lifecycle true

8 verbose lifecycle magicmirror@2.11.0~start: PATH: 
/usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games

9 verbose lifecycle magicmirror@2.11.0~start: CWD: /home/pi/MagicMirror

10 silly lifecycle magicmirror@2.11.0~start: Args: [

10 silly lifecycle   '-c',

10 silly lifecycle   'DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js'

10 silly lifecycle ]

11 info lifecycle magicmirror@2.11.0~start: Failed to exec start script

12 verbose stack Error: magicmirror@2.11.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`

12 verbose stack spawn ENOENT

12 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)

12 verbose stack     at ChildProcess.emit (events.js:315:20)

12 verbose stack     at maybeClose (internal/child_process.js:1026:16)

12 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)

13 verbose pkgid magicmirror@2.11.0

14 verbose cwd /home/pi/MagicMirror

15 verbose Linux 4.19.97-v7+

16 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"

17 verbose node v13.13.0

18 verbose npm  v6.14.4

19 error code ELIFECYCLE

20 error syscall spawn

21 error file sh

22 error errno ENOENT

23 error magicmirror@2.11.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`

23 error spawn ENOENT

24 error Failed at the magicmirror@2.11.0 start script.

24 error This is probably not a problem with npm. There is likely additional logging output above.

25 verbose exit [ 1, true ]
`

Any help appreciated! Thank you

sdetweil, 2020-04-28:
@sarahselby  is this on windows?  

see the instructions here 
https://github.com/MichMich/MagicMirror/issues/1972#issuecomment-619405930

sarahselby, 2020-04-28:
@sdetweil No sorry it's Raspberry Pi: Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux

sdetweil, 2020-04-28:
@sarahselby 

what device is this?  pi0 or pi 2> 
show results of
uname -m

[9 more replies]

closed 2020-07-04 by MichMich