MagicMirrorOrg/MagicMirror #1973: changes in socketclient.js stops MMM-RemoteControl to work

khassel, 2020-04-03:
The changes in `socketclient.js` introduced with this PR https://github.com/MichMich/MagicMirror/pull/1937 in v2.11.0 breaks MMM-RemoteControl, maybe other modules too.

See https://github.com/Jopyth/MMM-Remote-Control/issues/185

After reverting the changes in `socketclient.js` the module works again.

khassel, 2020-04-04:
Found 2 issues:

-  we can't use `window.location.pathname` because modules can use own html sites referencing `socketclient.js`. E.g. in `http://192.168.xxx.yyy:8080/remote.html` `window.location.pathname` has the value `/remote.html` instead of expected `/`. This could may solved defining a `subPath` value in the config and using this value instead of `window.location.pathname`.

-  in `index.html` the line
  `<script type="text/javascript" src="/socket.io/socket.io.js"></script>`
  was replaced in the above PR with
  `<script type="text/javascript" src="socket.io/socket.io.js"></script>`
  which is a good correction, but many module developers copied this line in their html site. 
  This is a minor issue: If someone will use such a module with a subpath <> `/`, the above line must corrected in the module html.

Legion2, 2020-04-05:
This can be resolved by defining a `basename` in the configuration with a default value of `/`. `basename` is the used instead of `window.location.pathname`.

Legion2, 2020-04-05:
@khassel I created a PR to fix the broken socket.io path #1976

MichMich, 2020-04-09:
Merged.

PR #1976 (2020-04-06): Added basename config
use basename in socket.io path fix #1973

The use of `window.location.pathname` in the socket.io path (introduced in #1937) broke some modules, because the url was not always the basename of MagicMirror. To resolve this the basename is now a configuration option which can be set in the configuration. The default value is `/` which results in the behavior as before #1937 was added.
commit 54542f7f (2020-04-06): added basename config
use basename in socket.io path fix #1973
PR #1980 (2020-04-09): socketclient.js: add backward compatibility for old module code
see https://github.com/MichMich/MagicMirror/pull/1976#issuecomment-610037700

closes #1973
closed 2020-04-09 by MichMich