MagicMirrorOrg/MagicMirror #1975: Black Screen After Update

SkyGuy385, 2020-04-05:
**Platform**: Raspberry Pi 3

**Node Version**: armv7l

**MagicMirror Version**: 2.11.0

**Description**: After updating to current MagicMirror version. Only a black screen is shown when booting up MagicMirror. No modules. My cursor appears while the MagicMirror window is displayed. 

**Steps to Reproduce**: Have the following modules installed:
- MMM-MoonPhase   
- MMM-Screencast
- MagicMirror-QuoteCatalog  
- MMM-MPD         
- MMM-WeatherDependentClothes
- MMM-GoogleAssistant
- MMM-PIR-Sensor
- The modules already included in MM2

1. Opened command prompt
2. Went to MagicMirror directory
3. git reset --hard; git clean -df;
4. git pull && npm install
5. Restarted Pi
6. Booted MagicMirror to find black screen

**Expected Results**: Expected to see my modules in working order like before.

**Actual Results**: Black screen with my cursor (I noticed the tab is called Electron, not even MagicMirror).

**Configuration**:

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
    address: "localhost", // Address to listen on, can be:
                          // - "localhost", "" to listen on loopback interface
                          // - another specific IPv4/6 to listen on a specific interface
                          // - "", "", "::" to listen on any interface
                          // Default, when address config is left out, is "localhost"
    port:,
    ipWhitelist: ["", "", ":"], // Set [] to allow all IP addresses
                                                           // or add a specific IPv4 of :
                                                           // ["", "", ", ""],
                                                           // or IPv4 range of --> use CIDR format :
                                                           // ["", "", "", ""],

    language: "en",
    timeFormat: 24,
    units: "imperial",
    // serverOnly:  true/false/"local" ,
                 // local for armv6l processors, default 
                 //   starts serveronly and then starts chrome browser
                 // false, default for all  NON-armv6l devices
                 // true, force serveronly mode, because you want to.. no UI on this device
    
    modules: [
        {
            module: "alert",
        },
        {
            module: "updatenotification",
            position: "top_bar"
        },
        {
            module: "clock",
            position: "top_left"
        },
        {
            module: "calendar",
            header: "US Holidays",
            position: "top_left",
            config: {
                calendars: [
                    {
                        symbol: "calendar-check",
                        url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"                   }
                ]
            }
        },
        {
            module: "compliments",
            position: "top_bar"
        },
        {
            module: "currentweather",
            position: "top_right",
            config: {
                location: "",
                locationID: "",
                appid: ""
            }
        },
        {
            module: "weatherforecast",
            position: "top_right",
            header: "Weather Forecast",
            config: {
                location: "",
                locationID: "",  //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
                appid: ""
            }
        },
        {
            module: "newsfeed",
            position: "bottom_bar",
            config: {
                feeds: [
                    {
                        title: "New York Times",
                        url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
                    }
                ],
                showSourceTitle: true,
                showPublishDate: true,
                broadcastNewsFeeds: true,
                broadcastNewsUpdates: true
            }
        },
            {   module: 'MMM-Screencast',
            position: 'center', // This position is for a hidden <div /> and not the screencast window
            config: {
            position: 'center',
            height: 700,
            width: 900,
            }
            },
        {
        module: 'MMM-PIR-Sensor',
        config: {
        // See 'Configuration options' for more information.
            }
        },
        {
        module: "mpd_client",
        position: "top_right",  // This can be any of the regions.
        config: {
            // See 'Configuration options' for more information.
            hostname: "",
            port: 
            }
        },
        {
            module: "MMM-MoonPhase",
            position: "top_center",
            config: {
                updateInterval: 43200000,
                hemisphere: "N",
                resolution: "detailed",
                basicColor: "white",
                title: true,
                phase: true,
                x: 200,
                y: 200,
                alpha: 0.7
                }
        },
        {
         module: 'MagicMirror-QuoteCatalog',
         header: 'The Quote Catalog',
         position: 'top_bar',
         config: {
                     // See below for configurable options
                     }
        },
    ]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

SkyGuy385, 2020-04-05:
If anyone can help me with this problem I would greatly appreciate it. Thank you!

sdetweil, 2020-04-05:
git pull and npm install on the base isn't enough

as part of the instructions, it said to run npm install in the MagicMirror folder.. 
but you also have to do it in all the modules which have a package.json file

but run my upgrade-script from here 
https://github.com/sdetweil/MagicMirror_scripts with **force** instead of **apply**
(because u already have the latest version installed)
it will try to fix all the updates required

user community has reported similar problems with PIR, GoogleAssistant, AssistantMK2, Hotword  needing update but failing (but work with fresh install, so script is updated to be more forceful)

SkyGuy385, 2020-04-05:
Thank you for the quick reply. How do I force your script instead of applying it?

sdetweil, 2020-04-05:
Use the word force at the end instead of apply

SkyGuy385, 2020-04-05:
Wow... was staring at me in the face the whole time. Worked too!

Thank you so much!

closed 2020-04-05 by SkyGuy385