MagicMirrorOrg/MagicMirror #1978: Package for Arch Linux

neilsimp1, 2020-04-06:
Hi, sorry if this is not the most appropriate means to reach out, but I'm not sure how else to do it.

I'm interested in packaging MagicMirror for Arch Linux in the [AUR](https://aur.archlinux.org/).
I'd like to use MagicMirror for a personal project with Arch and when I noticed that it wasn't in the AUR it just made sense to do it myself. I have everything ready, but I wanted your permission first. Please let me know if this is ok with you or not.

sdetweil, 2020-04-06:
what are u  packaging? just the base repo? at version 2.11?

neilsimp1, 2020-04-06:
2.11.0 from https://github.com/MichMich/MagicMirror/releases/tag/v2.11.0.

sdetweil, 2020-04-06:
thx.. do you have some helpers for me to install arch Linux?  I really don't want to learn 20 new commands to do it step by step..

I provide the the automated scripts for install, update and some recovery 
https://github.com/sdetweil/MagicMirror_scripts

there is a lot of detailed work to get this thing to install clean. 
the base is not tested beyond  PI 3/4

neilsimp1, 2020-04-07:
For installing Arch? Anything I wrote would be incomplete or eventually incorrect since things change semi-often. The only guide is on the [wiki](https://wiki.archlinux.org/index.php/Installation_guide).

However, if you don't want to go through all the work just to quickly test something on Arch, I have heard (but not personally used) that [archfi](https://github.com/MatMoul/archfi) is decent. Also, Arco Linux, EndeavorOS, and Manjaro are all pretty darn close to Arch and have graphical installers. That said, only Arch is Arch.

My PKGBUILD for Arch gets this running on x86_64 by the way. I have not tested it on Rasberry Pi or anything ARM based. Currently studying your scripts.

sdetweil, 2020-04-07:
cool.. my scripts work on mac, linux, (lots of different test beds), pi 0-pi 4, odroid, jetson nano, orange pi, 

I didn't start these, just took over what was there, and outdated..  now these have been removed from the base repo, and referenced in the doc, https://docs.magicmirror.builders/getting-started/installation.html#alternative-installation-methods

closed 2020-06-13 by stale[bot]