MagicMirrorOrg/MagicMirror #1985: Not adding optional module header results in 'undefined' as a header

nigel-daniels, 2020-04-13:
**Problem**
I just updated to v2.11.0 and modules that previously had no headers (as expected) are now displaying headers with the word 'undefined'.  

**Expected behavior**
The module configuration says the header is an optional property.  If this is not added then no header should be displayed.

rossburton, 2020-05-18:
Yeah, seeing this too.  Had to set `header: ""` in the forecast module so it didn't show a `undefined - Truro, UK` header.

sejka, 2020-06-16:
i believe that's connected to https://github.com/MichMich/MagicMirror/issues/1897
are you using an older browser by any chance?

nigel-daniels, 2020-06-16:
@sejka, in my case I'm using kweb a minimal kiosk browser.  This did work as expected with a back level version of MM it's only started showing 'undefined' since the MM upgrade to 2.11.0.  Hence raising the issue as it looks like a change in MM has introduced a regression defect.

sejka, 2020-06-16:
exactly, in #1897 i mentioned a commit i found to be problematic: https://github.com/MichMich/MagicMirror/commit/d8b7292d4bd7198ed6756fe5302c3f6ce2179b8c#diff-77ab163850ef0fa98d6a9ee9fb6bbcaf
It was released in 2.11.0.
Almost certain it's the same issue

rossburton, 2020-06-17:
FWIW I was seeing this with Safari 13 on macOS.

nigel-daniels, 2020-06-17:
Sure, did you want to reopen #1897 or resolve the regression under this defect?

sejka, 2020-06-17:
I've submitted PR https://github.com/MichMich/MagicMirror/pull/1908 but as you can see in the discussion we haven't found clear solution. Maybe worth another look @MichMich?
btw @nigel-daniels are you running MagicMirror serveronly using docker image?

sthuber90, 2020-06-25:
Looking for a solution as well, as I wanted to use MM to reuse my old tablet

sthuber90, 2020-07-07:
Apparently the new `logger.js` does not work on old devices with Safari < 11.

```
(function (root, factory) {
	if (typeof exports === "object") {
		// add timestamps in front of log messages
		require("console-stamp")(console, "yyyy-mm-dd HH:MM:ss.l");

		// Node, CommonJS-like
		module.exports = factory(root.config);
	} else {
		// Browser globals (root is window)
		root.Log = factory(root.config);
	}
})(this, function (config) {
	let logLevel = {
		info: Function.prototype.bind.call(console.info, console),
		log: Function.prototype.bind.call(console.log, console),
		error: Function.prototype.bind.call(console.error, console),
		warn: Function.prototype.bind.call(console.warn, console),
		group: Function.prototype.bind.call(console.group, console),
		groupCollapsed: Function.prototype.bind.call(console.groupCollapsed, console),
		groupEnd: Function.prototype.bind.call(console.groupEnd, console),
		time: Function.prototype.bind.call(console.time, console),
		timeEnd: Function.prototype.bind.call(console.timeEnd, console),
		timeStamp: Function.prototype.bind.call(console.timeStamp, console)
	};

	logLevel.setLogLevel = function (newLevel) {
		if (newLevel) {
			Object.keys(logLevel).forEach(function (key, index) {
				if (!newLevel.includes(key.toLocaleUpperCase())) {
					logLevel[key] = function () {};
				}
			});
		}
	};

	return logLevel;
});

```

MichMich, 2020-07-07:
As mentioned earlier, we don't officially support old browsers, but feel free to send a PR if you have a simple solution.

sthuber90, 2020-07-07:
I am trying to. The problem I am having is that the approach from @sejka worked with the last release but does not work with the new release because new issues for old browsers were introduced i.e. logger.js changes. I am aware that you don't want to add babel and/or webpack to your project, even though it would simplify keeping old browsers supported accross releases.

sejka, 2020-07-07:
we could add babel as a docker image step. I've tried it some time ago but couldn't set up babel so that before mentioned issues would be solved. (i don't think there's a polyfill for `HTMLElement API: style` or i couldn't find one)

sthuber90, 2020-07-07:
I also experimented around with a docker image that does the pre-compiling. Couldn't find a pre-compile solution for the HTML style element, but I believe we could resolve the style topic in here, if we can make logger.js work without too much hassle. For the docker image discussion, I think we could add it to the "official" docker image, see issue https://github.com/bastilimbach/docker-MagicMirror/issues/47

kjendal, 2020-07-30:
I think it may be a question of identifying the module that is the offender and adding the `header: '',` line to its config.  For me that was MMM-PIR.  Interestingly, it appears to only be modules in "top_bar" position.

rejas, 2020-08-01:
I made a new PR for this issue, would really appreciate for some feedback in #2103

closed 2020-09-22 by MichMich

TarikHaci, 2023-10-04:
I still have it in the latest karsten13/magicmirror:latest docker environment when I don't specify the header.

PR #2103 (2020-09-02): Fix "undefined" in weather modules header. 
The original PR for this issue (https://github.com/MichMich/MagicMirror/pull/2083) doesnt really fix it for current browsers or electron. 

The issue comes from the data.header being undefined but later filled with values as is the case in the weather modules where the API later fills the header with the name of the location.

This PR fixes #1985 in my opinion for good. It cleans up the header text generation in the weather modules as well as in the module class. It also adds some tests for displaying the header.