MagicMirrorOrg/MagicMirror #1986: Blank screen

likhitha224, 2020-04-14:
After update,when I run the magic mirror module, I am getting a blank screen.
Can someone help me with this issue ?

nubelcr, 2020-04-15:
I'm still new to this but a few things I encountered that you may also be encountering along with the solutions I found so far are;

1. Simply move your mouse and make sure its not "sleeping"
2. Reboot it
3) Did you follow the directions to copy the sample config ? Do you have a config.js file in your config folder ? 

Press the alt key on the screen and go the developer console and see what errors are showing up. Errors normally  bring up a placeholder error screen with the devs name but sometime's i've seen it blank too depending on the error I had in my config.js folder. Because it's js, your quotes, colons and commas must be correct or you're guaranteed to have a bad time. Good luck !

sdetweil, 2020-04-15:
likhitha224   because its after update, its more than likely a module dependency mismatch between then newer electron in this version and the older version...

so, to debug this
1st . look at the console where npm start was done.. 
if pm2, then pm2 logs --lines=100

secondly,  open the developers window, ctrl-shift-i on the keyboard, select the tab labeled console, and scroll up looking for errors, usually red text..
we've seen errors on PIR, and RemoteControl, and of course  Hotword and AssistantMk2.

how did u update? manually, or using the update script (which is suppose to fix these problems).

likhitha224, 2020-04-17:
I updated using the following lines of commands-
git reset --hard; git clean -df;
git pull && npm install

I also checked in console, I found no errors

sdetweil, 2020-04-17:
@likhitha224 there are two consoles, right?   the one where you issued npm start (which has all the output from the node_helpers) and the developers window console (which has all the output from the modules files)

rejas, 2020-04-18:
I can also report a blank screen since a few days. Used to run without a problem, maybe some raspbian update broke electron becuase this is what I get in the pm2 log:
```
0|MagicMirror  | > magicmirror@2.11.0 start /home/pi/MagicMirror
0|MagicMirror  | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
0|MagicMirror  | [2020-04-18 14:40:36.299] [LOG]    
0|MagicMirror  | Starting MagicMirror: v2.11.0
0|MagicMirror  | [2020-04-18 14:40:36.327] [LOG]    
0|MagicMirror  | Loading config ...
0|MagicMirror  | [2020-04-18 14:40:36.350] [LOG]    
0|MagicMirror  | Loading module helpers ...
0|MagicMirror  | [2020-04-18 14:40:36.544] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:36 <log> Initializing new module helper ... (/home/pi/MagicMirror/js/node_helper.js:14 Class.init)
0|MagicMirror  | [2020-04-18 14:40:36.550] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:36 <log> Module helper loaded: MMM-Logging (/home/pi/MagicMirror/js/node_helper.js:18 Class.loaded)
0|MagicMirror  | [2020-04-18 14:40:36.563] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:36 <log> Initializing new module helper ... (/home/pi/MagicMirror/js/node_helper.js:14 Class.init)
0|MagicMirror  | [2020-04-18 14:40:36.567] [LOG]    2020-04-18T14:40:36 <log> Module helper loaded: MMM-MotionDetector (/home/pi/MagicMirror/js/node_helper.js:18 Class.loaded)
0|MagicMirror  | [2020-04-18 14:40:36.697] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:36 <log> Initializing new module helper ... (/home/pi/MagicMirror/js/node_helper.js:14 Class.init)
0|MagicMirror  | [2020-04-18 14:40:36.700] [LOG]    2020-04-18T14:40:36 <log> Module helper loaded: updatenotification (/home/pi/MagicMirror/js/node_helper.js:18 Class.loaded)
0|MagicMirror  | [2020-04-18 14:40:36.710] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:36 <log> No helper found for module: clock. (/home/pi/MagicMirror/js/app.js:127 loadModule)
0|MagicMirror  | [2020-04-18 14:40:37.974] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:37 <log> Initializing new module helper ... (/home/pi/MagicMirror/js/node_helper.js:14 Class.init)
0|MagicMirror  | [2020-04-18 14:40:37.977] [LOG]    2020-04-18T14:40:37 <log> Module helper loaded: MMM-PublicTransportBerlin (/home/pi/MagicMirror/js/node_helper.js:18 Class.loaded)
0|MagicMirror  | [2020-04-18 14:40:37.982] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:37 <log> No helper found for module: compliments. (/home/pi/MagicMirror/js/app.js:127 loadModule)
0|MagicMirror  | [2020-04-18 14:40:37.992] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:37 <log> No helper found for module: currentweather. (/home/pi/MagicMirror/js/app.js:127 loadModule)
0|MagicMirror  | [2020-04-18 14:40:39.339] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Initializing new module helper ... (/home/pi/MagicMirror/js/node_helper.js:14 Class.init)
0|MagicMirror  | [2020-04-18 14:40:39.342] [LOG]    2020-04-18T14:40:39 <log> Module helper loaded: calendar (/home/pi/MagicMirror/js/node_helper.js:18 Class.loaded)
0|MagicMirror  | [2020-04-18 14:40:39.346] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> All module helpers loaded. (/home/pi/MagicMirror/js/app.js:171 loadNextModule)
0|MagicMirror  | [2020-04-18 14:40:39.741] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Starting server on port 8080 ...  (/home/pi/MagicMirror/js/server.js:35 new Server)
0|MagicMirror  | [2020-04-18 14:40:39.793] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Server started ... (/home/pi/MagicMirror/js/app.js:224 )
0|MagicMirror  | [2020-04-18 14:40:39.797] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Connecting socket for: MMM-Logging (/home/pi/MagicMirror/js/node_helper.js:97 Class.setSocketIO)
0|MagicMirror  | [2020-04-18 14:40:39.806] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Module helper started for MMM-Logging (/home/pi/MagicMirror/modules/MMM-Logging/node_helper.js:19 Class.start)
0|MagicMirror  | [2020-04-18 14:40:39.812] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Connecting socket for: MMM-MotionDetector (/home/pi/MagicMirror/js/node_helper.js:97 Class.setSocketIO)
0|MagicMirror  | [2020-04-18 14:40:39.894] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Connecting socket for: updatenotification (/home/pi/MagicMirror/js/node_helper.js:97 Class.setSocketIO)
0|MagicMirror  | [2020-04-18 14:40:39.902] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Connecting socket for: MMM-PublicTransportBerlin (/home/pi/MagicMirror/js/node_helper.js:97 Class.setSocketIO)
0|MagicMirror  | [2020-04-18 14:40:39.909] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Connecting socket for: calendar (/home/pi/MagicMirror/js/node_helper.js:97 Class.setSocketIO)
0|MagicMirror  | [2020-04-18 14:40:39.912] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:39 <log> Starting node helper for: calendar (/home/pi/MagicMirror/modules/default/calendar/node_helper.js:19 Class.start)
0|MagicMirror  | [2020-04-18 14:40:39.917] [LOG]    2020-04-18T14:40:39 <log> Sockets connected & modules started ... (/home/pi/MagicMirror/js/app.js:233 )
0|MagicMirror  | [2020-04-18 14:40:40.211] [LOG]    
0|MagicMirror  | 2020-04-18T14:40:40 <log> MMM-MotionDetector: monitor display_power=1
0|MagicMirror  |  (/home/pi/MagicMirror/modules/MMM-MotionDetector/node_helper.js:49 )
0|MagicMirror  | [2020-04-18 14:40:40.215] [LOG]    2020-04-18T14:40:40 <log> MMM-MotionDetector: monitor on true (/home/pi/MagicMirror/modules/MMM-MotionDetector/node_helper.js:8 )
[... cut ...]
0|MagicMirror  | > magicmirror@2.11.0 start /home/pi/MagicMirror
0|MagicMirror  | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
0|MagicMirror  | [2020-04-18 14:41:09.817] [LOG]    
0|MagicMirror  | Starting MagicMirror: v2.11.0
0|MagicMirror  | [2020-04-18 14:41:09.848] [LOG]    
0|MagicMirror  | Loading config ...
...
```

As you can see pm2 tries to start electron, it succeeds for a while but then seems to crash without a entry in the log (not even in the error part of the log). Is there any other place where I can look for clues?

sdetweil, 2020-04-18:
@rejas there are two pm2 logs, error and normal.. that looks like normal

pm2 logs --lines=100
or look at the ~/.pm2/logs folder 
MagicMirror-error.log
and 
MagicMirror-out.log

rejas, 2020-04-18:
Just like I wrote, the error log doesnt contain errors from the current timestamp.

I just looked in my .pm2/pm2.log found this a lot of the time:

```
2020-04-18T15:08:37: PM2 log: App [MagicMirror:0] starting in -fork mode-
2020-04-18T15:08:37: PM2 log: App [MagicMirror:0] online
2020-04-18T15:08:52: PM2 log: App [MagicMirror:0] exited with code [0] via signal [SIGINT]
2020-04-18T15:08:52: PM2 log: App [MagicMirror:0] starting in -fork mode-
2020-04-18T15:08:52: PM2 log: App [MagicMirror:0] online
```

likhitha224, 2020-04-18:
This is the error i got when i ran the following commands
cd MagicMirror/
npm start

pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.11.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-04-18 18:43:46.819] [LOG]    Starting MagicMirror: v2.11.0
[2020-04-18 18:43:46.834] [LOG]    Loading config ...
[2020-04-18 18:43:46.843] [LOG]    Loading module helpers ...
[2020-04-18 18:43:46.865] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'python-shell'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[2020-04-18 18:43:46.868] [LOG]    Loading module helpers ...
[2020-04-18 18:43:46.872] [ERROR]  App threw an error during load
[2020-04-18 18:43:46.873] [ERROR]  Error: Cannot find module 'python-shell'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js:4:21)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
[2020-04-18 18:43:46.876] [LOG]    Whoops! There was an uncaught exception...
[2020-04-18 18:43:46.877] [ERROR]  Error: Cannot find module 'python-shell'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js:4:21)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[2020-04-18 18:43:46.890] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-04-18 18:43:46.890] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Fontconfig warning: "/etc/fonts/fonts.conf", line 100: unknown element "blank"
[2020-04-18 18:43:47.113] [LOG]    Launching application.

sdetweil, 2020-04-18:
@likhitha224 

[2020-04-18 18:43:46.865] [ERROR] WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'python-shell'
Require stack:

/home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js

yep, that'll do it..

sdetweil, 2020-04-18:
@rejas

so, I would guess u need to redo the npm install in the MagicMirror folder 
do
```
stop MagicMirror
cd ~~/MagicMirror

rm -rf node_modules
rm package-lock.json
npm install --only=prod
```
repeat this (last three steps) for any module that contains a package.json file 
from each module folder

likhitha224, 2020-04-18:
@sdetweil 
i tried but i am getting the following 

pi@raspberrypi:~ $ cd MagicMirror/
pi@raspberrypi:~/MagicMirror $ /home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js
bash: /home/pi/MagicMirror/modules/MMM-Facial-Recognition/node_helper.js: Permission denied

sdetweil, 2020-04-18:
@likhitha224

why are u trying to execute the nod_helper??  the message says that file (nod_helper) cannot find one of its dependencies, 

Cannot find module 'python-shell'

so, is this a new install of this module?  if so, did you do the npm install in the module folder?

likhitha224, 2020-04-18:
@sdetweil 
I am new to this sir, so please tell me what commands to run and what exactly to do

sdetweil, 2020-04-18:
i do not know..  typically when this error occurs (module not found), it is because the instructions for the MagicMirror module installation were not executed

rejas, 2020-04-18:
@sdetweil try this:

```
cd /home/pi/MagicMirror/modules/MMM-Facial-Recognition
npm install
```

Then restart MagicMirror

sdetweil, 2020-04-18:
@rejas  not me.. @likhitha224

likhitha224, 2020-04-18:
@rejas 
omg it is working now...thank you so much @rejas 
I don't see black screen anymore. Thanks a lot...

rejas, 2020-04-18:
i noticed :-)

as for me: seems like another issues as likhitha224 one so I will open a new issue for that...

likhitha224, 2020-04-18:
okay @rejas thanks again for the help

scienceman1972, 2020-04-18:
hi all ,iam new here i would like to ask how can i doing my codes by myself?

likhitha224, 2020-04-18:
@rejas can i close this issue ?

rejas, 2020-04-18:
@scienceman1972 not sure what you mean, please open a new issue for any question. but pelase read the documentation first :-)

rejas, 2020-04-18:
@likhitha224 yes you can

closed 2020-04-18 by likhitha224

sdetweil, 2020-04-18:
@scienceman1972   u mean create a module? 

coding is using some descriptive structure to solve a problem. 
1st understand the problem , and how you would solve it manually.. what steps would you take, be detailed.. 

then use the environment and the available descriptive structure (language and libraries) to implement that...

u can use my sample module  as a starting point 
https://github.com/sdetweil/SampleModule

rejas, 2020-04-18:
@sdetweil re-intalling the dependencies did in fact help, mm starts again. just one quick question: is there anything in the log/errors that gives a hint that something is amiss with the node_modules/dependencies?

sdetweil, 2020-04-18:
@rejas  usually not.. its the weird behavior that points me there to start..  we have seen a lot of SD failures recently, and they also exhibit weird behavior.. 

I did not know that SD cards are a collection of write once bits.   each write marks some part of the card 'used',   when u run out, its toast.. great for cameras and games, seldom written, not so good for running system  paging,  constant  log writing, new software installs/updates etc..

sdetweil, 2020-04-18:
@rejas  this dependency problem is also encountered each MM update (once every 3 months), that is why I made the upgrade script  to do this for people that didn't understand the work.. 
see my repo for instructions
https://github.com/sdetweil/MagicMirror_scripts

rejas, 2020-04-20:
@sdetweil weird, I just re-installed with your script a few weeks ago (after 2.11 came out too). Thats why I didnt think about doing the old "rm & npm i" fix :-)
As for the SDcard: Where did you get the info of write once?`Thats new to me. And would render ALL raspberry software more or less useless...
Wrting less would neverthelesse be a nice thing. What is teh default log level when doing your install in pm2 ?

sdetweil, 2020-04-20:
@rejas >What is teh default log level when doing your install in pm2 ?

eh?   default log level?  
there is a log when installing  ~/install.log
and a log when installing pm2 later MagicMirror/installers/pm2_setup.log

I capture the normal stdout and stderr

rejas, 2020-04-21:
so the user cannot change for example that he only wants to log critical errors or warnings?

sdetweil, 2020-04-21:
no.. logging is for me, not them