MagicMirrorOrg/MagicMirror #1987: webView feature

bugsounet, 2020-04-14:
Hi,

I develop a module but i need special feature in webPreferences: { } of `js/electron.js` file
this feature is `webviewTag: true,`
i will make a PR but i don't know if i force to `true` or report it to config.js file like `zoom` feature

@MichMich : what do you think about that ?

regards, Cédric

MichMich, 2020-04-14:
Cant users use the `electronOptions` setting in the config for this?
https://docs.magicmirror.builders/getting-started/configuration.html#general

bugsounet, 2020-04-14:
yes, of course

like this:
```js
  electronOptions: {
    webPreferences: {
      webviewTag: true
    }
  },
```
but sincerly, i prefer `webView : true` like `zoom` feature
or directly force it to `electron.js`

just because hummm ...
you also know that not all users are very comfortable with the configurations ...

MichMich, 2020-04-14:
That would mean we get a lot of custom options for all types of modules. I think the best solution is to write clear documentation.

bugsounet, 2020-04-14:
no problem I understand, I simply offered this option.
so i will use `electronOptions`

closed 2020-04-14 by MichMich