MagicMirrorOrg/MagicMirror #1993: White screen with mouse cursor

BKeyport, 2020-04-19:
MagicMirror 2.11 bug

**Platform**: Raspbian GNU/Linux 10 (buster) on a Pi 4 using HDMI0 output. 
**Node Version**:  10.15.2 

**Description**: White screen with mouse cursor for 2-5 seconds before magicmirror continues to load, then mirror loads normally. 

**Steps to Reproduce**: Start Magicmirror (I use PM2 to launch mm.sh - which is) 
```
cd ~/MagicMirror
DISPLAY=:0 npm start
```

**Expected Results**: clearing of GUI to black screen, with MagicMirror starting right away. 

**Actual Results**: See Description

**Configuration**: 
```
/**************************** 
 *                          *
 *       Magic Mirror       *
 *       Live  Config       *
 *    By Brendan Keyport    *
 *                          *
 ****************************/
var config = {
	address: "0.0.0.0", // Address to listen on, can be:
	port: 60001,
	ipWhitelist: [], // Set [] to allow all IP addresses
	language: "en",
	timeFormat: 12,
	units: "imperial",
// Modules Follow
	modules: [
		
// Calendar Functionality
		{
			module: 'MMM-CalendarExt2',
				config: {
					calendars : [
						{
							url: "<private>",
							name: "appt", 
							maxItems: 999,
							scanInterval: 30000.00, 
							beforeDays: 0,
							afterDays: 999,
							maxIterations: 999,
							forceLocalTZ: false,
							className: "regcal",
						},
						{
							url: "http://sports.yahoo.com/mlb/teams/sea/ical.ics",
							name: "mariners", 
							maxItems: 999,
							scanInterval: 86400000,
							beforeDays: 0,
							afterDays: 999,
							maxIterations: 999,
							forceLocalTZ: false,
						},
						{
							url: "http://www.trumba.com/calendars/seattle-seahawks-schedule.ics",
							name: "seahawks", 
							maxItems: 999,
							scanInterval: 60000, // once an hour
							beforeDays: 0,
							afterDays: 999,
							maxIterations: 999,
							forceLocalTZ: false,
							className: "hawks",
						}, 
					],
					views: [
						{
							mode: "daily",
							title: "Upcoming Events",
							position: "top_left",
							slotCount: 90,
							timeFormat:"h:mm A", // clock format per event
							//slotTitleFormat: " ", // Left side format
							slotSubTitle: " ", // Right side format
							hideOverflow: false, // if too big, it will hide extras if true 
							name: "Daily View", // required name
							className: "remove_empty_slot", // CSS class to remove empty events
							filterPassedEvent: true, // remove completed events - true yes, false no. 
							calendars: ["appt","trash","seahawks"],
							transform: function(event) {  
								if (event.title.search("Recycle") > -1) {
									event.icon = "mdi:recycle";
								}
								if (event.title.search("Yard Waste") > -1) {
									event.icon = "entypo-leaf";
								}
								if (event.title.search("Seahawk") > -1) {
									event.icon = "fa-solid:football-ball";
								}
								if (event.title.search("Bowling") > -1) {
									event.icon = "maki-bowling-alley-15";
								}
								if (event.title.search("Mariners") > -1) {
									event.icon = "ion-baseball";
								}
								return event;
							},
						},
					],
					scenes: [
						{
							name: "DEFAULT",
						},
					],
				},
		},
		{
			module: "MMM-Multimonth",
			position: "bottom_center",
			config: {
				startMonth: -1, 
				monthCount: 3, 
			}
		},

// Time 
		{
			module: "clock",
			position: "top_right",
			config: {
				displayType: "digital",
				analogFace: "face-005",
				secondsColor: "#FFFFFF",
			}
		},
		{
			module: "worldclock",
			position: "top_right",
			config: {
				timeFormat: 'h A',
				style: 'left',
				clocks: [
					{
						title: "Eastern",
						timezone: "America/New_York",
					},
					{
						title: "Central",
						timezone: "America/Chicago",
					},
				]
			}
		},
// Weather
	{
			module: 'MMM-DarkSkyForecast',
			position: 'top_right',
			header: "Weather",
			classes: "default everyone",
			config: {
  				apikey: '<private>', 
				latitude:  <nope>,
				longitude: <haha>,
				forecastLayout: "table",
				iconset: "2c",
				units: "us",
				hourlyForecastInterval: 1,
				maxHourliesToShow: 3,
				maxDailiesToShow: 4,
				concise: true,
				useAnimatedIcons: false,
				updateFadeSpeed: 0,
			}
		},
// News
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				animationSpeed: 0,
				feeds: [
					{
						title: "Field Gulls",
						url: "https://www.fieldgulls.com/rss/current.xml",
					},
					{
						title: "ESPN NFL Headlines",
						url: "https://www.espn.com/espn/rss/nfl/news",
					},
					{
						title: "ESPN MLB Headlines",
						url: "https://www.espn.com/espn/rss/mlb/news",
					},
					{
						title: "CBS Sports NFL Headlines",
						url: "https://www.cbssports.com/rss/headlines/nfl/",
					},
					{
						title: "CBS Sports MLB Headlines",
						url: "https://www.cbssports.com/rss/headlines/mlb/",
					},
					{
						title: "King5 Local News",
						url: "http://rssfeeds.king5.com/king5/local&x=1",
					},
				],
			},
		},
// Local Temp
		{
			module: "MMM-DHT-Sensor",
			position: "top_right",
			header: "Indoor",
			config: {
				sensorPin: 21,
				sensorType: 22,
				units: "imperial",
				updateInterval: 30000,
				animationSpeed: 0,
				
			},
		},
// System Stats 
		{
			module: 'MMM-SystemStats',
			position: 'top_right', // This can be any of the regions.
			classes: 'small', // Add your own styling. OPTIONAL.
			// header: 'System Stats', // Set the header text OPTIONAL
			config: {
				updateInterval: 10000, // every 10 seconds
				align: 'center', // align labels
				header: 'System Stats', // This is optional
				units: 'imperial', // default, metric, imperial
				label: 'text',
			},
		},
// Prettify 
		{
			module: "MMM-Wallpaper",
			position: "fullscreen_below",
			config: { 
				source: "firetv",
				slideInterval: 1800000, // every 30 minutes
				//slideInterval: 5000, // every 5 seconds for testing
				filter: "grayscale(.75) brightness(0.5)",
			}
		},
// Admin
		{
			module: "updatenotification",
			position: "bottom_center"
		},
		{
			module: "alert",
		},
		{
			module: 'MMM-Remote-Control',
		},
		
// End modules before this line, do not edit below
	]
};
if (typeof module !== "undefined") {module.exports = config;}

sdetweil, 2020-04-19:
white screen flash is an electron problem, with no fix.

khassel, 2020-04-19:
... and already discussed in #1919

closed 2020-04-21 by BKeyport