MagicMirrorOrg/MagicMirror #1994: MM shows black Screen.

Arerel, 2020-04-19:
Hi,
after a littlebit testing with my Magic Mirror, it doesn't show any screen anymore. I can still see my mouse but no modules. I tired to remove my changes but it didn't work, here is my Output

`pi@raspberrypi:/usr/local/bin/MagicMirror $ DISPLAY=:0 npm start &
[2] 1797
pi@raspberrypi:/usr/local/bin/MagicMirror $
> magicmirror@2.11.0 start /usr/local/bin/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-04-19 12:34:44.585] [LOG]    Starting MagicMirror: v2.11.0
[2020-04-19 12:34:44.598] [LOG]    Loading config ...
[2020-04-19 12:34:44.607] [LOG]    Loading module helpers ...
[2020-04-19 12:34:44.610] [LOG]    No helper found for module: alert.
[2020-04-19 12:34:44.685] [LOG]    Initializing new module helper ...
[2020-04-19 12:34:44.686] [LOG]    Module helper loaded: updatenotification
[2020-04-19 12:34:44.687] [LOG]    No helper found for module: clock.
[2020-04-19 12:34:45.333] [LOG]    Initializing new module helper ...
[2020-04-19 12:34:45.334] [LOG]    Module helper loaded: calendar
[2020-04-19 12:34:45.335] [LOG]    No helper found for module: compliments.
[2020-04-19 12:34:45.336] [LOG]    No helper found for module: currentweather.
[2020-04-19 12:34:45.337] [LOG]    No helper found for module: weatherforecast.
[2020-04-19 12:34:45.380] [LOG]    Initializing new module helper ...
[2020-04-19 12:34:45.381] [LOG]    Module helper loaded: newsfeed
[2020-04-19 12:34:45.381] [LOG]    All module helpers loaded.
[2020-04-19 12:34:45.384] [ERROR]  WARNING! Could not find config file. Please create one. Starting with default configuration.
[2020-04-19 12:34:45.385] [LOG]    Loading module helpers ...
[2020-04-19 12:34:45.386] [LOG]    No helper found for module: alert.
[2020-04-19 12:34:45.387] [LOG]    Initializing new module helper ...
[2020-04-19 12:34:45.388] [LOG]    Module helper loaded: updatenotification
[2020-04-19 12:34:45.389] [LOG]    No helper found for module: clock.
[2020-04-19 12:34:45.390] [LOG]    Initializing new module helper ...
[2020-04-19 12:34:45.391] [LOG]    Module helper loaded: calendar
[2020-04-19 12:34:45.392] [LOG]    No helper found for module: compliments.
[2020-04-19 12:34:45.393] [LOG]    No helper found for module: currentweather.
[2020-04-19 12:34:45.395] [LOG]    No helper found for module: weatherforecast.
[2020-04-19 12:34:45.396] [LOG]    Initializing new module helper ...
[2020-04-19 12:34:45.396] [LOG]    Module helper loaded: newsfeed
[2020-04-19 12:34:45.397] [LOG]    All module helpers loaded.
[2020-04-19 12:34:45.398] [ERROR]  App threw an error during load
[2020-04-19 12:34:45.400] [ERROR]  Error: ENOENT: no such file or directory, open
    at Object.openSync (fs.js:447:3)
    at Object.func (electron/js2c/asar.js:138:31)
    at Object.func [as openSync] (electron/js2c/asar.js:138:31)
    at Object.readFileSync (fs.js:349:35)
    at Object.fs.readFileSync (electron/js2c/asar.js:580:40)
    at Object.fs.readFileSync (electron/js2c/asar.js:580:40)
    at new Server (/usr/local/bin/MagicMirror/js/server.js:26:12)
    at /usr/local/bin/MagicMirror/js/app.js:223:18
    at loadNextModule (/usr/local/bin/MagicMirror/js/app.js:172:5)
    at /usr/local/bin/MagicMirror/js/app.js:167:6
[2020-04-19 12:34:45.403] [LOG]    Whoops! There was an uncaught exception...
[2020-04-19 12:34:45.403] [ERROR]  Error: ENOENT: no such file or directory, open
    at Object.openSync (fs.js:447:3)
    at Object.func (electron/js2c/asar.js:138:31)
    at Object.func [as openSync] (electron/js2c/asar.js:138:31)
    at Object.readFileSync (fs.js:349:35)
    at Object.fs.readFileSync (electron/js2c/asar.js:580:40)
    at Object.fs.readFileSync (electron/js2c/asar.js:580:40)
    at new Server (/usr/local/bin/MagicMirror/js/server.js:26:12)
    at /usr/local/bin/MagicMirror/js/app.js:223:18
    at loadNextModule (/usr/local/bin/MagicMirror/js/app.js:172:5)
    at /usr/local/bin/MagicMirror/js/app.js:167:6 {
  errno: -2,
  syscall: 'open',
  code: 'ENOENT'
}
[2020-04-19 12:34:45.413] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-04-19 12:34:45.414] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-04-19 12:34:45.532] [LOG]    Launching application.
`

sdetweil, 2020-04-19:
/usr/local/bin/MagicMirror??

u cannot run this thing as root..

please reinstall as normal user , do NOT  use sudo

use my install script
see https://github.com/sdetweil/MagicMirror_scripts, 
read  the readme fo the command ti issue

Ybbet, 2020-05-07:
Hi,

I’ve updated tonight my MagicMirror and I’ve got the same error. I’m not in root :-)

sdetweil, 2020-05-07:
@Ybbet, how did you update? git pull and npm install??

please the npm install in each module folder that has a package.json file as the system level has changed and all module dependencies need to be updated as well..

if you used the automated upgrade script 
see here https://github.com/sdetweil/MagicMirror_scripts
then it should have been successful... please post the 
~/MagicMirror/installers/upgrade.log

closed 2020-07-04 by MichMich