MagicMirrorOrg/MagicMirror #1999: Deafault weather module https instead of http openweather API

Travelbacon, 2020-04-26:
## I found a bug in MagicMirror
Very small change request. The weathermodule uses http://openweathermap.org/api instead of the https. Can this be changed?

I checked the repository, and it in the current verions weather.js has on line 48 
`apiBase: "http://api.openweathermap.org/data/",`
Because it is just one character, I wouldn't make a pull request.

**Platform**: Every platform
**MagicMirror Version**:  All verions.

sdetweil, 2020-04-26:
>Because it is just one character, I wouldn't make a pull request.

SOMEBODY has to make a pull request, thats the _**ONLY**_ way changes get made

Travelbacon, 2020-04-27:
Okay and why? If somebode who is actively programming in the code adds the character, the result is the same.

sdetweil, 2020-04-27:
no one is actively coding.. its all volunteer, and all our changes are tracked thru pull requests.

rejas, 2020-04-29:
And it's very easy to go from "there is this one line that could be fixed" to "oh, there are a ton of lines to be fixed" as can be seen in my PR :-)

Travelbacon, 2020-04-29:
The reason why I did it here, before more people are putting words in my mouths, is I assumed that more people are working on this project. It ‘s just one character what I’ve found. Why make a complete pull in a version control for one character? I think that’s an overkill. 
Secondly, my working copy is that much changed that a pull doesn’t make sense. 

Good to read there is a complete request to change all the HTTP to secure. I overlooked these pulls. It would make sense for me to add it there.

closed 2020-04-29 by Travelbacon

rejas, 2020-04-30:
Might I ask @Travelbacon what you changed in your copy / fork of the MirrorSoftware? Because maybe you had some other things done that might be useful in here too. So even if you cant make a PR because its deviated too much, feature input is always appreciated.

Travelbacon, 2020-04-30:
https://github.com/Travelbacon/MMM-DutchWeatherStatic
That is the result. Nothing is left of the original code and adapted to my own needs. Two other modules have been published with complete readme and instructions
[TomTom Traffice and Incidents map](https://github.com/Travelbacon/MMM-TomTomTrafficIncidents)
[Departure times of public transport grouped by destination grouped and stops](https://github.com/Travelbacon/MMM-NLDepartureTimes)

rejas, 2020-04-30:
ah, didnt get that you ment module development. I thought you ment the whole MagicMirror project. Anyway, nice module! Cheerio!

PR #2002 (2020-04-30): Replace insecure links with https 
Fixes #1999 and replaces most of the http urls in the code with secure ones