MagicMirrorOrg/MagicMirror #2003: Random crashes to black screen with mouse cursor

bblaha, 2020-04-29:
I realize similar issues have been opened recently, but I was not able to find a solution:
My Magic Mirror randomly stops showing all modules and defaults to a black screen showing just the mouse cursor.
I have installed MMM-Logging, however the logs only show problems related to what I suppose is an internet outage (which does happen occasionally).
Errors look such as the following
```
0|mm       | [2020-04-27 21:50:01.138] [ERROR]  2020-04-27T21:50:01 <error> Calendar Error. Could not fetch calendar:  http://XXX/Calendar/personal.ics Error: getaddrinfo EAI_AGAIN XXX
0|mm       |     at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:60:26) {
0|mm       |   errno: 'EAI_AGAIN',
0|mm       |   code: 'EAI_AGAIN',
0|mm       |   syscall: 'getaddrinfo',
0|mm       |   hostname: 'XXX'
0|mm       | } (/home/pi/MagicMirror/modules/default/calendar/node_helper.js:63 )

```
My guess is that those cannot or at least should not be the cause for the problem.
I have not found any patterns in the crash, it happens at random times with no interaction via SSH or any other input modes.

**Platform**: Raspberry Pi 3

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: 2.11.0

Config file:
```
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
        address: "0.0.0.0", // Address to listen on, can be:
                              // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
                              // - another specific IPv4/6 to listen on a specific interface
                              // - "0.0.0.0", "::" to listen on any interface
                              // Default, when address config is left out or empty, is "localhost"
        port: 8080,
        ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1", "192.168.0.1/24"], // Set [] to allow all IP addresses
                                                               // or add a specific IPv4 of 192.168.1.5 :
                                                               // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
                                                               // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
                                                               // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

        useHttps: false,                // Support HTTPS or not, default "false" will use HTTP
        httpsPrivateKey: "",    // HTTPS private key path, only require when useHttps is true
        httpsCertificate: "",   // HTTPS Certificate path, only require when useHttps is true

        language: "en",
        timeFormat: 24,
        units: "metric",
        // serverOnly:  true/false/"local" ,
                             // local for armv6l processors, default
                             //   starts serveronly and then starts chrome browser
                             // false, default for all  NON-armv6l devices
                             // true, force serveronly mode, because you want to.. no UI on this device

        modules: [
                {
                module: 'MMM-Logging',
                config: {
                // Module functions out of the box with nothing set here.
                // See below for configurable options.
                        }
                },
                {
[... 146 lines cut ...]
                                showSourceTitle: true,
                                showPublishDate: true,
                                broadcastNewsFeeds: true,
                                broadcastNewsUpdates: true,
                                },
                },
                {
                        module: 'MMM-Remote-Control',
        // uncomment the following line to show the URL of the remote control on the mirror
        // , position: 'bottom_left'
        // you can hide this module afterwards from the remote control itself
                        config: {
                                apiKey: ''
                        }
                },

        ]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
(END)

```

sdetweil, 2020-04-29:
when it goes to black screen again, open the developers console, f12 or ctrl-shift-i on the keyboard, and select the console tab.  there should be a fatal error  somewhere..

MichMich, 2020-07-04:
Closing due to inactivity. Feel free to reopen.

closed 2020-07-04 by MichMich