MagicMirrorOrg/MagicMirror #2006: Launches black screen with mouse - New to coding

vDiglet, 2020-05-02:
pi@raspi:~/MagicMirror $ npm start

> magicmirror@2.11.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

/home/pi/MagicMirror/node_modules/electron/dist/electron: /lib/arm-linux-gnueabihf/libdbus-1.so.3: no version information available (required by /home/pi/MagicMirror/node_modules/electron/dist/electron)
/home/pi/MagicMirror/node_modules/electron/dist/electron: /lib/arm-linux-gnueabihf/libdbus-1.so.3: no version information available (required by /home/pi/MagicMirror/node_modules/electron/dist/electron)
[2020-05-02 19:45:24.490] [LOG]    Starting MagicMirror: v2.11.0
[2020-05-02 19:45:24.540] [LOG]    Loading config ...
[2020-05-02 19:45:24.553] [LOG]    Loading module helpers ...
[2020-05-02 19:45:24.557] [LOG]    No helper found for module: alert.
[2020-05-02 19:45:24.700] [LOG]    Initializing new module helper ...
[2020-05-02 19:45:24.702] [LOG]    Module helper loaded: updatenotification
[2020-05-02 19:45:24.704] [LOG]    No helper found for module: clock.
[2020-05-02 19:45:26.062] [LOG]    Initializing new module helper ...
[2020-05-02 19:45:26.063] [LOG]    Module helper loaded: calendar
[2020-05-02 19:45:26.065] [LOG]    No helper found for module: compliments.
[2020-05-02 19:45:26.067] [LOG]    No helper found for module: currentweather.
[2020-05-02 19:45:26.069] [LOG]    No helper found for module: weatherforecast.
[2020-05-02 19:45:26.188] [LOG]    Initializing new module helper ...
[2020-05-02 19:45:26.189] [LOG]    Module helper loaded: newsfeed
[2020-05-02 19:45:26.190] [LOG]    All module helpers loaded.
[2020-05-02 19:45:26.195] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'socket.io'
Require stack:
- /home/pi/MagicMirror/js/server.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[2020-05-02 19:45:26.197] [LOG]    Loading module helpers ...
[2020-05-02 19:45:26.198] [LOG]    No helper found for module: alert.
[2020-05-02 19:45:26.199] [LOG]    Initializing new module helper ...
[2020-05-02 19:45:26.200] [LOG]    Module helper loaded: updatenotification
[2020-05-02 19:45:26.201] [LOG]    No helper found for module: clock.
[2020-05-02 19:45:26.202] [LOG]    Initializing new module helper ...
[2020-05-02 19:45:26.204] [LOG]    Module helper loaded: calendar
[2020-05-02 19:45:26.207] [LOG]    No helper found for module: compliments.
[2020-05-02 19:45:26.208] [LOG]    No helper found for module: currentweather.
[2020-05-02 19:45:26.209] [LOG]    No helper found for module: weatherforecast.
[2020-05-02 19:45:26.210] [LOG]    Initializing new module helper ...
[2020-05-02 19:45:26.210] [LOG]    Module helper loaded: newsfeed
[2020-05-02 19:45:26.211] [LOG]    All module helpers loaded.
[2020-05-02 19:45:26.212] [ERROR]  App threw an error during load
[2020-05-02 19:45:26.214] [ERROR]  Error: Cannot find module 'socket.io'
Require stack:
- /home/pi/MagicMirror/js/server.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at new Server (/home/pi/MagicMirror/js/server.js:33:11)
    at /home/pi/MagicMirror/js/app.js:223:18
    at loadNextModule (/home/pi/MagicMirror/js/app.js:172:5)
    at /home/pi/MagicMirror/js/app.js:167:6
[2020-05-02 19:45:26.216] [LOG]    Whoops! There was an uncaught exception...
[2020-05-02 19:45:26.217] [ERROR]  Error: Cannot find module 'socket.io'
Require stack:
- /home/pi/MagicMirror/js/server.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at new Server (/home/pi/MagicMirror/js/server.js:33:11)
    at /home/pi/MagicMirror/js/app.js:223:18
    at loadNextModule (/home/pi/MagicMirror/js/app.js:172:5)
    at /home/pi/MagicMirror/js/app.js:167:6 {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/js/server.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[2020-05-02 19:45:26.242] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-05-02 19:45:26.242] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Fontconfig warning: "/etc/fonts/fonts.conf", line 86: unknown element "blank"
[2020-05-02 19:45:26.795] [LOG]    Launching application.

sdetweil, 2020-05-02:
can u tell us more? is this right after install?  how did u install? manual or scripted?  
what pi are you using

vDiglet, 2020-05-02:
right after install, using the manual install method, and on pi 3

sdetweil, 2020-05-02:
can u show results of

node v 
npm -v

might want to try scripted install
rename or erase the MagicMirror folder and then run the install script from here
https://github.com/sdetweil/MagicMirror_scripts
read the readme for the command to use

closed 2020-07-04 by MichMich