MagicMirrorOrg/MagicMirror #2015: maybe no internet connection?-process tickandrejection

jimmydew1, 2020-05-09:
Raspberry pi 4 
everything up to date

Magic mirror is set up freshly, but it gives me error when it runs:

[2020-05-09 14:52:14.035] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-05-09 14:52:14.037] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-05-09 14:52:30.541] [LOG]    Whoops! There was an uncaught exception...
[2020-05-09 14:52:30.542] [ERROR]  Error: premature close
    at onclosenexttick (/home/pi/MagicMirror/modules/MMM-MirrorMirrorOnTheWall/node_modules/end-of-stream/index.js:54:86)
    at processTicksAndRejections (internal/process/task_queues.js:82:9)

any idea's???

sdetweil, 2020-05-09:
@jimmydew1 

Error: premature close
at onclosenexttick (/home/pi/MagicMirror/modules/MMM-MirrorMirrorOnTheWall/node_modules/end-of-stream/index.js:54:86)
at processTicksAndRejections (internal/process/task_queues.js:82:9)

looks like MirrorMirrorOnTheWall  is having trouble

jimmydew1, 2020-05-09:
agreed, any ideas on how to fix this?

sdetweil, 2020-05-09:
how did u install? some of these things have big bundles

jimmydew1, 2020-05-09:
i followed every step of these instructions : 
https://joanaz.github.io/MirrorMirrorOnTheWallSkill/

sdetweil, 2020-05-09:
thanks.  i don't believe this is a MagicMirror problem, but rather a module problem, 

I would open an issue here 
https://github.com/joanaz/MirrorMirrorOnTheWallSkill/issues

I see there are some pending pull requests for handling async events, and the aws iot code has changed..

rejas, 2020-05-26:
Can this issue be closed now since you opened one at https://github.com/joanaz/MirrorMirrorOnTheWallSkill/issues/45 @jimmydew1 ?

closed 2020-05-26 by MichMich