MagicMirrorOrg/MagicMirror #2018: Weather forecast ignores "maxNumberOfDays"

mirko3000, 2020-05-11:
Hi there,

currently the config parameter "maxNumberOfDays" of the module "weatherforecast" seems to be ignored - at least it does not work increasing it to more than 7 days. That is because the API needs an additional parameter "cnt", which indicates how many days are to be retrieved. Adding it to the weatherforecast.js in line 305 solves the issue for me:

`params += "&cnt=" + this.config.maxNumberOfDays;`

Best regards,
Mirko

MichMich, 2020-05-11:
Please send a PR.

Thanks!

hangorazvan, 2020-05-16:
params += "&cnt=" + (((this.config.maxNumberOfDays < 1) || (this.config.maxNumberOfDays > 128)) ? 40 : this.config.maxNumberOfDays);

rejas, 2020-05-16:
Maybe add a "Good first bug" to this Issue so people who want to start helping can fix it? Otherwise I will do it sooner than later ;-)

Ekristoffe, 2020-05-29:
> 
> 
> `params += "&cnt=" + (((this.config.maxNumberOfDays < 1) || (this.config.maxNumberOfDays > 128)) ? 40 : this.config.maxNumberOfDays);`

I the documentation it said:
```
How many days of forecast to return. Specified by config.js

Possible values: 1 - 16
Default value: 7 (7 days)
This value is optional. By default the weatherforecast module will return 7 days.
```
I think only a limit of 16 would be nice.
And i don't think anyone need a forecast of more than 16 days.

sdetweil, 2020-05-29:
>And i don't think anyone need a forecast of more than 16 days.

and more importantly, can you GET  a forecast from any api for more that 15 days?

Ekristoffe, 2020-05-30:
Well I can make it work up to 17  after there is no answer.

lavolp3, 2020-07-02:
Still seems to be an issue
https://forum.magicmirror.builders/topic/13187/weatherforecast-showing-only-two-days/6

closed 2020-07-04 by MichMich

ember1205, 2020-09-28:
I wanted to add a comment here that this issue is not actually "fixed" as the new version now reverts forecasts back to showing only two days (at least when using the Free API). In doing some research, it appears that the actual URL being used to retrieve the forecast is not the correct / current one.

From what I can tell, the base URL is being built out as:

`https://api.openweathermap.org/data/2.5/forecast??id={id}&appid={API key}`

According to the API docs, it should be built out as:

`https://api.openweathermap.org/data/2.5/onecall?lat={lat}&lon={lon}&exclude={part}&appid={API key}`

PR #2066 (2020-07-03): Adjusts maxNumberOfDays depending on the endpoint 
Fixes #2018 

Now the `cnt` is set depending on the endpoint
- for 'forecast/daily' the `maxNumberOfDays` is used directly
- for 'forcast' `maxNumberOfDay` is adjusted. The maximum here is 5 days

The documentation for this module need to be adjusted as well:
https://docs.magicmirror.builders/modules/weatherforecast.html#configuration-options

If a free API token is used this description fits the maxNumberOfDays config:
Possible values: 1 - 5
Default value: 5 (5 days) 

Else the config description remains the same
PR #2077 (2020-07-17): Fixes getting only full day forecasts
Finally I had time to examine the API well enough.
It seems that the correct function to use here is the round function.
Now the forecast requested from the API will always fetch full days only.

Fixes #2018 
Corrects the PR #2066