MagicMirrorOrg/MagicMirror #2019: How to build / buy MagicMirror?

orsenthil, 2020-05-12:
I learned about MagicMirror(2) today, and I wanted to instructions to either buy/build the hardware using raspberry pi and additional hardware pieces.

I do not find that information in https://magicmirror.builders/

Where can I find the complete documentation for a beginner who wants to set up MagicMirror?

nikobix, 2020-05-12:
Hi Orsenthil
The MagPi magazine has several tutorials on building magic mirrors, from materials to modules. 
Its seems to be a regular feature in the mag so far.
https://magpi.raspberrypi.org/articles/category/tutorials
other than that there are many websites with details on Magic Mirror builds using MichMich platform.

I am sure others will comment as well.

Cheers
Nik (NOOB builder)

orsenthil, 2020-05-12:
Thanks, Nik,

This was the starting tutorial https://magpi.raspberrypi.org/articles/build-a-magic-mirror - I was looking for.  I will look at the rest of the links and suggest the good addition to the documentation.

closed 2020-05-24 by MichMich