MagicMirrorOrg/MagicMirror #2022: Calendar fails to parse my iCal

rossburton, 2020-05-18:
With MagicMirror 2.11.0 (tag, from git), pointing the calendar module at my private iCal causes an error:

```
[2020-05-18 16:35:07.063] [ERROR]  Error: Unknown RRULE property 'RRULE'
    at /Users/ross/Programming/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1191:23
    at Array.forEach (<anonymous>)
    at parseRrule (/Users/ross/Programming/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1149:11)
    at parseLine (/Users/ross/Programming/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1138:20)
    at Array.map (<anonymous>)
    at Function.parseString (/Users/ross/Programming/MagicMirror/node_modules/rrule/dist/es5/rrule.js:1110:41)
    at Function.RRule.fromString (/Users/ross/Programming/MagicMirror/node_modules/rrule/dist/es5/rrule.js:2135:32)
    at Object.ical.objectHandlers.END (/Users/ross/Programming/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:73:23)
    at Object.handleObject (/Users/ross/Programming/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:400:41)
    at Object.parseICS (/Users/ross/Programming/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:441:20)
```

rossburton, 2020-05-18:
Added some debugging to the `END` object handler in `node-ical.js`:

```
[2020-05-18 16:47:38.087] [ERROR]  got event  {
  type: 'VEVENT',
  params: [],
  start: 2008-08-07T13:00:00.000Z { tz: 'Europe/London' },
  end: 2008-08-07T15:30:00.000Z { tz: 'Europe/London' },
  rrule: 'RRULE;X-EVOLUTION-ENDDATE=20080828T130000Z:FREQ=WEEKLY;COUNT=4;BYDAY=TH',
  dtstamp: 2020-05-18T15:47:37.000Z { tz: undefined },
  uid: '20080727T171740Z-5353-1000-1-3243@blackadder',
  class: 'PUBLIC',
  created: 2010-08-03T12:40:59.000Z { tz: undefined },
  description: '',
  lastmodified: 2010-08-16T03:18:27.000Z { tz: undefined },
  sequence: '2',
  status: 'CONFIRMED',
  summary: ' [... ] ',
  transparency: 'OPAQUE'
}
```

Looks like a slightly mangled event, but MagicMirror shouldn't be erroring out like that (instead of skipping the event, it skips reading the entire calendar).

sdetweil, 2020-05-20:
well, the reader crashes.. (not our code)

MichMich, 2020-07-04:
Closing due to inactivity. Feel free to reopen.

edmondchiu92, 2020-07-10:
Hello, wondering if we are able to re-open this issue. I am setting up my Magic Mirror for the first time on a Raspberry Pi. It seems my Google Calendar experiences a similar error to above. The sample calendar, and another one from calendarlabs.com worked fine. Along with a Office365 published ICS calendar.

Here is the error log when I try to load in my google calendar:

[2020-07-10 02:11:07.784] [ERROR]  Error: Unknown RRULE property 'RRULE'
    at Function.parseString (/home/pi/MagicMirror/node_modules/ical/node_modules/rrule/dist/es5/rrule.js:1341:23)
    at Function.RRule.fromString (/home/pi/MagicMirror/node_modules/ical/node_modules/rrule/dist/es5/rrule.js:1607:32)
    at Object.ical.objectHandlers.END (/home/pi/MagicMirror/node_modules/ical/node-ical.js:43:23)
    at Object.handleObject (/home/pi/MagicMirror/node_modules/ical/ical.js:406:41)
    at Object.parseICS (/home/pi/MagicMirror/node_modules/ical/ical.js:462:20)
    at Request._callback (/home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js:61:22)
    at Request.self.callback (/home/pi/MagicMirror/node_modules/request/request.js:185:22)
    at Request.emit (events.js:200:13)
    at Request.<anonymous> (/home/pi/MagicMirror/node_modules/request/request.js:1154:10)
    at Request.emit (events.js:200:13)
[2020-07-10 02:11:07.790] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-07-10 02:11:07.790] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues

If it is indeed the reader crashing, are the steps that can be taken to resolve?

MichMich, 2020-07-10:
Will need to catch these exceptions. Reopened this issue.

rejas, 2020-07-10:
Catching that exception is easy. The problem is only that it would mean we couldnt show anything at all of that calendar since the triggering of that exception is so deep in the stack (in the rrule lib).

edmondchiu92, 2020-07-10:
Thank you, let me know how I can help

rejas, 2020-07-10:
Could you provide a sample ics file which throws an error? I dont have Office365 :-)

rejas, 2020-07-20:
@edmondchiu92 @rossburton can anyone of you provide me with an ics file that throws the error so I can reproduce it locally?

rossburton, 2020-07-20:
Sorry, I ended up deleting the event that was breaking this for me.

rejas, 2020-07-20:
Any chance you can recreate a similar dummy event @rossburton ?

rossburton, 2020-07-20:
If I recall correctly the malformed bit was a mangled RRULE line:

```
RRULE=RRULE;X-EVOLUTION-ENDDATE=20080828T130000Z:FREQ=WEEKLY;COUNT=4;BYDAY=TH
```

rejas, 2020-07-20:
Mh.... putting that into an event doesnt trigger the exception for me :-(

rejas, 2020-09-19:
Maybe @rossburton can still help me create an event that triggers the exception?

MichMich, 2020-09-22:
Could you give this a try with the current `develop` branch? I just merged a PR by @sdetweil which might solve this.

sdetweil, 2020-09-22:
testing with the node-ical lib indicates that the  prefix 
RRULE= is ignored, 

as all lines in ICS should be 
keyword:

PR #2210 (2020-12-08): Catch errors when parsing calendar data with ical
Fixes #2022
closed 2020-11-29 by stale[bot]