MagicMirrorOrg/MagicMirror #2028: Calendar: .hide() method?

njoerd114, 2020-05-26:
I just finished setting up my [face recognition](https://github.com/nischi/MMM-Face-Reco-DNN) and am pretty happy so far. However I'd like to show and hide calendar elements depending on somebody's presence (i.e. being logged in/out).

This works quite good with whole modules, but not for calendars. I think it's a common scenario, isn't it?

Show Calendar A to Person A, but not to Person B, Show Calendar B to Person B but not to Person A, Show Calendar C to both, but not to everybody and so on...

Any idea on how I could solve this?

MichMich, 2020-05-26:
Yeah, this is possible by using classes:

```
		{
			module: "calendar",
			position: "top_left",
			classes: "calendarA"
			config: { ... }
		},
		{
			module: "calendar",
			position: "top_left",
			classes: "calendarB"
			config: { ... }
		},
```

and use that class to target the module:

```
MM.getModules().withClass('calendarB').enumerate((module) => {
    module.show();
});

MM.getModules().withClass('calendarA').enumerate((module) => {
    module.hide();
});
```

njoerd114, 2020-05-26:
Ok, did not think about having multiple instances, works like a charm...

closed 2020-05-26 by njoerd114