MagicMirrorOrg/MagicMirror #2030: Clock can be off by a minute

Martin-Palringo, 2020-05-27:
With MagicMirror (confirmed on 2.11.0), when the clock module is configured to show only minutes, it updates every minute. This update, however, does not correctly happen on (or just after) the minute changes - so the displayed time often drifts to be a minute out from the real time.

eg. time is 22:15, MM2 shows 22:14.

I would propose calculating how long is left till the next minute, setting the next setTimeout to refresh the clock shortly after the next minute happens.

rejas, 2020-05-31:
That seems reasonable. Would you be able to send a PR for this or do you need help`with it?

Martin-Palringo, 2020-06-06:
I'm happy to send a PR - hopefully I'll get a moment to do this in the next few days.

commit 2d3849a6 (2020-07-01): Merge pull request #2061 from ZoneMR/patch-1

Fix #2030 - Clock can be off by a minute
commit ea31d346 (2020-07-01): Fix #2030 - Clock can be off by a minute

Set minute/second in our model based on the actual time from moment() rather than our own attempt to track the time - which can drift or fail to respond to time changes.

Also, schedule time refreshes to happen 50ms after the minute/second is expected to change - preventing premature fires and rapid re-firings of notifications due to accuracy limits in setTimeout
closed 2020-07-01 by MichMich