MagicMirrorOrg/MagicMirror #2033: forum down

sdetweil, 2020-05-28:
unable to logon 
```
Login Unsuccessful
<div class="alert alert-danger"> <strong>Internal Error.</strong> <p>Oops! Looks like something went wrong!</p> <p>&#x2F;login</p> <p>Cannot find module &#x27;..&#x2F;encodings&#x27; Require stack: - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;node_modules&#x2F;body-parser&#x2F;node_modules&#x2F;iconv-lite&#x2F;lib&#x2F;index.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;node_modules&#x2F;body-parser&#x2F;node_modules&#x2F;raw-body&#x2F;index.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;node_modules&#x2F;body-parser&#x2F;lib&#x2F;read.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;node_modules&#x2F;body-parser&#x2F;lib&#x2F;types&#x2F;json.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;node_modules&#x2F;body-parser&#x2F;index.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;node_modules&#x2F;express&#x2F;lib&#x2F;express.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;node_modules&#x2F;express&#x2F;index.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;src&#x2F;webserver.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;src&#x2F;start.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;require-main.js - &#x2F;home&#x2F;forge&#x2F;forum.magicmirror.builders&#x2F;app.js</p> </div>

```

MichMich, 2020-05-28:
Thanks for the heads up!

Should be fixed now. Can you give it a try?

sdetweil, 2020-05-28:
yep, thanks

closed 2020-05-28 by sdetweil