MagicMirrorOrg/MagicMirror #2034: Update module

Doudy, 2020-05-29:
hello
I have two MMM-Domoticz modules.
MMM-Domoticz-1 : displays the temperatures
MMM-Domoticz-2 : displays other informations
With the MMM-Domoticz-1.js and MMM-Domoticz-2.js  depending on the module.
MM tells me that I have to update my modules.
I made a copy of the .js file because I made changes to it.

I tried with git pull but MM refuses to update.

As I changed the name is there not a problem for the update?

[image]

sdetweil, 2020-05-29:
because you made changes,  you will have to make the same changes manually

i suggest u install a new copy and then migrate the changes over to the new folder and swap folders one for each set of modifications..
git clone  ... new-name

did u change the file for code reasons, or settings?

rejas, 2020-05-29:
you can also "stash" your current changes with git and re-apply them after updating

Doudy, 2020-05-30:
>@ sdetweil
>did u change the file for code reasons, or settings?

I made the changes in order to have a display of certain data in color.
Thanks for your advices.

closed 2020-07-04 by MichMich