MagicMirrorOrg/MagicMirror #2038: BLACK SCREEN .

aamir6629, 2020-06-05:
HERE IS TERMINAL LOG.

pi@raspberrypi:~/MagicMirror $ npm start dev

> magicmirror@2.11.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js "dev"

[2020-06-05 08:24:47.909] [LOG]    Starting MagicMirror: v2.11.0
[2020-06-05 08:24:47.922] [LOG]    Loading config ...
[2020-06-05 08:24:47.931] [LOG]    Loading module helpers ...
[2020-06-05 08:24:47.934] [LOG]    No helper found for module: alert.
[2020-06-05 08:24:48.023] [LOG]    Initializing new module helper ...
[2020-06-05 08:24:48.024] [LOG]    Module helper loaded: updatenotification
[2020-06-05 08:24:48.025] [LOG]    No helper found for module: clock.
[2020-06-05 08:24:48.707] [LOG]    Initializing new module helper ...
[2020-06-05 08:24:48.708] [LOG]    Module helper loaded: calendar
[2020-06-05 08:24:48.709] [LOG]    No helper found for module: compliments.
[2020-06-05 08:24:48.710] [LOG]    No helper found for module: currentweather.
[2020-06-05 08:24:48.711] [LOG]    No helper found for module: weatherforecast.
[2020-06-05 08:24:48.718] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module '@bugsounet/lpcm16'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Hotword/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[2020-06-05 08:24:48.720] [LOG]    Loading module helpers ...
[2020-06-05 08:24:48.721] [LOG]    No helper found for module: alert.
[2020-06-05 08:24:48.722] [LOG]    Initializing new module helper ...
[2020-06-05 08:24:48.722] [LOG]    Module helper loaded: updatenotification
[2020-06-05 08:24:48.723] [LOG]    No helper found for module: clock.
[2020-06-05 08:24:48.724] [LOG]    Initializing new module helper ...
[2020-06-05 08:24:48.725] [LOG]    Module helper loaded: calendar
[2020-06-05 08:24:48.725] [LOG]    No helper found for module: compliments.
[2020-06-05 08:24:48.726] [LOG]    No helper found for module: currentweather.
[2020-06-05 08:24:48.727] [LOG]    No helper found for module: weatherforecast.
[2020-06-05 08:24:48.730] [ERROR]  App threw an error during load
[2020-06-05 08:24:48.732] [ERROR]  Error: Cannot find module '@bugsounet/lpcm16'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Hotword/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Hotword/node_helper.js:9:16)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
[2020-06-05 08:24:48.734] [LOG]    Whoops! There was an uncaught exception...
[2020-06-05 08:24:48.735] [ERROR]  Error: Cannot find module '@bugsounet/lpcm16'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Hotword/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Hotword/node_helper.js:9:16)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-Hotword/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[2020-06-05 08:24:48.750] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-06-05 08:24:48.750] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-06-05 08:24:48.959] [LOG]    Launching application.

aamir6629, 2020-06-05:
Issue is because of MMM-Hotword module. 

As soon as I removed it from my config file MAGICMIRROR2 started to work flawlessly. 

MMM-ASSISTANTMK2 also started and greeted with my name and time. 

The only issue is MMM-HOTWORD module I think. Can be wrong too.

sdetweil, 2020-06-05:
Hotword and Assistant have been replaced, and will no longer receive any development or support

please move to the new MMM-GoogleAssistant module
https://github.com/bugsounet/MMM-GoogleAssistant

aamir6629, 2020-06-05:
Thanks for the reply that was really helpful.

Installed successfully and installation was very easy .

My output device is configured perfectly . I do hear the Google chime
sound. But assistant does not outputs sound.

Only text is displayed no sound from assistant.

Help me again please

sdetweil, 2020-06-05:
@aamir6629  I hink you will have to come to the forum for the author to help with this.  I know that he has been working on solving sound problems..  what is your speaker connection?  tv via hdmi,
speaker via usb, or the jack on the pi?

as a test, if u do from a terminal window

arecord test.wav
then say something
and then ctrl-c to end arecord 
and aplay test.wav
does it work and do you hear your words?

you may need a sound control file made
~/.asoundrc
aplay -l will show you the list of devices for outputing sound.

aamir6629, 2020-06-06:
Arecord tested works fine.
Output audio tested works fine.

Played youtube via chromium
Played test.wav

Sound output hw is usb sound card.

sdetweil, 2020-06-12:
you solved this with help from the forums right?

MichMich, 2020-07-04:
Closing due to inactivity. Feel free to reopen.

closed 2020-07-04 by MichMich