MagicMirrorOrg/MagicMirror #2039: PM2 autostart is not working in 2.11 in case of electron

El-Tonno, 2020-06-08:
## PM2 autostart is not working with 2.11 in case of electron...

PM2 Logs:

0|mm  | > magicmirror@2.11.0 start /home/pi/MagicMirror
0|mm  | > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
0|mm  | [13529:0608/154800.047435:FATAL:atom_main_delegate.cc(210)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.

[image]

adding "--no-sandbox" to package.json worked for me.

  "scripts": {
    "start": "DISPLAY=\"${DISPLAY:=:0}\" ./node_modules/.bin/electron js/electron.js **--no-sandbox**",

sdetweil, 2020-06-08:
how did u install?     my install script updates the sandbox files with the right permissions so you shouldn't need the --no-sandbox parm

El-Tonno, 2020-06-08:
did a fresh buster installation these days, and followed the manual installation instruction...

https://docs.magicmirror.builders/getting-started/installation.html

Xorg is set to autologin as pi ...

don´t know why erverything is running as root.... could be also an issue with PM2

sdetweil, 2020-06-08:
ok, down about 5 lines

Alternative Installation Methods

i have an automated script that handles all these problems.  (installs pm2, turns off screensaver) both if u say yes

https://github.com/sdetweil/MagicMirror_scripts

update coming in 3 weeks, my upgrade script hopefully will help

El-Tonno, 2020-06-08:
ok might be ... but there is a short advise to go for manual installation...:

**"Therefore the only officially supported way of installation is by using a manual installation. "**

...and if you do so ... you will stuck on pm2 with this issue

would be nice to have an automatic solution which can handle this, but therefore this solution in my opinion has to be prefered at all ? ... and not an advise to go for manual installation

ok works for me ... it´s a workarround for all out there ... used the manual way

looking forward to have a one click adventure .... nice job at all :-)

thanks a lot Sam

sdetweil, 2020-06-09:
the scripts were removed from the repo in version 2.11, because they require constant attention and updates, but the project is on a 3 month release cycle.. 

the wording is unlikely to change

MichMich, 2020-07-04:
Closing due to inactivity. Feel free to reopen.

closed 2020-07-04 by MichMich

websterjazz, 2021-05-30:
To setup the Startup Script, copy/paste the following command:

> sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi