MagicMirrorOrg/MagicMirror #2042: RFC: Should notifications be considered as part of the public interface?

edward-shen, 2020-06-11:
(Let me know if this is isn't appropriate here, but I think it's more appropriate here since the wiki's here and not on the forums.)

#### Who am I?
I'm not active in the development of MagicMirror so if I'm stepping on some toes I apologize. I am the author of MMM-pages, a small module that gets enough attention that I still get new issues from time to time asking for help, so I think I am in at least some position to voice my suggestions.

#### Why I'm asking
I've noticed that more and more of the issues I've been getting have been with regards to interoperability with modules that expose the notifications API to the user. This makes sense: as the platform matures, more and more modules are going to offer more powerful ways to customize a magic mirror to the user. Exposing the notifications system (either by providing ways to send what notification and when or by mentioning which notifications that this module listens to) is often the easiest way to do this. However, it seems that there's a difference in what modules are doing now and how Core treats them.

#### What's present
At the very least, notifications are treated as an internal concept only, to third-party modules to interact with Core and with each other. You can see this by looking on the sidebar of the wiki: Under the Module Development Documentation page (which 404s, by the way), there's a short page on notifications. The page itself is barely helpful to end users, as it only provides a short description of what notifications are followed by links to internal documentation (which also 404s) and an explanation of what internal modules are sent. 

For users that are new to programming or have no experience at all but want to customize their mirror further, this is effectively the end of the line for them. The forums don't provide an explanation of what notifications are (they'll kindly point back to the wiki) and some module authors (myself included) will think that this is out of the scope for support, as it's more of a Core topic than it is with a 3rd party module issue specifically.

#### What I propose
I don't think it's right to ask module developers to limit the flexibility of their modules, so I think it's best if we improve the visibility of notifications instead. Instead, treat them as an exposed official feature that users should get familiar with (like with 3rd party modules), and provide documentation and whatnot to have some sort of documentation parity with 3rd party modules. Since we don't have modules that provide automatic interoperability between certain modules (and I think that's generally against the DIY philosophy here), we should instead empower users to understand the system itself. This would reduce the disconnect between using a module that exposes the notifications system, and in the spirit of DIY learning.

----

While I'd like to see this change, I'm unable to provide the time & effort it deserves. I do think that this would be a good first task for someone looking to get started with contributing to Core, though, if this gets accepted.

Thanks for reading.

sdetweil, 2020-06-11:
while I don't think its a core responsibility to document notification usage by 3rd party modules, I do think its a useful idea.. if we can find author(s) and maintainer(s)
i personally don't think this will be useful for much of the user community, maybe more for developers

I think and entry that is

  module name (could be a link to the 3rd party  list entry
        produces
             notification-name, data, version, date, purpose, description
        consumes
             notification-name, data, version, date, source module name, usage 

etc

edward-shen, 2020-06-11:
What I meant was to provide an introduction to the concept of notifications to the end user, rather than just hiding it away for developers—I don't think Core should document how 3rd party modules use notifications either.

MichMich, 2020-06-12:
I think it's a good idea to add some information about this topic to the docs. In the docs we can link to a wiki page which documents the existing notifications. There we can make a list of used notification with their payload structure and a list of modules that broadcast or consume those notifications.

Feel free to create such a wiki page. And more importantly. Feel free to send a PR to the documentation repo: https://github.com/MichMich/MagicMirror-Documentation

MichMich, 2020-07-04:
Closing issue. Feel free to reopen.

closed 2020-07-04 by MichMich