MagicMirrorOrg/MagicMirror #2047: Weatherforecast module bug

sipuli93, 2020-06-17:
**Platform**: Electron@6.1.7 on Raspberry Pi 4

**Node Version**: 10.20.1

**MagicMirror Version**: 2.11.0

**Description**: Using free openweathermap account and limiting forecast to under 7 days (6 days in my case) causes the last day of the forecast show only night data. 

[image]
Also nighttime weather icon is shown.

**Steps to Reproduce**: 
Use  "forecast" as endpoint for openweathermap api (free account)
Limit maxNumberOfDays to under 7

**Expected Results**: I expect to see Min and Max temperatures based on full day and to have day icon in forecast.

**Actual Results**: The last day of the forecast doesn't show data from whole day.

**Configuration**: 
var config = {
	address: "localhost", 
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"],
	useHttps: false,
	httpsPrivateKey: "",
	httpsCertificate: "",
	language: "fi",
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,

	modules: [
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left",
			config: {
				displaySeconds: false,
				timezone: "Europe/Helsinki"
			}
		},
		{
			module: "calendar",
			header: "Tapahtumat",
			position: "top_left",
			config: {
				maximumNumberOfDays: 90,
				calendars: [
					{
						symbol: "calendar-check",
						url: "https://rocketcal.s3.eu-central-1.amazonaws.com/launches.ics"
					}
				]
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Sääennuste",
			config: {
				location: "Alavus",
				tableClass: "large",
				maxNumberOfDays: 6,
				forecastEndpoint: "forecast",
				roundTemp: true,
				locationID: "661352",
				appid: "xxxxxxxxxxxxxxxx"
			}
		}
	]

};

**Additional Notes**: Moving forecast length check from lines 363-365 to line 351 seems to fix problem as we want prevent extra days from showing rather than stopping right after reaching the last day.

hangorazvan, 2020-06-17:
using the forecast endpoit give you the 3 by 3 hour forecast

We = 09h
Th = 12h
Fr = 15h
Sa = 18h
Su = 21h
mo = 00h

so last one is night

to display hours instead days you need to change line 343

`day = moment(forecast.dt_txt, "YYYY-MM-DD hh:mm:ss").format("ddd");`

with

`day = moment(forecast.dt_txt, "YYYY-MM-DD hh:mm:ss").format("HH");`

but if you want to use forecast for days you must use endpoint `forecast/daily`

sipuli93, 2020-06-18:
> using the forecast endpoit give you the 3 by 3 hour forecast
> 
> Your hour is 11:43 not yet 12.00 so
> 
> We = 09h
> Th = 12h
> Fr = 15h
> Sa = 18h
> Su = 21h
> mo = 00h
> 
> then last one is in the night

/forecast endpoint gives forecast for next 5 days in 3 hour steps ([Openweathermap](https://openweathermap.org/forecast5))
processWeather function iterates those 40 forecasts to get min and max temperatures for each day.

> hour = ... has no sens

The hour is used to get weather icon within the day. At line 374.

> but if you want to use forecast for days you must use endpoint `forecast/daily`

That would work as I wanted, but it's only for paid users.

MichMich, 2020-07-04:
This is not a bug, but a limitation based on open weather. Feel free to send a PR if you know a nice solution. Closing this issue for now.

closed 2020-07-04 by MichMich

sthuber90, 2020-09-02:
To warm this up and to checkin whether @sipuli93 has already found a solution, wouldn't it be possible to get for exampe one more day than you actually want to show?

sipuli93, 2020-09-02:
I ended up to use weather module with type forecast. It uses same endpoint and does same thing properly.