MagicMirrorOrg/MagicMirror #2050: Recurring events don't show up w/ the default Calendar

chamakura, 2020-06-19:
## I found a bug in MagicMirror

**Platform**: Chrome on Windows

**Node Version**: 14.4

**MagicMirror Version**: v2.11.0

**Description**: Recurring Calendar events don't show up on MagicMirror. I suspect this is because the logic to handle `maximumEntries` is incorrectly handling events in the past. In the example below, there are six recurring events scheduled for the day. As `maximumEntries` is set to 2, 'Test 5' and 'Test 6' should show up on the Calendar as 'Test [1 - 4]' have already occurred in the day. But MagicMirror does not show any future events in this case as it probably considers all six events as eligible candidates, but after applying the `maximumEntries` condition, it probably picks 'Test 1' and 'Test 2' as the two entities. But it doesn't show them as they are in the past.

Please note that this issue doesn't happen if the events are not recurring.

[image]
[image]

**Steps to Reproduce**:
* Create multiple recurring events on the Calendar
* Set `maximumEntries` to a number that's smaller than the events on a given day
* Load MagicMirror after `maximumEntries` events have already happened

**Expected Results**:
The remaining events for the day should show up.

**Actual Results**:
MM does not show any events.

**Configuration**: 
`{
            module: "calendar",
            header: "US Holidays",
            position: "top_left",
            config: {
                calendars: [
                    {
                        symbol: "calendar-check",
                        url: "***",
                    },
                ],
                fade: false,
                maximumNumberOfDays: 1,
                timeFormat: "absolute",
                dateFormat: "LT",
                getRelative: 0,
                maximumEntries: 2,
                excludedEvents: ['Set Status'],
            }
        }
`

**Additional Notes**:
I was able to get it to work correctly by commenting out [this line]( https://github.com/MichMich/MagicMirror/blob/master/modules/default/calendar/calendarfetcher.js#L344). I can submit this as a fix if it looks ok.

sdetweil, 2020-06-19:
yes, the calendar entry fetcher reduces the count to the first maximumEntries, but the 
display code then removes entries in the past. and ends up with nothing to display
the work should be combined on the module js and removed from the fetcher, which should just fetch

chamakura, 2020-06-20:
I'm getting an error while pushing the fix for this issue:

```
MagicMirror git:(chamakura_mm_fix_2050) ✗ git push origin chamakura_mm_fix_2050:develop
ERROR: Permission to MichMich/MagicMirror.git denied to chamakura.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

Do I need any special permission to be able to submit a change? I couldn't find anything specific in the instructions.

rejas, 2020-06-20:
Did you fork the repo? You can only push to your repository and then submit a PullRequest to the original repository here.

chamakura, 2020-06-20:
Thanks, @rejas! I've submitted the fix now.

closed 2020-06-21 by chamakura