MagicMirrorOrg/MagicMirror #2056: ReSpeaker 2-Mics Pi HAT no audio in- or output after MagicMirror installation

EHammer98, 2020-06-24:
I want to use MMM-GoogleAssistant for my Raspberry Pi Zero W.
for that some people suggest getting [this ](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/)card.
It's a nice card and works well on a clean install of the latest Rasbian.
But it stops working when i install MagicMirror using this command: `yes | bash -c "$(curl -sL  https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/raspberry.sh)" `
Alsamixer and my pi's GUI do still see the card and i can adjust it. But it just won't work, no audio in- or output.
A simple USB-card or audio from HDMI do work fine.

I also tried installing the driver again [following ](https://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/)these instructions.
Also my .asoundrc file is set as it should be.

Also a weird thing: It auto-mutes on start up. this has been solved by unstalling pulseaudio and doing some other comands.

sdetweil, 2020-06-24:
i would reopen this in the installer repo
https://github.com/sdetweil/MagicMirror_scripts

if you do the manual install 
```
git clone ..... 
cd MagicMirror
npm install 
```
does it still fail?

EHammer98, 2020-06-24:
> i would reopen this in the installer repo
> https://github.com/sdetweil/MagicMirror_scripts
> 
> if you do the manual install
> 
> ```
> git clone ..... 
> cd MagicMirror
> npm install 
> ```
> 
> does it still fail?

@sdetweil Thanks for the fast reply! I will try this tomorrow or Friday (busy with work...). I will let you know!

lxne, 2020-07-01:
I had similar issues on mic RPI3 with this hat.
Found a strange solution: I removed the audio control from the taskbar and audio in GoogleAssistant began to work. Maybe you have to delete the .asoundrc file, too.

MichMich, 2020-07-04:
This is no MagicMirror² core issue. Feel free to discuss further on http://forum.magicmirror.builders 

Closing this issue.

closed 2020-07-04 by MichMich