MagicMirrorOrg/MagicMirror #2065: NumberofDay stuck on 2 after updating MagicMirror 2-7-20

BGH-H, 2020-07-02:
After running the update the weatherforecast app it seems to be stuck to 2 days. It ran on 5 or 7 days before. I have read this thread and double checked all tips mentioned here. Am I the only one who is now stuck on 2 days of forecast? 

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

kjendal, 2020-07-02:
I can confirm this issue as well.

AWSW2017, 2020-07-02:
I can confirm this after update on 3 different MM as well. The setting is in my case still at 5 days max, but shows 2 only.

Souln-Crypto, 2020-07-02:
I can confirm it too. I have the same issue after the update. The module shows only 2 days but in my settings are 7 days configured.

MichMich, 2020-07-02:
It might have something to do with https://github.com/MichMich/MagicMirror/pull/2035

@Ekristoffe do you have any idea?

Souln-Crypto, 2020-07-02:
I replaced the file weatherforecast.js from the older version, and now it works fine. I think there is a problem with the code. I have checked the code from #2035, but I can't find any problems there.

MichMich, 2020-07-02:
If anyone is able to send a PR, that would be awesome.

rejas, 2020-07-02:
Maaaybe I find time tonight to Look at it

khassel, 2020-07-02:
see last post [here](https://forum.magicmirror.builders/topic/13187/weatherforecast-showing-only-two-days)

khassel, 2020-07-02:
and now https://github.com/MichMich/MagicMirror/pull/2066

MichMich, 2020-07-03:
I merged a fix by @oemel09: https://github.com/MichMich/MagicMirror/pull/2066

Can you check to see if the develop branch fixes the issue?

```
git checkout develop
git pull
```

BGH-H, 2020-07-03:
I’ll try asap . Thank you for the prompt response .
Wat lul ik nou . Wat ik bedoel te zeggen is : Dankje wel ik ga er zsm mee aan de slag. 

Groet ,
Bernard 

Verstuurd vanaf mijn iPhone

DustinBryant, 2020-07-04:
@MichMich can confirm the develop branch fixes this issue.

MichMich, 2020-07-04:
Ok, closing this issue for now. Let me know if I need to reopen this.

closed 2020-07-04 by MichMich

AWSW2017, 2020-07-05:
Just did a complete update. Still 2 days shown only. Is the fix already available?

sdetweil, 2020-07-05:
@AWSW2017 you have to change to the develop branch
do
```
cd ~/MagicMirror
git checkout develop
```
then restart mm

AWSW2017, 2020-07-05:
Thanks for your reply. I have done this and it replies:

error: Path specification 'develop' does not fit to any git-known files

Thanks in advance

rejas, 2020-07-05:
You probably used Sams Installation Scripts?

AWSW2017, 2020-07-05:
Maybe... I don’t remember... The installation was done over a year ago...

rejas, 2020-07-05:
I think you dont have all branches locally therefor... Try out a "git fetch --all" and then do the checkout again...

AWSW2017, 2020-07-05:
Did this and it relies to receive from origins. The checkout afterwards tells the same error...
When the patch is brought to the master branch I can live with this so far then... I am still at master it says...

rejas, 2020-07-05:
Ok, missing one line :-) Try the solution form here: https://stackoverflow.com/a/39986737/2244646

```
git config remote.origin.fetch '+refs/heads/*:refs/remotes/origin/*'
git fetch --all
```

AWSW2017, 2020-07-05:
That helped. Thanks a lot!

BGH-H, 2020-07-06:
Ok Ive done the above , and the module works now, only I keep getting the update notification on my mirror after switching to:
git checkout develop
"already on develop" 
" your branch is up to date with origin/develop" 

Still the update notification says im 2 branches behind :( Sorry Im kinda nooby

MichMich, 2020-07-06:
Run `git pull` to update to the latest version of your current branch. 👍🏻

BGH-H, 2020-07-06:
did that ....and its says its already up to date :( thats why im so confused.

MichMich, 2020-07-06:
Did you restart mm?

BGH-H, 2020-07-06:
Yeah I did ...its so weird. Could it be a permissions thing ? Everything says its up to date (npm reinstall , git pulls , everything). Ive had some permission problems in the past. Maybe thats were the problem lies. For now im kinda stuck (is there any code to hide the update warning as a temp solution?

BGH-H, 2020-07-06:
Withdraw , issue closed. my bad