MagicMirrorOrg/MagicMirror #2068: All events in standard calendar module are shown in UTC following the update to MM 2.12.0

eliahru, 2020-07-03:
## I found a bug in MagicMirror

Following the update to 2.12.0 all Google Calendar events are shown incorrectly. 

**Platform**: Raspberry Pi 3

**Node Version**: 10.20.1

**MagicMirror Version**: 2.12.0

**Description**: 
I use Google calendars configured in MM standards "calendar" module via .ical URLs.
My local timezone is MSK (UTC+3, no daylight savings time).
Following the update to MM version 2.12.0, all events are shown as they start/finish 3 hours earlier. I.e. if I cave an event that starts at "15:00" local time, it is now shown in MM as starting at "12:00". It seems that UTC timezone is used instead of local one.

**Configuration**: 
Skipping this step for now as config.js file contains nothing related to time-zones.

MichMich, 2020-07-04:
It might have something to do with this PR: https://github.com/MichMich/MagicMirror/commit/6d60baa2d61deb43aed0769b0527b9475f725064#diff-4163d379a03c855907bc93633f16efa0

@rejas: Any idea if this could be the case?

eliahru, 2020-07-04:
Made some more tests. It looks that the problem is impacting only recurring events, one-time events are shown correctly.

MichMich, 2020-07-04:
https://github.com/MichMich/MagicMirror/pull/2057

MichMich, 2020-07-04:
Ok, I've merged #2057, could you check if this fixes the issue by running the develop branch?

```
git checkout develop && git pull
```

eliahru, 2020-07-06:
The given merge makes the situation for me event worse, the difference with real time becomes larger.
But it works fine if I remove the minus sign for the "offset" variable:
```
// correct data for the current timezone
var offset = -(new Date().getTimezoneOffset() * 60 * 1000);
           ^ remove this one
event.startDate = event.startDate - offset;
event.endDate = event.endDate - offset;
```

eliahru, 2020-07-06:
After applying #2071 everything seems to be working fine.

rejas, 2020-07-06:
Is there a test for this scenario in the code by the way?

mpegram3rd, 2020-07-07:
Just wanted to weigh in to confirm:
- It was happening with recurring events only
- The fix seems to have worked (pulled develop.. looking good). 

Thank you!  This is the most crucial thing on my MM at the moment since I use it to track work meetings.

MichMich, 2020-07-07:
Thanks for your confirmation. I'm closing this issue for now. Let me know if I need to reopen this issue.

closed 2020-07-07 by MichMich

sdetweil, 2020-07-08:
you should leave these fixed in develop issues open for people to search against, close them on actual release to master

khassel, 2020-07-08:
or think about bugfix releases. We will have discussions about this issue and weather bug from now on until next release (here, in the forum, in discord). And telling people to use `develop` branch or wait 3 months is no solution. Same thing was the socket.io bug in the release before. 

May it is a solution to make one bugfix release 2 weeks after main release (if there are such bugs).

sdetweil, 2020-07-08:
I agree, but Mich didn't to point releases, even tho there is numbering for them