MagicMirrorOrg/MagicMirror #2069: 2.12 user upgrade from 2.11, server mode fails

sdetweil, 2020-07-03:
npm run server
getting 
Cannot find module 'proxy-addr'
looks like from express

https://forum.magicmirror.builders/topic/10859/new-update-upgrade-script-ready-for-testing/285?_=1593780454858

cannot debug more, in hospital

MichMich, 2020-07-04:
No issues here. Did you try it with a default config?

MichMich, 2020-07-04:
More importantly, how are you doing Sam? All well?

khassel, 2020-07-04:
`proxy-addr` is used by `express` and `express-ipfilter`, but there are no version changes between MM 2.11 and 2.12

sdetweil, 2020-07-04:
I am doing pretty good. range of motion is better than expected for 5 days, so, I'll take it!!

thx for asking

GitHubbdream, 2020-07-07:
@MichMich `Did you try it with a default config?`
Just took the sample and a backed up one.
Quit interesting, as the whole MM seems to start now, but at the end I get always the same issue.
Not connecting web page.

@khassel what does it `proxy-addr is used by express and express-ipfilter` mean? Is there any I have to change?

@sdetweil furthe all the best wishes to you and your knee!

Best regards, bdream

MichMich, 2020-07-07:
@GitHubbdream how did you upgrade? What commands are you using to start the mirror?

GitHubbdream, 2020-07-07:
@MichMich I used Sams script `bash -c  "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/upgrade-script.sh)"` which enabled me to run MM on my RPi0 💯 
I changed package.json at "start" `"start": "./run-start.sh",`
and 
`pm2 flush
pm2 start 0`

Best regards, bdream

GitHubbdream, 2020-07-07:
May I install from scatch an new one?

sdetweil, 2020-07-07:
@GitHubbdream 

just rename the MagicMirror folder to something else, then use my install script

khassel, 2020-07-07:
i was only commenting the error message sam posted `Cannot find module 'proxy-addr'` which is strange because this dependency didn't change between MM 2.11 and 2.12

GitHubbdream, 2020-07-08:
@khassel thank you. that make sense.

@sdetweil just installed a new with your script. All fine but installing dependencies:
`Installing dependencies ...
npm WARN deprecated rrule-alt@2.2.8: Please use rrule instead
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated left-pad@1.3.0: use String.prototype.padStart()
npm WARN deprecated webdriverio@4.14.4: outdated version, please use @next
npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
npm WARN deprecated mkdirp@0.5.1: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated

Unable to install dependencies!
`
reboot starts MM into black screen with mousepointer, then raspian comes up, white screen with mousepointer, black screen....

sdetweil, 2020-07-08:
can u send me the ~/install.log?

sdetweil, 2020-07-08:
log says there is a requested reboot before the actual .mm install phase

GitHubbdream, 2020-07-08:
Ineresting.
Rebootet before starting the script anyway.
Just erased MagicMirror and rebooted again.
install-script started again and it works as expected.

Do I need to install all modules again or is copy from save-folder possible?

sdetweil, 2020-07-08:
u can copy modules from save folder. if they have a package.json file remember to remove the node_modules folder and package-lock.json files and run nom install after the copy

GitHubbdream, 2020-07-09:
@sdetweil huiii that was a long way
Brought all modules back with `remove the node_modules folder and package-lock.json files and run npm install after the copy`

had some issues don't know how to handle
`
> magicmirror@2.12.0 install /home/pi/MagicMirror
> echo "Installing vendor files ...
" && cd vendor && npm install --loglevel=error

Installing vendor files ...

audited 150 packages in 103.548s
found 8 vulnerabilities (7 low, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details

> magicmirror@2.12.0 postinstall /home/pi/MagicMirror
> npm run install-fonts && echo "MagicMirror installation finished successfully!"

> magicmirror@2.12.0 install-fonts /home/pi/MagicMirror
> echo "Installing fonts ...
" && cd fonts && npm install --loglevel=error

Installing fonts ...

audited 1 package in 18.481s
found 0 vulnerabilities

MagicMirror installation finished successfully!

audited 1042 packages in 421.863s

77 packages are looking for funding
  run `npm fund` for details

found 3 low severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details
`

I tried: 
`npm audit fix
up to date in 209.16s

77 packages are looking for funding
  run `npm fund` for details

fixed 0 of 3 vulnerabilities in 1040 scanned packages
  1 vulnerability required manual review and could not be updated
  2 package updates for 2 vulnerabilities involved breaking changes
  (use `npm audit fix --force` to install breaking changes; or refer to `npm audit` for steps to fix these manually)
`

Got the same as before :(

I copied over the original config,js and it works :D Don't know why it was not working in updated setup as MichMich mentioned to use original config.js....

Now I see the module MMM-RTSPStream fecing the issue.
Maybe that has to do with npm errors, maybe not. I'm not familair with this errors.

sdetweil, 2020-07-09:
@GitHubbdream this are not errors but warnings if you had a public website which could be attacked. even tho there are issues identified, there may not be fixes, and some fixes change code such that apps have to change ( called breaking changes). I generally recommend to ignore the audit stuff as you can't fix all of them

rtpstream, sorry no idea. seems finicky based on the recurring posts to the forum

MichMich, 2020-07-17:
So just to be sure, is this still a issue which can be replicated?

RaspberryTech01, 2020-07-18:
I had this issue today. I tried booting up the magicmirror software with npm run start but it spat out errors. I then tried a completely fresh install only server mode worked, no errors on normal mode except that no windows would pop up after the command was ran. I have a feeling its related to nodejs, ran sudo apt update and getting many dpkg errors with it.

sdetweil, 2020-07-18:
@RaspberryTech01   I am unsure what you said..

was MM running before?  or is this new install?
you don't need a new install to run server mode, just start a different pgm.. it has no UI.. YOU have to open a browser back to the 'server'

>no errors on normal mode except that no windows would pop up after the command was ran.
no idea what that means? in server mode? or full mode? 

did u use the automated install script? see the install command here 
https://github.com/sdetweil/MagicMirror_scripts

RaspberryTech01, 2020-07-18:
I used the manual install script. I fixed the problem by reinstalling the whole magicmirror software and nodejs. NodeJS has dpkg problems and I had to fix that.

MichMich, 2020-07-18:
Good to hear. Can I close this issue?

GitHubbdream, 2020-07-21:
I was off the last time ;)

@MichMich  asked: "So just to be sure, is this still a issue which can be replicated?"

I was not able to find out the reason.
I run in problems after I activated the MMM-RTSPStream Module. from the backup.

I intalled an configured MMM-RTSPStream again and it works.

As @sdetweil mentioned I had not errors but warnings while npm install again.

So, from my side nothing more to do. I can't replicate the issue. But all ist running as estimated.

Many thansk to all!

sdetweil, 2020-08-15:
need to review apt-get upgrade for required reboot before continuing.

closed 2020-08-15 by sdetweil