MagicMirrorOrg/MagicMirror #2074: weatherforecast only showing 2 days, not the default 7.

hervidero, 2020-07-05:
## I found a bug in MagicMirror

Default module weatherforecast started showing only a 2 day forecast recently instead of the default 7 day forecast.
I think openweathermap.org changed something, they now have 8 forecast moments per day.
As a result the mirror only shows 2 days of forecast with a default maxNumberOfDays: 7 in the config.js

If I change the weatherforecast.js so that &cnt=56 the mirror again shows a 7 day forecast. (as 8 moments * 7 days = 56)

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.
2.12.0

khassel, 2020-07-05:
This is may a duplicate of https://github.com/MichMich/MagicMirror/issues/2065 which is already fixed on the `develop` branch.

hervidero, 2020-07-05:
After reading #2065, I can confirm it is the same issue.

MichMich, 2020-07-05:
Duplicate. Closing issue.

closed 2020-07-05 by MichMich