MagicMirrorOrg/MagicMirror #2076: OpenWeather onecall API endpoint support

cjbrunner, 2020-07-08:
## I'm not sure if this is a bug

Not sure if it's a bug, per se, but the fallback endpoint the `weatherforecast` module uses when free accounts fail to access the `forecast/daily` endpoint doesn't appear to work anymore.  The fallback endpoint (simply `forecast`) returns data in 3-hour increments, which the `weatherforecast` module does not currently seem to handle correctly.

Rather than try to make the `forecast` API work, OpenWeather seem to be pushing their [onecall](https://openweathermap.org/api/one-call-api) API endpoint for free users, so I wrote up a little PR for MagicMirror to handle that, mostly so my own setup would work again, but if it's useful I can make a PR here.

oemel09, 2020-07-08:
The bug with the the `forecast` API should be fixed in the latest develop branch already.

But I think using the new onecall API seems like a good idea, I'd prefer to use it because it features a 7 day forecast instead of only 5 days.
From my side: Feel free to post a PR with your implementation

MichMich, 2020-07-17:
Merged.

closed 2020-07-17 by MichMich

hangorazvan, 2020-09-05:
can someone help me please to add dewpoint, humidity, pressure and uvi from onecall api in the second row of each day?