MagicMirrorOrg/MagicMirror #2085: Calendar event issues with DST and organizer timezone being different

chamakura, 2020-07-12:
**Node Version**: 10.19.0

**MagicMirror Version**: 2.13

**Description**: There's two issues with the way the Calendar module displays events:
1. Ignores organizers timezone: For events that are scheduled by someone who is in a different timezone, the 'ical' parser incorrectly parses the start time as being in the local timezone. This results in all of the local times being offset by the difference between the timezones of the organizer and the user. It's not possible to easily fix the issue in 'ical' (I tried https://github.com/peterbraden/ical.js/pull/111), so the solution here is to instead use the 'node-ical' library which is a fork of 'ical' to better support node installations. The parsing issue is already fixed in 'node-ical'.
2. The parsing of times/events from Calendar are all done in UTC, and don't consider Daylight Savings Time (DST) changes. For recurring events, what this means is that if the recurrence spans across the point when there's a DST change, all of the subsequent times will be off (because UTC does not have any DST changes). The fix here is to apply the necessary DST changes whenever either the organizer or the user has undergone a DST transition.

**Steps to Reproduce**: 
1. Create a Calendar event in a different timezone than yours
2. Create a recurring event that spans a DST change (either starts during DST and extends to no-DST or the other way around)

**Expected Results**: The event times displayed in Calendar should be aligned with the local time by factoring in DST changes.

**Actual Results**: Event times are incorrect.

**Configuration**: My Calendar config:
```
{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
                        url: "https://calendar.google.com/calendar/ical/tvejbv8vv27a16v76hmsgr4tbo%40group.calendar.google.com/private-d7185592c52d177beb17ad2e915e0fdc/basic.ics"
                    }
				],
                maximumEntries: 25,
                fade: false,
                timeFormat: "absolute",
                getRelative: 0,
                dateFormat: "MM/DD HH:mm:ss",
                urgency: 0,
			}
		}
```

**Additional Notes**: This issue has been discussed exhaustively in https://github.com/MichMich/MagicMirror/issues/1798, but there was no solution that addressed all of the problems.

I will be submitting a pull request with the fix.

kjendal, 2020-07-13:
I believe there are two parts to this problem (at least there were for me).
1) ical.js doesn't really handle timezones.
2) the timezone format may not be support by ical.js (none of the microsoft timezones are).
See [MMM-CalendarWeek/Issues/42](https://github.com/heskja/MMM-CalendarWeek/issues/42) for a description of the fix, I suspect it applies here as well.

sdetweil, 2020-09-22:
this issue can be closed

closed 2020-09-22 by MichMich