MagicMirrorOrg/MagicMirror #2087: Manual dependency installation instructions outputs error.

stopspazzing, 2020-07-13:
On Fresh install on Raspberry Pi Zero W with Raspberry Pi OS (Full)

`curl -sL https://deb.nodesource.com/setup_10.x | sudo -E bash -
`

```
## Installing the NodeSource Node.js 10.x repo...

## You appear to be running on ARMv6 hardware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js 4 and later.
```

https://docs.magicmirror.builders/getting-started/installation.html#manual-installation

sdetweil, 2020-07-13:
yes, you need to use my scripted install which configures the system in split mode, server with chromium  over it. 

see the installation script here 
https://github.com/sdetweil/MagicMirror_scripts

khassel, 2020-07-13:
you can use sam's script for this as he already mentioned above. The [official version from the docs](https://docs.magicmirror.builders/getting-started/requirements.html#hardware): pi 0/1 are not supported.

closed 2020-07-15 by stopspazzing