MagicMirrorOrg/MagicMirror #2088: i have done correctly stepwise.... :(  Module not found

abhishekpreet, 2020-07-13:
pi@raspberrypi:~/MagicMirror $ npm run restart

> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-07-14 00:48:22.952] [LOG]    Starting MagicMirror: v2.12.0
[2020-07-14 00:48:22.959] [LOG]    Loading config ...
[2020-07-14 00:48:22.969] [LOG]    Loading module helpers ...
[2020-07-14 00:48:22.971] [LOG]    No helper found for module: alert.
[2020-07-14 00:48:23.270] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: Failed to load gRPC binary module because it was not installed for the current system
Expected directory: electron-v6.1-linux-arm-glibc
Found: [node-v64-linux-arm-glibc]
This problem can often be fixed by running "npm rebuild" on the current system
Original error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/electron-v6.1-linux-arm-glibc/grpc_node.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client_interceptors.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[2020-07-14 00:48:23.272] [LOG]    Loading module helpers ...
[2020-07-14 00:48:23.273] [LOG]    No helper found for module: alert.
[2020-07-14 00:48:23.289] [ERROR]  App threw an error during load
[2020-07-14 00:48:23.290] [ERROR]  Error: Failed to load gRPC binary module because it was not installed for the current system
Expected directory: electron-v6.1-linux-arm-glibc
Found: [node-v64-linux-arm-glibc]
This problem can often be fixed by running "npm rebuild" on the current system
Original error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/electron-v6.1-linux-arm-glibc/grpc_node.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client_interceptors.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js:53:17)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client_interceptors.js:144:12)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
[2020-07-14 00:48:23.292] [ERROR]  Whoops! There was an uncaught exception...
[2020-07-14 00:48:23.292] [ERROR]  Error: Failed to load gRPC binary module because it was not installed for the current system
Expected directory: electron-v6.1-linux-arm-glibc
Found: [node-v64-linux-arm-glibc]
This problem can often be fixed by running "npm rebuild" on the current system
Original error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/electron-v6.1-linux-arm-glibc/grpc_node.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client_interceptors.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js:53:17)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client_interceptors.js:144:12)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10) {
  code: 'MODULE_NOT_FOUND'
}
[2020-07-14 00:48:23.298] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-07-14 00:48:23.299] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-07-14 00:48:23.388] [LOG]    Launching application.

sdetweil, 2020-07-13:
I suggest you post the issue to the MMM-GoogleAssistant repo , this is not an MM problem
@bugsounet  will help you
https://github.com/bugsounet/MMM-GoogleAssistant

MichMich, 2020-07-17:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2020-07-17 by MichMich

abhishekpreet, 2020-07-17:
Now it's okay....

bugsounet, 2020-07-17:
wonderfull so ;)

abhishekpreet, 2020-07-17:
I had downloaded this again after erasing....