MagicMirrorOrg/MagicMirror #2090: MagicMirror refresh modules on "focus" event

sthuber90, 2020-07-15:
Can I somehow refresh MagicMirror and the running modules on an event, such as the focus event in a browser window? 

Given that not all mirror are always on and running, I want the view on my screen to be refreshed on "wake-up". This wake-up could be a manual push of a bower button of the screen or a movement detected by a camera which wakes up the screen from its power save sleep.

sdetweil, 2020-07-15:
if the mirror is self hidden or the screen is turned off, the page will get a refresh signal on screen on or unhide. modules are already supposed to be aware of the functions to handle this.

you could program something extra to send a ctrl-r keystroke to the mm window to cause a reload

sthuber90, 2020-07-16:
Thank you for the prompt response. Is this also true on screens connected to MagicMirror, running in server-only mode? I couldn't reproduce this, when running MM on my PC. Turning off and on my external monitor did not trigger a refresh as far as I can tell from logs and what's shown on screen.

sdetweil, 2020-07-16:
that is Software driven turn off.. using the dpms xset command or the tvservice command.. 

if you just flick the external switch, then there is no notification to the software.. 

one could create some hardware that detects power flow and toggle one of the gpio pins that some software could monitor to cause that event.

closed 2020-07-17 by MichMich