MagicMirrorOrg/MagicMirror #2093: forum error 503 excessive load

sdetweil, 2020-07-19:
## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

MichMich, 2020-07-19:
Fixed.

sdetweil, 2020-07-19:
still same results for me

MichMich, 2020-07-19:
Are you sure you're not just refreshing the 503 page (see url)

sdetweil, 2020-07-19:
entered magicmirror.builders, then selected forum, got 503

MichMich, 2020-07-19:
Could you try cleaning your cache or refresh. No signs of being down here. There are multiple users online at the moment.

sdetweil, 2020-07-19:
I see it from two different systems. same after clearing cache, both systems

sdetweil, 2020-07-19:
disabled wifi on phone, same results

khassel, 2020-07-19:
forum is working for me

sdetweil, 2020-07-19:
still not for me. now three systems. phone, laptop and desktop

sdetweil, 2020-07-19:
power off/on phone clear cache. no change.

try to connect, long delay, error 503

MichMich, 2020-07-19:
Seems you are cursed. 🥴

khassel, 2020-07-19:
strange ... had similar problems only with some websites a few days ago because of this issue: https://blog.cloudflare.com/cloudflare-outage-on-july-17-2020/

MichMich, 2020-07-19:
What happens if you try incognito mode?

sdetweil, 2020-07-19:
incognito, same. long delay, 503

sdetweil, 2020-07-19:
back

closed 2020-07-19 by sdetweil