MagicMirrorOrg/MagicMirror #2094: weatherforecast module is not showing the correct amount of days

blackthroat, 2020-07-19:
Possibly related to https://github.com/MichMich/MagicMirror/pull/2066. This worked fine for me prior to upgrading to 2.12.0

**Platform**: Raspberry Pi 2/3

**Node Version**: 10.21.0

**MagicMirror Version**: 2.12.0

**Description**: weatherforecast module is not showing the correct amount of days
[image]

**Steps to Reproduce**:  Modify values for `maxNumberOfDays` in config.js for weatherforecast module

**Expected Results**: Expected to see between 1 and 16 days.

**Actual Results**: Only maximum of 2 or 3 days are shown

**Configuration**: 
```
module: "weatherforecast",
position: "top_right",                                                                                                                                                                config: {
	locationID: "[REDACTED]",  //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
	appid: "[REDACTED]",
	roundTemp: true,
	fade: false,
	appendLocationNameToHeader: false,
	}     
```
**Additional Notes**: 
Leaving `maxNumberOfDays` out of config.js will result in only 2 days being shown, while setting the value to 7 will also show 2 days. Setting the value to 16 will show 3 days.

hangorazvan, 2020-07-19:
if you mentioned #2066 at least did you tried to solve with https://github.com/MichMich/MagicMirror/commit/37237d9c10a0698fe61a3084a6467b7d5f213d64 ?

in weatherforecast replace line 297 with this:
```
		let numberOfDays;
		if (this.config.forecastEndpoint === "forecast") {
			numberOfDays = this.config.maxNumberOfDays < 1 || this.config.maxNumberOfDays > 5 ? 5 : this.config.maxNumberOfDays;
			// don't get forecasts for the 6th day, as it would not represent the whole day
			numberOfDays = numberOfDays * 8 - (Math.floor(new Date().getHours() / 3) % 8);
		} else {
			numberOfDays = this.config.maxNumberOfDays < 1 || this.config.maxNumberOfDays > 17 ? 7 : this.config.maxNumberOfDays;
		}
		params += "&cnt=" + numberOfDays;
```

blackthroat, 2020-07-19:
Thanks, that did the trick. I didn't look close enough at the 2.12.0 release date and thought the PR made it into the latest release.

closed 2020-07-19 by blackthroat