MagicMirrorOrg/MagicMirror #2097: Calendar iCal RRule Results in Incorrect Time Offset

ykuzmycz, 2020-07-21:
## I found a bug in MagicMirror

**Platform**: Rasperry PI

**MagicMirror Version**: 2.12

**Description**: Calendar parses iCal file (ics) incorrectly when using a reoccurring event.  Confirmed that timezone of raspberry pi and ics event are the same (America/Toronto).

**Steps to Reproduce**: Created a daily repeating event starting at 9am.  Calendar shows daily correctly but starts at 1pm.  Created a single event starting at 9am.  Event shows correctly.  Added "RRULE:FREQ=DAILY;UNTIL=20200801T035959Z" to the newly created event without changing anything else it correctly shows as repeating but time start changes to 1pm.

BEGIN:VEVENT
CREATED:20200721T010209Z
DTEND;TZID=America/Toronto:20200727T100000
DTSTAMP:20200721T010209Z
DTSTART;TZID=America/Toronto:20200727T090000
LAST-MODIFIED:20200721T010209Z
SEQUENCE:0
SUMMARY:Test
UID:B54FA134-516B-4686-8A3F-744653968A44
URL;VALUE=URI:
TRANSP:OPAQUE
BEGIN:VALARM
ACTION:DISPLAY
DESCRIPTION:Reminder
TRIGGER:-PT1H
UID:103B56E0-48C5-4A7C-9471-4FE8CD989AF0
X-APPLE-DEFAULT-ALARM:TRUE
X-WR-ALARMUID:103B56E0-48C5-4A7C-9471-4FE8CD989AF0
END:VALARM
END:VEVENT

sdetweil, 2020-07-21:
what app did you use to create the cal entry? pi doesn't create, only view.
4 hour diff 9-1, sounds like app is not timezone sensitive. but created entry in UTC.

ykuzmycz, 2020-07-21:
Events are created via Calendar app on iPhone (iCloud).  Vdirsyncer then syncs the iCloud ics locally and the MagicMirror Calendar module is pointed to the local ics file.  Confirmed that the iCloud event is created with a Toronto timezone.  The ics extract in the original entry shows the event is also in Toronto timezone.  That was my first thought but I confirmed all parts of the flow are on the same timezone.

ykuzmycz, 2020-07-21:
I manually edited the local ics file and edited the “test” event. Without the daily rule it shows up fine. Manually adding the rrule to daily creates the issue. This shows that it’s not the creator of the ics but the reader I would think.

rejas, 2020-07-21:
Could you provide that example ics File?

sdetweil, 2020-07-21:
@ykuzmycz I didn't see the tz value in the cal entry. having the 'right' hour value is not enough.
we have all these crazy problems

tz where the app was, tz where the calendar server is ??, tz where the mirror is. then add on daylight adjustments when two ends of a connection don't change time together. and a recurring entry was created before daylight time, and  the next instance is after

ykuzmycz, 2020-07-24:
@sdetweil sorry, didn't follow your comment.  All these entries were made recently so wouldn't expect daylight time to be an impact.

@rejas  I've attached my ics file as a txt file (tried to scrub it of personal information where I could).

It seems something to do with if an event is reoccurring.  If it's not, it's fine.  My "test" event test indicated that.

[home.txt](https://github.com/MichMich/MagicMirror/files/4969645/home.txt)

sdetweil, 2020-07-24:
@ykuzmycz I was just commenting that getting the cal entry right is a challenge, with a lot of moving parts.  we had a private copy of the iCal lib, cause the public one was unstable at some time. now we have moved up to the public one again, as the private version doesn't handle a bunch of features showing up in our users calendars. and it looks like that public iCal. lib is still trouble

ykuzmycz, 2020-07-24:
@sdetweil ah.  Understood, thanks.  Happy to shift to a private lib to test if it helps.  If memory serves, I didn't have issues with my data flow in previous versions.

rejas, 2020-07-25:
@ykuzmycz will try to see if I can write a test against your ics file. with a test we can see if and what solution would fix that while not breaking other stuff :-)

just a question: the "golf camp" entries in there are problematic, correct?

ykuzmycz, 2020-07-25:
@rejas Thanks!  Holler if I can help in any way.  Happy to create further test cases on my end locally.

Correct.  The "golf camp" entries are problematic.  Interesting point - when I took the "test" entry and made it repetitive across the week (rrule) it acted the same way as the "golf camp" (wrong start time).  When I removed the rrule on the "test" entry it went back to starting at 9am.

rejas, 2020-07-26:
Thx for the clarification. I also noticed on my personal mirror that the birthdays (daylong recurring yearly events) were off by one day (so tomorrows birthday was shown as happening today)
That bug was in master and got fixed in develop by commit 862bf78e6337bbc2e70d5f7a37c4104b7fb23b53
Could you check out the develop branch and see if your issue is in there still? Maybe that fix is also good for your bug.

ykuzmycz, 2020-07-26:
@rejas no worries.  Happy to help.
Assume you just need me to pull from modules/default/calendar from the develop branch?

sdetweil, 2020-07-26:
@ykuzmycz just use the git checkout command to checkout the develop branch. no copying or anything. 

if u used my script to install, I only load the master branch 1 iteration back, to minimize space usage. so to get the develop branch to the pi do
```
cd ~/MagicMirror
git fetch origin develop:develop
```
then checkout in either case
```
git checkout develop
```
and restart mm

ykuzmycz, 2020-07-26:
@sdetweil I did use your script to update, thanks!

Will kick off the checkout and keep the group posted of results.

ykuzmycz, 2020-07-26:
Success!  Shifted to develop branch and restarted using the commands provided.  The "golf camp" events now start at 9am correctly.  Happy to place a few other test events to test any further cases as needed.

FYI: I had to do a git reset to clear the below

error: Your local changes to the following files would be overwritten by checkout:
	package-lock.json
	vendor/package-lock.json
Please commit your changes or stash them before you switch branches.
Aborting

ykuzmycz, 2020-07-26:
@sdetweil I assume I'll have to shift to master branch when changes are merged to master and we update to a version.  Will I have to do anything other than execute your normal update script (I expect I'll have to fetch origin master prior)?

sdetweil, 2020-07-26:
the update script will handle it all. and tell u what u need to do (save files), if anything

ykuzmycz, 2020-07-26:
Thanks.  Seems like we can close this one then if https://github.com/MichMich/MagicMirror/commit/862bf78e6337bbc2e70d5f7a37c4104b7fb23b53 sorts it out.

rejas, 2020-07-27:
@sdetweil regarding those changes in the lock files after your script runs: do you do a "npm i" or "npm ci" ? The later doesnt change the lock files and would be more appropiate: see https://blog.npmjs.org/post/171556855892/introducing-npm-ci-for-faster-more-reliable to quote: 
`npm ci bypasses a package’s package.json to install modules from a package’s lockfile. This ensures reproducible builds—you are getting exactly what you expect on every install.`

@MichMich with those calendar and other bugs popping up again and again, maybe it makes sense to release a little earlier or do a bugfix release? what are your thoughts on that?

sdetweil, 2020-07-27:
I erase the lock files and run npm i , same as what you would do for manual update.

the files changed, and u can restore them, but they will be changed, or erase them and they will be rebuilt

sdetweil, 2020-07-27:
as I support more platforms than the base repo, I don't use ci.

khassel, 2020-07-27:
> @MichMich with those calendar and other bugs popping up again and again, maybe it makes sense to release a little earlier or do a bugfix release? what are your thoughts on that?

concerning this I added already my thoughts here: https://github.com/MichMich/MagicMirror/issues/2068#issuecomment-655681811

MichMich, 2020-07-28:
> @MichMich with those calendar and other bugs popping up again and again, maybe it makes sense to release a little earlier or do a bugfix release? what are your thoughts on that?

The thing is that I'm 100% confident that new bugs will arise after that release, meaning I would need to do an other release (and so forth). For a lot of users a new release is annoying (due to the update notification) and unnecessary: most users don't encounter these issues. 

For the users that do encounter this issue, the solution is simple: use the develop branch. As an added bonus this means we have extra `develop`-testers. (Which we really need!) 

Usually I only release X.X.1 updates whenever there really is a blocking issue most users encounter (ie. MM won't boot). For all other issues, I like to refer people to the `develop` branch.

rejas, 2020-07-28:
> As an added bonus this means we have extra `develop`-testers. (Which we really need!)

I won't argue with you about that reasoning :-) +1

sdetweil, 2020-09-22:
this issue can be closed

closed 2020-09-22 by MichMich

ykuzmycz, 2020-11-01:
@sdetweil this one popped up again.  I'm on the latest version of MM.  Different calendar events but everything is one hour behind.  I executed the below to jump on the development branch (I like the idea of more users on develop = more testers).  Happy to help in whatever way necessary.

I'm using your script to install.  Executed the below:

git fetch origin develop:develop

git checkout develop  (had to do a git reset --hard first as there were some locks)

pm2 restart mm

Outside of executing the above, how can I ensure it's using the development branch (ie pulled down the necessary updates)?  Calendar entries still one hour behind.  Holler if ics file needed and happy to create test entries.

sdetweil, 2020-11-01:
@ykuzmycz 

can u post one VENET from your ICS, or send to me , same userid at gmail

thanks..

are these full day repeating ? (more fixes coming)

ykuzmycz, 2020-11-01:
Full ics attached.  They're weekly repeating events at a certain time.

Happy to create sample ones to cover any use cases needed.

[icsfile.txt](https://github.com/MichMich/MagicMirror/files/5470981/icsfile.txt)

sdetweil, 2020-11-01:
@ykuzmycz save the ~/MagicMirror/modules/default/calendar/calendarfecther.js
and install
[calendarfetcher.js.txt](https://github.com/MichMich/MagicMirror/files/5471073/calendarfetcher.js.txt)
 this one

remove the .txt extension of course

ykuzmycz, 2020-11-01:
Done.  Times moved but oddly in the wrong direction it seems.

With new js
Piano Lessons 5pm
Dancing 4pm
School 7:30am

With old js
Piano Lessons 6pm
Dancing 5pm
School 8:30am

Should be
Piano 7pm
Dancing 6pm
School 9:30am

ykuzmycz, 2020-11-01:
Trying again, I might have missed the updated file.

ykuzmycz, 2020-11-01:
version 2 did it.  Times are now correct.

Holler if there are other use cases that might pop up that you need me to create - happy to help test.

sdetweil, 2020-11-01:
@ykuzmycz  awesome.. thanks for the fast test..I also saw the problem om my own calendars (but hadn't brought up MM today yet)

ykuzmycz, 2020-11-01:
@sdetweil my pleasure - always happy to test the calendar stuff.  My setup is a bit unique (iPhone -> vdirsync -> pi) so I appreciate it may need extra testing.  Ping me if there's further stuff that needs testing (I'm happy to create dummy entries on my side if it helps).

Huge thanks for the quick help!

(I'll stay on the dev branch)

ykuzmycz, 2021-03-05:
> @ykuzmycz just use the git checkout command to checkout the develop branch. no copying or anything.
>
> if u used my script to install, I only load the master branch 1 iteration back, to minimize space [...]

> @ykuzmycz just use the git checkout command to checkout the develop branch. no copying or anything.
>
> if u used my script to install, I only load the master branch 1 iteration back, to minimize space [...]

@sdetweil - apologies for going into an old thread with a quick "how-to".  Just trying to troubleshoot a few things before raising a bug.  I'm using your install script and pointing to the development branch.  How do I correctly shift to the master branch to check if the issue has to do with dev branch?  Basically kick off the above but "point back" to master.  Thanks (and again sorry about going into an old thread).

sdetweil, 2021-03-06:
git checkout master
npm install

ykuzmycz, 2021-03-06:
Much appreciated!  Going back to master broke the calendar again (and didn't solve my module issue) so I reverted to dev with an updated calendarfetcher.  Shrug, seems I know what I'm doing this weekend!

sdetweil, 2021-03-06:
@ykuzmycz   >and didn't solve my module issue

what module problem is that?

note that develop NOW is 2.14 + 
a new electron, 
a rewrite of calendarfetcher
a rewrite of weather & currentweather
and a whole lot more...

ykuzmycz, 2021-03-06:
MMM-Unsplash.  Totally stopped working this week so I'm slowly going through the "what could have changed" options as well as the logs.

Heads up, I repointed to dev first without new calendarfetcher this morning.  It did not pick up my first calendar entry.  I had to rewrite with your version to get the first entry.

sdetweil, 2021-03-06:
after switch branches you MUST npm install again.. right?  maybe erase package-lock.json too before

ykuzmycz, 2021-03-06:
Yup, I did exactly what you commented:

git checkout master
npm install

I very much had to grab the old calendarfetcher when shifting from master back to dev.  Shrug.  I did have to execute git checkout master -f though.

sdetweil, 2021-03-06:
hm.. i don't see an apikey with unsplash

ykuzmycz, 2021-03-06:
I've got one from ages ago.  It has to be something silly - was working on my mm just this week.  Everything else working except the background.  So time to step through it.  May have to go into the "webpage" itself remotely and inspect each div.  Groan.