MagicMirrorOrg/MagicMirror #2099: excessive load error , 503, again

sdetweil, 2020-07-26:
## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

MichMich, 2020-07-26:
That's correct. Every Sunday around this time the server sends the weekly forum updates to the users who have this enabled. 

It will take a few minutes and during that time the server could suffer from heavy load. 

It should solve itself after a while.

sdetweil, 2020-07-26:
should move those to some low usage time of day. after midnight maybe

MichMich, 2020-07-28:
That depends on the time zone where you are ... ;)

sdetweil, 2020-07-28:
well, find the typical idle time and do it then. West coast USA, after midnight, would seem like a smaller population impacted. I need to go configure out of those. totally useless given my online time

MichMich, 2020-07-28:
Made some changes. Let's see how it goes next week. Closing this for now. Let me know if you run into issues.

closed 2020-07-28 by MichMich