MagicMirrorOrg/MagicMirror #2102: How does one move the bottom bar to the right?

yurMoThEr, 2020-07-31:
This isn't an issue, but I have a module pretty near the bottom bar and was wondering how I could move it for the sake of aesthetics. Thanks for any help.

sdetweil, 2020-07-31:
you could use css in the custom.css file to change the size, location of the bottom bar.

if css/custom.css doesn't exist, just create it
```
touch ~/MagicMirror/css/custom.css
```

yurMoThEr, 2020-08-01:
I hate to ask you this, but I know nothing of CSS. Could you tell me how to do it?

sdetweil, 2020-08-01:
read this
https://forum.magicmirror.builders/topic/6808/css-101-getting-started-with-css-and-understanding-how-css-works?_=1596288932213

I am not looking to make changes for other users.

closed 2020-08-03 by yurMoThEr