MagicMirrorOrg/MagicMirror #2106: MM2 on half the screen?

ksdehoff, 2020-08-02:
This is not a bug.

I'd like MM2 to coexist with another home control App I use - hstouch from homeseer.  I'd like to have HSTouch on the left 25% of the display and MM2 on the remaining 75%.  I'm using an infrared touch frame so I can select the HSTouch buttons and currently if I 'click' on MM2 its window pops over and hides HSTouch.  I'd like to get these two apps to coexist

sdetweil, 2020-08-07:
use the electronOptions in config.js to set the size and placement of the MM window

while not specifically about your request, the description and link to doc are in this forum topic
https://forum.magicmirror.builders/topic/13431/magic-mirror-with-2-displays/2

ksdehoff, 2020-08-12:
Thanks much – this works great.  Unfortunately the RTSP module apparently expects a fullscreen  so the video window is in the wrong place.  I’ll ping them to try to find a workaround

From: sam detweiler
Sent: Friday, August 7, 2020 3:13 PM
To: MichMich/MagicMirror
Cc: ksdehoff; Author
Subject: Re: [MichMich/MagicMirror] MM2 on half the screen? (#2106)

use the electronOptions in config.js to set the size and placement of the MM window
while not specifically about your request, the description and link to doc are in this forum topic
https://forum.magicmirror.builders/topic/13431/magic-mirror-with-2-displays/2

closed 2020-09-18 by MichMich