MagicMirrorOrg/MagicMirror #2109: config:check dependencies listed in devdependencies

sdetweil, 2020-08-06:
to reduce space on SD card, my installer  does  npm install only=prod, which leaves off devDependencies, and all the prettier, etc stuff is listed there. needs to be moved to the production dependencies. 

npm run config:check fails missing prettier etc

khassel, 2020-08-06:
same setup, and after running `npm install`, which installs devDependencies, still the same error:

````
> magicmirror@2.13.0-develop config:check /opt/magic_mirror
> node js/check_config.js

[2020-08-06 22:52:57.222] [INFO]   Checking file...  /opt/magic_mirror/config/config.js
[2020-08-06 22:52:57.267] [LOG]    Line 1 col 1 Definition for rule 'prettier/prettier' was not found.
/opt/magic_mirror/js/check_config.js:66
                        throw new Error("Wrong syntax in config file!");
                        ^

Error: Wrong syntax in config file!
    at /opt/magic_mirror/js/check_config.js:66:10
````

The syntax of `config.js` is correct.

Could be related to prettier stuff, so may @rejas can take a look at this?

rejas, 2020-08-07:
Took a look, made a PR :-)

sdetweil, 2020-08-07:
see note on pr

sdetweil, 2020-09-07:
looks good

PR #2112 (2020-08-13): Fix config check after merge of prettier
The verifying of the config file doesnt need the eslint config file to run. SO remove it and jsut check the syntax and not the format. Fixes #2109
closed 2020-09-07 by sdetweil