MagicMirrorOrg/MagicMirror #2110: Calendar not showing reoccurring full-day events on the day of the event (static calendar file)

Feedy88, 2020-08-06:
_Hey there, first of all, thanks for this amazing project. I started setting up a few weeks ago and enjoy every time working on it._

**Platform**: Raspberry Pi 4, 2 GB, Raspbian 10 (buster); Electron 6.1.7

**Node Version**: 10.21.0

**MagicMirror Version**: 2.13

**Description**: For birthdays I am using a static calendar file saved in a folder using `https://localhost:8080/PATHNAME/calendar.ics` to feed it to the mirror. Before the actual date of the event the Birthdays are displayed correctly. On the actual day of the event, the events don't show up in the mirror. This goes only for **reoccurring full day events.** Full day events without reoccurance and reoccurring events which are not full day work fine.

**Steps to Reproduce**:

- Create and ics file with the following format:
```
BEGIN:VCALENDAR
VERSION:2.0
PRODID:-//Static Birthdays//Rafal Rüßmann//EN
CALSCALE:GREGORIAN
X-WR-CALNAME;VALUE=TEXT:Birthdays_static_short
X-WR-TIMEZONE:Europe/Berlin
BEGIN:VTIMEZONE
TZID:Europe/Berlin
LAST-MODIFIED:20200806T101500Z
BEGIN:STANDARD
DTSTART:20191027T030000
RRULE:FREQ=YEARLY;BYMONTH=10;BYDAY=-1SU
TZNAME:CET
TZOFFSETFROM:+0200
TZOFFSETTO:+0100
END:STANDARD
BEGIN:DAYLIGHT
DTSTART:20200329T020000
RRULE:FREQ=YEARLY;BYMONTH=3;BYDAY=-1SU
TZNAME:CEST
TZOFFSETFROM:+0100
TZOFFSETTO:+0200
END:DAYLIGHT
END:VTIMEZONE
BEGIN:VEVENT
SUMMARY:Max Mustermann
DESCRIPTION:
DTSTAMP:20050806T080000Z
DTSTART;VALUE=DATE:20050806
DTEND;VALUE=DATE:20050806
RRULE:FREQ=YEARLY
END:VEVENT
END:VCALENDAR
```
- Save calendar locally on MagicMirror
- Feed calendar to via `localhost`

**Expected Results**: Seeing the full day even throughout the day of the event in the calendar.

**Actual Results**: Event only showing a day before but not on the actual day.

**Configuration**: 

```javascript
{
    module: "calendar",
    header: "Termine & Feiertage",
    position: "top_left",
    config: {
        fetchInterval: 60000,
        displaySymbol: true,
        showLocation: false,
        fade: true,
        fadePoint: 0.4,
        dateFormat: "D. MMM HH:mm",
        fullDayEventDateFormat: "D. MMM",
        timeFormat: "absolute",
        getRelative: 12,
        urgency: 0,
        maximumEntries: 5,
        calendars: [
            {
                symbol: "cocktail",
                url: "webcal://www.ecoline-media.de/ics/nordrhein-westfalen.ics"
            },
            {
                symbol: 'calendar-check',
                url: 'webcal://localhost:8080/modules/calendars/9*******-****-****-****-***********D.ics',
            },
            {
                symbol: 'birthday-cake',
                url: 'webcal://localhost:8080/modules/calendars/birthdays.ics',
            }
        ]
    }
},
```

**Additional Notes**: I already discussed this issue in the forum, recommendation was to craete a ticket, as config of MM and ics look fine.

castletonroad, 2020-08-07:
Hi, I think my issue #2111 maybe the same, albeit manifesting itself via a different route.

rejas, 2020-08-30:
The PR that moved ical to v0.8.0 didnt revert the #1798 fix. That came later with #2071.

hangorazvan, 2020-08-30:
Ok, but #1798 fix was good at the time, can be done again?

hangorazvan, 2020-09-01:
https://github.com/MichMich/MagicMirror/pull/2121 I think this solve the problem

Feedy88, 2020-09-02:
Just checked, #2121 fixed the issue. Closing this one. Thanks a lot to all participants

closed 2020-09-02 by Feedy88