MagicMirrorOrg/MagicMirror #2111: On the day of event, Calendar does not show recurring events starting within TZ difference from midnight

castletonroad, 2020-08-07:
**Platform**: Raspberry Pi 4, 2 GB, Raspbian 10 (buster); Electron 6.1.12

**Node Version**: 10.22.0

**MagicMirror Version**: 2.13.0-develop

**Description**: On the day of an event, Calendar module does not show any recurring events where the start time is BEFORE 10am - these events start within the TZ difference (10hrs) from midnight. (My TZ is AEST, currently GMT+10.00). If the event is NOT recurring, it will show on the MM. If the event IS recurring, but starts after 10.00, it will also show on the MM

**Steps to Reproduce**:
1. Create a new recurring (two-weekly, 4 occurrences, 09.20-10.15) event (via either of iPhone Calendar app, or Nextcloud web-based Calendar app) for next day - recurring event shows correctly on the MM ('Tomorrow at 09.20').
2. On the day of the event, the recurring event starting before 10.00 does not show; any recurring event starting after 10.00 shows correctly.
3. Create a NON-recurring event for same day (09.20-10.15) and restart the MM - the event shows correctly on the MM.
4. Make this event RECURRING and restart the MM - the event DOES NOT show.
5. Change this event's start time to 10.30 (vs. 09.20) - this recurring event shows correctly on the MM!
6. Change start time to 10.05 - recurring event shows correctly on the MM.
7. Change start time to 10.00 - recurring event shows correctly on the MM.
8. Change start time to 09.55 - recurring event DOES NOT show on the MM.
9. Remove the recurrence from the event - recurring event shows correctly on the MM.

**Expected Results**:
Single instance events starting before 10.00 show correctly on the MM
Recurring events starting before 10.00 show correctly on the MM
Recurring events starting on/after 10.00 show correctly on the MM

**Actual Results**:
Single instance events starting before 10.00 show correctly on the MM
**Recurring events starting before 10.00 DO NOT show on the MM**
Recurring events starting on/after 10.00 show correctly on the MM

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!
NextCloud: 18.0.06, Calendar App 2.0.3

Export of .ICS file to MM/modules/default/calendar folder

~/MagicMirror/config/config.js:

                {
                        module: "calendar",
                        header: "School Timetable",
                        position: "top_right",
                        config: {
                                calendars: [
                                        {
                                                symbol: "calendar-check-o ",
                                                url: "webcal://127.0.0.1:8080/modules/default/calendar/school-timetable.ics"
                                        }
                                ]
                        }
                },
 
**Additional Notes**:
Non-recurrent event that shows correctly before 10.00:
BEGIN:VEVENT
CREATED:20200806T235605Z
DTEND;TZID=Australia/Sydney:20200807T100000
DTSTAMP:20200806T235856Z
DTSTART;TZID=Australia/Sydney:20200807T090000
LAST-MODIFIED:20200806T235856Z
SEQUENCE:1
SUMMARY:Test5
TRANSP:OPAQUE
UID:8BDDEF67-33C4-4C32-B5D4-F7D31D83B490
URL;VALUE=URI:
LOCATION:TESTING
END:VEVENT

Recurrent event that does not show correctly before 10.00:
BEGIN:VEVENT
CREATED:20200806T235535Z
DTEND;TZID=Australia/Sydney:20200807T100000
DTSTAMP:20200806T235842Z
DTSTART;TZID=Australia/Sydney:20200807T090000
LAST-MODIFIED:20200806T235842Z
RRULE:FREQ=WEEKLY;INTERVAL=2;UNTIL=20200925T135959Z
SEQUENCE:1
SUMMARY:Test4 recurrent
TRANSP:OPAQUE
UID:E7A31472-B600-4EC6-B06E-19A054343989
URL;VALUE=URI:
LOCATION:TESTING
END:VEVENT

I see that CREATED date time shows as 10hrs behind by TZ, so GMT

MichMich, 2020-09-22:
Can I close this issue?

castletonroad, 2020-09-22:
Hi, I'm not clear how the matter has been fixed? My MM is fully up-to-date but the issue persists.

Did I miss the solution?

Thanks :)

rejas, 2020-09-22:
Did you try the develop branch of MM?

castletonroad, 2020-09-22:
> 
> 
> Did you try the develop branch of MM?

I'm not so experienced with git. Happy to try that if you could point me in the right direction of how to..?

Thanks very much.

sdetweil, 2020-09-22:
this issue can be closed

closed 2020-09-22 by MichMich

castletonroad, 2020-09-23:
Apologies - how do I 'try the develop branch of MM'?

Thank you.

MichMich, 2020-09-23:
In your MagicMirror folder try the following commands:
`git checkout develop && git pull`

castletonroad, 2020-09-23:
> 
> 
> In your MagicMirror folder try the following commands:
> `git checkout develop && git pull`

Thank you very much for the very quick reply.

MM now runs to a blank screen with a mouse cursor dead-centre. :(

MichMich, 2020-09-23:
Maybe try `npm install`

castletonroad, 2020-09-23:
> Maybe try `npm install`

Great - MM back up and running! (I'll remember that...)

However, my recurrent events are STILL messed up. For example, it is currently 21:22 (I am AEST, i.e. GMT+9h) but my weekly recurring 10.15-10.30am appointment from this morning is showing as 'Ends in 3 hrs'!

Let me see how things look first thing in the morning...

sdetweil, 2020-09-23:
@castletonroad 

can u make a sample ics with a calendar block
and a vevent block that shows the problem

I will work to resolve before mm release next week. u can paste the text here, or email to me. same userid on gmail

castletonroad, 2020-09-23:
> 
> 
> @castletonroad
> 
> can u make a sample ics with a calendar block
> and a vevent block that shows the problem
> 
> I will work to resolve before mm release next week. u can paste the text here, or email to me. same userid on gmail

BEGIN:VCALENDAR
PRODID:-//Microsoft Corporation//Outlook 16.0 MIMEDIR//EN
VERSION:2.0
METHOD:PUBLISH
X-CALSTART:20200719T222500Z
X-CALSTART:20200720T000000
X-CALEND:20280825T000000
X-WR-RELCALID:{0000002E-4069-AC36-7A41-8D022446F3D7}
X-WR-CALNAME:School Timetable
BEGIN:VTIMEZONE
TZID:Australia/Sydney
BEGIN:STANDARD
DTSTART:16010401T030000
RRULE:FREQ=YEARLY;BYDAY=1SU;BYMONTH=4
TZOFFSETFROM:+1100
TZOFFSETTO:+1000
END:STANDARD
BEGIN:DAYLIGHT
DTSTART:16011007T020000
RRULE:FREQ=YEARLY;BYDAY=1SU;BYMONTH=10
TZOFFSETFROM:+1000
TZOFFSETTO:+1100
END:DAYLIGHT
END:VTIMEZONE
BEGIN:VEVENT
CLASS:PUBLIC
CREATED:20200830T150748Z
DTEND;TZID=Australia/Sydney:20200722T103000
DTSTAMP:20200625T233123Z
DTSTART;TZID=Australia/Sydney:20200722T101500
LAST-MODIFIED:20200920T021150Z
LOCATION:School
PRIORITY:5
RRULE:FREQ=WEEKLY;COUNT=10;BYDAY=WE
SEQUENCE:0
SUMMARY;LANGUAGE=en-us:P3 - Lesson
TRANSP:OPAQUE
UID:ec400f81-02d5-46b0-ba6a-22f4d007c07d
X-MICROSOFT-CDO-BUSYSTATUS:BUSY
X-MICROSOFT-CDO-IMPORTANCE:1
END:VEVENT
END:VCALENDAR

sdetweil, 2020-09-23:
ok, I can't recreate that. because its sept 23, wed, the wed morning event has passed, so next in 6 days..  I set my time to AEST (which is +10, not +11)

can u check three things

1. edit ~/MaguicMirror/modules/default/calendar/calendarfetcher.js
 line 8 should be 
```
const ical = require("node-ical");
```
2. ls ~/MagicMirror/node_modules/node-ical

3. git branch
should show 
```
* develop
  master
```

castletonroad, 2020-09-23:
1. Yes:
`const ical = require("node-ical");`

2. This:
pi@raspberrypi2:~ $ ls ~/MagicMirror/node_modules/node-ical
ical.js  LICENSE  node-ical.d.ts  node-ical.js  node_modules  package.json  README.md  SECURITY.md  windowsZones.json

3. This:
```
pi@raspberrypi2:~ $ git branch
fatal: not a git repository (or any of the parent directories): .git
```

sdetweil, 2020-09-23:
git branch from inside the MM folder.. , but u have the new code..  hmm... 

weird...  want to do some quick debugging

edit calendarfetcher.js

add a line , new line at line 97
``` 
			Object.entries(data).forEach(([key, event]) => {
				const now = new Date();
				const today = moment().startOf("day").toDate();
				const future = moment().startOf("day").add(maximumNumberOfDays, "days").subtract(1, "seconds").toDate(); // Subtract 1 second so that events that start on the middle of the night will not repeat.
				let past = today;

				console.log("event="+JSON.stringify(event))   // < --- add this line
```
also at approx line 219
```
						const dates = rule.between(pastLocal, futureLocal, true, limitFunction);

						console.log("rule dates="+ JSON.stringify(dates))  // < --- add this line
```

then start mm 

npm start >somefile

after the calendar displays 

copy/paste the  somefile here...  or attach it..

mine looks like 
```
[2020-09-23 15:06:23.911] [LOG]    event=["20200719T222500Z","20200720T000000"]
[2020-09-23 15:06:23.911] [LOG]    event={"type":"VTIMEZONE","params":[],"tzid":"Australia/Sydney","9ebb6868-92f1-43a4-b975-426446220014":{"type":"STANDARD","params":[],"start":"1601-04-01T08:50:36.000Z","datetype":"date-time","rrule":"RRULE:FREQ=YEARLY;BYDAY=1SU;BYMONTH=4","tzoffsetfrom":"+1100","tzoffsetto":"+1000"},"d55b9c18-ef77-4d10-89d8-0f63dfec94c2":{"type":"DAYLIGHT","params":[],"start":"1601-10-07T07:50:36.000Z","datetype":"date-time","rrule":"RRULE:FREQ=YEARLY;BYDAY=1SU;BYMONTH=10","tzoffsetfrom":"+1000","tzoffsetto":"+1100"}}

[2020-09-23 15:06:23.911] [LOG]    event={"type":"VEVENT","params":[],"class":"PUBLIC","created":"2020-08-30T15:07:48.000Z","end":"2020-07-22T00:30:00.000Z","dtstamp":"2020-06-25T23:31:23.000Z","start":"2020-07-22T00:15:00.000Z","datetype":"date-time","lastmodified":"2020-09-20T02:11:50.000Z","location":"School","priority":"5","rrule":{"_cache":{"all":false,"before":[],"after":[],"between":[]},"origOptions":{"dtstart":"2020-07-22T00:15:00.000Z","freq":2,"count":15,"byweekday":[{"weekday":3}]},"options":{"freq":2,"dtstart":"2020-07-22T00:15:00.000Z","interval":1,"wkst":0,"count":15,"until":null,"bysetpos":null,"bymonth":null,"bymonthday":[],"bynmonthday":[],"byyearday":null,"byweekno":null,"byweekday":[3],"bynweekday":null,"byhour":[0],"byminute":[15],"bysecond":[0],"byeaster":null}},"sequence":"0","summary":{"params":{"LANGUAGE":"en-us"},"val":"P3 - Lesson"},"transparency":"OPAQUE","uid":"ec400f81-02d5-46b0-ba6a-22f4d007c07d","MICROSOFT-CDO-BUSYSTATUS":"BUSY","MICROSOFT-CDO-IMPORTANCE":"1"}

[2020-09-23 15:06:23.912] [LOG]     tz offsets past=300 future=300
[2020-09-23 15:06:23.917] [LOG]    rule dates=["2020-09-24T00:15:00.000Z","2020-10-01T00:15:00.000Z","2020-10-08T00:15:00.000Z","2020-10-15T00:15:00.000Z","2020-10-22T00:15:00.000Z","2020-10-29T00:15:00.000Z"]
```
blank lines added by me added by me

castletonroad, 2020-09-23:
I added the lines (you had the correct locations).

```
pi@raspberrypi2:~/MagicMirror $ more somefile

> magicmirror@2.13.0-develop start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-09-24 06:51:26.301] [LOG]    Starting MagicMirror: v2.13.0-develop
[2020-09-24 06:51:26.324] [LOG]    Loading config ...
[2020-09-24 06:51:26.340] [LOG]    Loading module helpers ...
[2020-09-24 06:51:26.346] [LOG]    No helper found for module: alert.
[2020-09-24 06:51:26.349] [LOG]    No helper found for module: clock.
[2020-09-24 06:51:27.889] [LOG]    Initializing new module helper ...
[2020-09-24 06:51:27.894] [LOG]    Module helper loaded: calendar
[2020-09-24 06:51:27.897] [LOG]    No helper found for module: compliments.
[2020-09-24 06:51:27.900] [LOG]    No helper found for module: currentweather.
[2020-09-24 06:51:27.901] [LOG]    No helper found for module: weatherforecast.
[2020-09-24 06:51:29.280] [LOG]    Initializing new module helper ...
[2020-09-24 06:51:29.282] [LOG]    Module helper loaded: MMM-NowPlayingOnSpotify
[2020-09-24 06:51:29.283] [LOG]    All module helpers loaded.
[2020-09-24 06:51:29.624] [LOG]    Starting server on port 8080 ...
[2020-09-24 06:51:29.666] [LOG]    Server started ...
[2020-09-24 06:51:29.669] [LOG]    Connecting socket for: calendar
[2020-09-24 06:51:29.671] [LOG]    Starting node helper for: calendar
[2020-09-24 06:51:29.672] [LOG]    Connecting socket for: MMM-NowPlayingOnSpotify
[2020-09-24 06:51:29.674] [LOG]    Sockets connected & modules started ...
[2020-09-24 06:51:30.039] [LOG]    Launching application.

```

sdetweil, 2020-09-23:
its after the launching application 
```
[2020-09-23 15:06:19.007] [LOG]    Launching application.
 Create new calendar fetcher for url: http://localhost:8090/modules/default/calendar/australia.ics - Interval: 300000
[2020-09-23 15:06:23.808] [INFO]   Checking git for module: MMM-Ring
[2020-09-23 15:06:23.911] [LOG]    event=["20200719T222500Z","20200720T000000"]
[2020-09-23 15:06:23.911] [LOG]    event={"type":"VTIMEZONE","params":[],"tzid":"Australia/Sydney","9ebb6868-92f1-43a4-b975-426446220014":{"type":"STANDARD","params":[],"start":"1601-04-01T08:50:36.000Z","datetype":"date-time","rrule":"RRULE:FREQ=YEARLY;BYDAY=1SU;BYMONTH=4","tzoffsetfrom":"+1100","tzoffsetto":"+1000"},"d55b9c18-ef77-4d10-89d8-0f63dfec94c2":{"type":"DAYLIGHT","params":[],"start":"1601-10-07T07:50:36.000Z","datetype":"date-time","rrule":"RRULE:FREQ=YEARLY;BYDAY=1SU;BYMONTH=10","tzoffsetfrom":"+1000","tzoffsetto":"+1100"}}
[2020-09-23 15:06:23.911] [LOG]    event={"type":"VEVENT","params":[],"class":"PUBLIC","created":"2020-08-30T15:07:48.000Z","end":"2020-07-22T00:30:00.000Z","dtstamp":"2020-06-25T23:31:23.000Z","start":"2020-07-22T00:15:00.000Z","datetype":"date-time","lastmodified":"2020-09-20T02:11:50.000Z","location":"School","priority":"5","rrule":{"_cache":{"all":false,"before":[],"after":[],"between":[]},"origOptions":{"dtstart":"2020-07-22T00:15:00.000Z","freq":2,"count":15,"byweekday":[{"weekday":3}]},"options":{"freq":2,"dtstart":"2020-07-22T00:15:00.000Z","interval":1,"wkst":0,"count":15,"until":null,"bysetpos":null,"bymonth":null,"bymonthday":[],"bynmonthday":[],"byyearday":null,"byweekno":null,"byweekday":[3],"bynweekday":null,"byhour":[0],"byminute":[15],"bysecond":[0],"byeaster":null}},"sequence":"0","summary":{"params":{"LANGUAGE":"en-us"},"val":"P3 - Lesson"},"transparency":"OPAQUE","uid":"ec400f81-02d5-46b0-ba6a-22f4d007c07d","MICROSOFT-CDO-BUSYSTATUS":"BUSY","MICROSOFT-CDO-IMPORTANCE":"1"}
[2020-09-23 15:06:23.912] [LOG]     tz offsets past=300 future=300
[2020-09-23 15:06:23.917] [LOG]    rule dates=["2020-09-24T00:15:00.000Z","2020-10-01T00:15:00.000Z","2020-10-08T00:15:00.000Z","2020-10-15T00:15:00.000Z","2020-10-22T00:15:00.000Z","2020-10-29T00:15:00.000Z"]
[2020-09-23 15:06:23.920] [INFO]   Calendar-Fetcher: Broadcasting 6 events.
```

castletonroad, 2020-09-23:
> its after the launching application
>
> ```
> [2020-09-23 15:06:19.007] [LOG]    Launching application.
>  Create new calendar fetcher for url: http://localhost:8090/modules/default/calendar/australia.ics [...]

That is all that I have in somefile.

I'm conscious that you think this issue is closed and yet I'm taking up your time.

I'll do a clean install of MM and see where that gets me.

Thanks for trying to help me with this.

sdetweil, 2020-09-23:
you have to switch to the develop branch  even with a new install, the fixes are not promoted yet, on the 1st of Oct.. 

so, do you have two MagicMirror installations?  cause I don't see the fetcher message either..
>Create new calendar fetcher for url:

its ok, I just did a ton of work on the timezone adjustments and windows timezone names.. so I am familiar with the code, I am the best to do it.. nothing else going on (retired)... 

I have to go watch grandson play hockey... back later

castletonroad, 2020-09-23:
> you have to switch to the develop branch even with a new install, the fixes are not promoted yet, on the 1st of Oct..
>
> so, do you have two MagicMirror installations? cause I don't see the fetcher [...]

Just the one installation. And it was all from a clean raspberryOS install a few months back - it's an old pi2 that does nothing else (I have a pi3 for fun and a pi4 as a mail server and vpn).

I'll do an clean install and switch to dev, then report back 👍 

And thanks again for your help on this, and support across the whole MM project.

sdetweil, 2020-09-23:
@castletonroad also. more stops when the page is full. I would use cat

castletonroad, 2020-09-23:
> 
> 
> @castletonroad also. more stops when the page is full. I would use cat

There was no more content in the file...

castletonroad, 2020-09-23:
Clean MM install, MM working.

Switched to Dev.

Now on start, I get:
```
[2020-09-24 09:20:42.685] [LOG]    Loading module helpers ...
[2020-09-24 09:20:42.688] [LOG]    No helper found for module: alert.
[2020-09-24 09:20:42.690] [LOG]    Initializing new module helper ...
[2020-09-24 09:20:42.691] [LOG]    Module helper loaded: updatenotification
[2020-09-24 09:20:42.693] [LOG]    No helper found for module: clock.
[2020-09-24 09:20:42.703] [ERROR]  App threw an error during load
[2020-09-24 09:20:42.706] [ERROR]  Error: Cannot find module 'node-ical'
Require stack:
- /home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js
- /home/pi/MagicMirror/modules/default/calendar/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
-
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js:8:14)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
[2020-09-24 09:20:42.711] [ERROR]  Whoops! There was an uncaught exception...
[2020-09-24 09:20:42.713] [ERROR]  Error: Cannot find module 'node-ical'
Require stack:
- /home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js
- /home/pi/MagicMirror/modules/default/calendar/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
-
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js:8:14)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js',
    '/home/pi/MagicMirror/modules/default/calendar/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[2020-09-24 09:20:42.734] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-09-24 09:20:42.736] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-09-24 09:20:42.969] [LOG]    Launching application.

```

sdetweil, 2020-09-23:
after switching to dev have to do 
npm install
again

castletonroad, 2020-09-24:
> 
> 
> after switching to dev have to do
> npm install
> again

Of course, doh!

Up, running, and seems to being well-behaved.  Let me check tomorrow morning that ALL recurrent events show correctly.

Thanks again.

sdetweil, 2020-09-24:
thanks for the update. we have a few days. oct 1 is next release

castletonroad, 2020-09-24:
No joy.

First two events of the day, as per my originally documented issue, not showing. 3rd event of the day, showing as starts in 3 hrs.

Just can't my head around this...

sdetweil, 2020-09-24:
hm.. can u build me a ics with those three  entries..  the two above at the top are past events..

or add the debugging previously..

sdetweil, 2020-09-24:
also, if this is the event that starts at 10:15, it is now 7:38, so 2:45 rounded up is 3 hours..

sdetweil, 2020-09-24:
try this... edit calendarfetcher.js again and change 
```
						let pastLocal = 0;
						let futureLocal = 0;
						if (isFullDayEvent(event)) {
```
to 
```
						let pastLocal = 0;
						let futureLocal = 0;
						if (true) {  //< ---- changing this line
```

castletonroad, 2020-09-24:
> 
> 
> try this... edit calendarfetcher.js again and change
> 
> ```
> 						let pastLocal = 0;
> 						let futureLocal = 0;
> 						if (isFullDayEvent(event)) {
> ```
> 
> to
> 
> ```
> 						let pastLocal = 0;
> 						let futureLocal = 0;
> 						if (true) {  //< ---- changing this line
> ```

Done.

Events 1, 2 from YESTERDAY show first for TODAY (ending in 15hrs, 43 mins respectively ?!).

Events 3-7 for today show correctly.

Events 1-2 for today show as following 3-7, but for TOMORROW.

castletonroad, 2020-09-24:
`pi@raspberrypi2:~/MagicMirror $ pm2 start mm`

```
pi@raspberrypi2:~/MagicMirror $ npm start >somefile
[2020-09-25 08:41:33.757] [ERROR]  Whoops! There was an uncaught exception...
[2020-09-25 08:41:33.768] [ERROR]  Error: listen EADDRINUSE: address already in use 127.0.0.1:8080
    at Server.setupListenHandle [as _listen2] (net.js:1226:14)
    at listenInCluster (net.js:1274:12)
    at GetAddrInfoReqWrap.doListen [as callback] (net.js:1413:7)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:65:10) {
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '127.0.0.1',
  port: 8080
}
[2020-09-25 08:41:33.795] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-09-25 08:41:33.797] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues

^C
pi@raspberrypi2:~/MagicMirror $
```

castletonroad, 2020-09-24:
I then get no output in 'somefile':

```
pi@raspberrypi2:~/MagicMirror $ more somefile

> magicmirror@2.13.0-develop start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-09-25 08:41:31.592] [LOG]    Starting MagicMirror: v2.13.0-develop
[2020-09-25 08:41:31.615] [LOG]    Loading config ...
[2020-09-25 08:41:31.631] [LOG]    Loading module helpers ...
[2020-09-25 08:41:31.638] [LOG]    No helper found for module: alert.
[2020-09-25 08:41:31.641] [LOG]    No helper found for module: clock.
[2020-09-25 08:41:33.098] [LOG]    Initializing new module helper ...
[2020-09-25 08:41:33.100] [LOG]    Module helper loaded: calendar
[2020-09-25 08:41:33.102] [LOG]    No helper found for module: compliments.
[2020-09-25 08:41:33.104] [LOG]    No helper found for module: currentweather.
[2020-09-25 08:41:33.105] [LOG]    No helper found for module: weatherforecast.
[2020-09-25 08:41:33.107] [LOG]    No helper found for module: MMM-NowPlayingOnSpotify.
[2020-09-25 08:41:33.108] [LOG]    All module helpers loaded.
[2020-09-25 08:41:33.461] [LOG]    Starting server on port 8080 ...
[2020-09-25 08:41:33.503] [LOG]    Server started ...
[2020-09-25 08:41:33.506] [LOG]    Connecting socket for: calendar
[2020-09-25 08:41:33.508] [LOG]    Starting node helper for: calendar
[2020-09-25 08:41:33.509] [LOG]    Sockets connected & modules started ...
[2020-09-25 08:41:33.816] [LOG]    Launching application.

```

sdetweil, 2020-09-24:
can't have both, pm2 start AND npm start

pm2 stop mm

npm start

sdetweil, 2020-09-24:
can I get those cal entries.. in an ics ..

castletonroad, 2020-09-24:
> pm2 stop mm

```
pi@raspberrypi2:~/MagicMirror $ npm start

> magicmirror@2.13.0-develop start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-09-25 08:47:59.072] [LOG]    Starting MagicMirror: v2.13.0-develop
[2020-09-25 08:47:59.099] [LOG]    Loading config ...
[2020-09-25 08:47:59.117] [LOG]    Loading module helpers ...
[2020-09-25 08:47:59.124] [LOG]    No helper found for module: alert.
[2020-09-25 08:47:59.127] [LOG]    No helper found for module: clock.
[2020-09-25 08:48:00.590] [LOG]    Initializing new module helper ...
[2020-09-25 08:48:00.592] [LOG]    Module helper loaded: calendar
[2020-09-25 08:48:00.595] [LOG]    No helper found for module: compliments.
[2020-09-25 08:48:00.597] [LOG]    No helper found for module: currentweather.
[2020-09-25 08:48:00.600] [LOG]    No helper found for module: weatherforecast.
[2020-09-25 08:48:00.602] [LOG]    No helper found for module: MMM-NowPlayingOnSpotify.
[2020-09-25 08:48:00.603] [LOG]    All module helpers loaded.
[2020-09-25 08:48:00.955] [LOG]    Starting server on port 8080 ...
[2020-09-25 08:48:00.998] [LOG]    Server started ...
[2020-09-25 08:48:01.001] [LOG]    Connecting socket for: calendar
[2020-09-25 08:48:01.006] [LOG]    Starting node helper for: calendar
[2020-09-25 08:48:01.008] [LOG]    Sockets connected & modules started ...
[2020-09-25 08:48:01.275] [LOG]    Launching application.
[2020-09-25 08:48:06.317] [LOG]    Create new calendar fetcher for url: http://127.0.0.1:8080/modules/default/calendar/personal.ics - Interval: 300000
[2020-09-25 08:48:06.404] [LOG]    Create new calendar fetcher for url: http://127.0.0.1:8080/modules/default/calendar/school-assessments-assignments.ics - Interval: 300000
[2020-09-25 08:48:06.421] [LOG]    Create new calendar fetcher for url: http://127.0.0.1:8080/modules/default/calendar/school-timetable.ics - Interval: 300000
[2020-09-25 08:48:07.174] [INFO]   Calendar-Fetcher: Broadcasting 0 events.
[2020-09-25 08:48:09.425] [INFO]   Calendar-Fetcher: Broadcasting 10 events.
[2020-09-25 08:48:11.557] [INFO]   Calendar-Fetcher: Broadcasting 10 events.
```

castletonroad, 2020-09-24:
> 
> 
> can I get those cal entries.. in an ics ..

[school-timetable.zip](https://github.com/MichMich/MagicMirror/files/5279645/school-timetable.zip)

sdetweil, 2020-09-25:
ok, Term 3 - end, looks correct , all day event, ends at midnight your time (without the (if true), back to original)
"start":"2020-09-24T14:00:00.000Z","end":"2020-09-25T14:00:00.000Z"

i see a japanese at 082500, - 10 = 22:25
rule dates=["2020-09-25T22:25:00.000Z"]
the rule got it wrong..   

and biology at  092000 - 10 = 23:20,  "dtstart":"2020-07-30T23:20:00.000Z"
 rule dates=["2020-09-25T23:20:00.000Z"]
the rule got it wrong, friday at 09:20 - 10 is thursday utc

Ok, this is the event back from the calendar parser, it correctly adjusted the start date from 7/31 08:25 to 7/30: 22:25

[2020-09-25 09:52:03.662] [LOG]    event={"type":"VEVENT","params":[],"class":"PUBLIC","created":"2020-08-30T15:10:18.000Z","end":"2020-07-30T23:20:00.000Z","dtstamp":"2020-06-26T00:04:24.000Z","start":"2020-07-30T22:25:00.000Z","datetype":"date-time","lastmodified":"2020-09-20T02:11:48.000Z","location":"SCHOOL","priority":"5","rrule":{"_cache":{"all":false,"before":[],"after":[],"between":[]},"origOptions":{"dtstart":"2020-07-30T22:25:00.000Z","freq":2,"count":5,"interval":2,"byweekday":[{"weekday":4}],"wkst":{"weekday":6}},"options":{"freq":2,"dtstart":"2020-07-30T22:25:00.000Z","interval":2,"wkst":6,"count":5,"until":null,"bysetpos":null,"bymonth":null,"bymonthday":[],"bynmonthday":[],"byyearday":null,"byweekno":null,"byweekday":[4],"bynweekday":null,"byhour":[22],"byminute":[25],"bysecond":[0],"byeaster":null}},"sequence":"1","summary":{"params":{"LANGUAGE":"en-us"},"val":"P1 - IB Japanese B HL (N.2.02)"},"transparency":"OPAQUE","uid":"1cb9eab0-0f50-42f6-a416-060bbab58c0e","MICROSOFT-CDO-BUSYSTATUS":"BUSY","MICROSOFT-CDO-IMPORTANCE":"1"}
[2020-09-25 09:52:03.662] [LOG]     tz offsets past=-600 future=-600
[2020-09-25 09:52:03.663] [LOG]    rule dates=["2020-09-25T22:25:00.000Z"] start=Fri Sep 25 2020 19:52:03 GMT+1000 (Australian Eastern Standard Time) end=Sat Sep 25 2021 09:59:59 GMT+1000 (Australian Eastern Standard Time)

but the rrule processor did NOT we said find the rule dates between NOW(Fri Sep 25 2020 19:52:03 GMT+1000) and a year from now (Sat Sep 25 2021 09:59:59)
but it came back with Friday?, right time, wrong day 
"2020-09-25T22:25:00.000Z"

will investigate more

castletonroad, 2020-09-25:
Thanks @sdetweil for continuing to look through this.

I'm relieved there might be 'something' there that's causing an issue and that I'm not losing my mind!

As I've mentioned previously, this was working fine up until a few months back.

sdetweil, 2020-09-25:
man, I don't know how it worked.. the calendar handler was not using tz at all.. 

we'll get it... (I think!!)

sdetweil, 2020-09-25:
well, i have reported the problem and debugged it to the line of code.. 
https://github.com/jakubroztocil/rrule/issues/420

don't know what to do about it yet

sdetweil, 2020-10-01:
@castletonroad can u look at this output, 2.13 with your school-timetable-ics file , and your timezone
https://www.dropbox.com/s/gls8jab28xngv3o/2.13.school-timetable.png?dl=0
(showEnd:true and maximumEntries:25)

castletonroad, 2020-10-01:
@sdetweil the behaviour looks good. P1 should always start at 8.25am and finish 9.20am...

sdetweil, 2020-10-02:
well, thats not how the rule processor works.. P1 -IB  Japanese, shows 08:25-09:20
the parser gets UTC time, 22:25:00 (for start) (timezone offset +10 lost)
oops, then daylight  switch this weekend (Oct 3) and next appt is Oct 10, after daylight, now offset is +11
utc+11 = 9:25 start - 10:20 end

once we get todays date past oct 3, then it shift will be lost, and offset will be used to utc as 21:25

good news is u won't be late on the 10th, cause the cal will correct on the 4rd. (tomorrow your time already)

castletonroad, 2020-10-02:
@sdetweil so the issue is now resolved?

What do I need to do in order to implement the fix?

Thanks very much.

sdetweil, 2020-10-02:
install 2.13 and replace the modules/default/calendar/calendarfetcher.js with the linked one below (second link) ( save the original)
see this pic (time corrected too)
https://www.dropbox.com/s/f8qd6hslamofjsi/Screenshot%20at%202020-10-03%2006-38-13.png?dl=0

here is the new file 
https://www.dropbox.com/s/t7m1f49kvv3x2b1/calendarfetcher.js?dl=0

castletonroad, 2020-10-03:
2.13 not 2.14dev?

sdetweil, 2020-10-03:
yes, 2.13,  we are testing.

sdetweil, 2020-10-04:
I uploaded a new version of calendarfetcher.js just now, same dropbox link
now that you have dst switched,  the time would be wrong ... corrected