MagicMirrorOrg/MagicMirror #2113: Module config not merging in deep

bugsounet, 2020-08-12:
Hi, i have some problem when i create some module.
`this.defaults` and `this.config` is not merged properly
`Object.assign()` don't do all job in deep
So I create `configMerge()`

this is some test : 

**So I do some another test and not worked without `configMerge()`**

Module name : MMM-Freebox (in dev for new version)
in MMM-Freebox.js:

```js
  defaults: {
    updateDelay:  5 * 1000,
    token: "",
    activeOnly: false,
    showIcon: true,
    showButton: true,
    showBandWidth: true,
    showRate: true,
    showClient: true,
    showClientRate: true,
    showClientCnxType: true,
    showFreePlayer: true,
    showMissedCall: true,
    showVPNUsers: true,
    maxMissed: 3,
    showIP: true,
    showPing: true,
    pingAdress: "google.fr",
    textWidth: 250,
    excludeMac: [],
    sortBy: null,
    debug: false,
    verbose: false,
    dev: false,
    debitText: "Débit total utilisé : ",
    player : {
      showPlayerInfo: false,
      // depuis le firmware 4.2.3, problemes d'affichage des logos
      // essayez avec les ips :  "192.168.0.254" (l'ip du freebox server)
      //                         "mafreebox.free.fr" ou le resultat de l'ip de mafreebox.free.fr
      //                         "212.27.38.253" l'ip de mafreebox.free.fr (a voir si cela fonctionne pour vous)
      ServerIP: "212.27.38.253",
      UseEPGDayURL: true,
      EPGDelay: 2* 60 *60 *1000
    }
  },
```

I try to merge config with MMM-Freebox.js :
```js
  start: function () {
    this.config = Object.assign({}, this.defaults, this.config)
```

I send config to node_helper.js with:
```js
  notificationReceived: function (notification, payload) {
    switch(notification) {
      case "DOM_OBJECTS_CREATED":
        this.sendSocketNotification("INIT", this.config)
        break
    }
  },
```

module configuration in config.js:
```js
{
  module: "MMM-Freebox",
  disabled: false,
  header: "INFO Freebox",
  position: "top_center",
  config: {
    token: "xxxx",
    updateDelay:  5 * 1000,
    activeOnly: false,
    sortBy: "type",
    showButton: true,
    debug: true,
    //dev: true,
    //verbose: true,
    showVPNUsers: true,
    textWidth: 250,
    player: {
      showPlayerInfo: true
    }
  }
},
```

result in node_helper.js:
```sh
[2020-08-10 16:53:52.705] [LOG]    {
  updateDelay: 5000,
  token: 'xxxx',
  activeOnly: false,
  showIcon: true,
  showButton: true,
  showBandWidth: true,
  showRate: true,
  showClient: true,
  showClientRate: true,
  showClientCnxType: true,
  showFreePlayer: true,
  showMissedCall: true,
  showVPNUsers: true,
  maxMissed: 3,
  showIP: true,
  showPing: true,
  pingAdress: 'google.fr',
  textWidth: 250,
  excludeMac: [],
  sortBy: 'type',
  debug: true,
  verbose: false,
  dev: false,
  debitText: 'Débit total utilisé : ',
  player: { showPlayerInfo: true }
}
```
so config is not merged !

now ... Let's try with configMerge() with SAME configuration
just modify : 
```js
  start: function () {
    this.config = configMerge({}, this.defaults, this.config)
```
result in node_helper.js:

```js
[2020-08-10 17:08:14.984] [LOG]    {
  updateDelay: 5000,
  token: 'xxxx',
  activeOnly: false,
  showIcon: true,
  showButton: true,
  showBandWidth: true,
  showRate: true,
  showClient: true,
  showClientRate: true,
  showClientCnxType: true,
  showFreePlayer: true,
  showMissedCall: true,
  showVPNUsers: true,
  maxMissed: 3,
  showIP: true,
  showPing: true,
  pingAdress: 'google.fr',
  textWidth: 250,
  excludeMac: [],
  sortBy: 'type',
  debug: true,
  verbose: false,
  dev: false,
  debitText: 'Débit total utilisé : ',
  player: {
    showPlayerInfo: true,
    ServerIP: '212.27.38.253',
    UseEPGDayURL: true,
    EPGDelay: 7200000
  }
}
```

You can't see that `player: {}` value is now set and merged with default value

@MichMich : 

in MM core `js/module.js`
```
	/* setConfig(config)
	 * Set the module config and combine it with the module defaults.
	 *
	 * argument config object - Module config.
	 */
	setConfig: function (config) {
		this.config = Object.assign({}, this.defaults, config);
	},
```
Can I send you a patch with `configMerge()` script for solve this developer (and users) issue (It take head !) ?
I use it in all my module for merging and why not take place in MM core ;)

sdetweil, 2020-08-15:
could also use loadash.merge

bugsounet, 2020-08-15:
I will be see that when I return back vacations ;)

sdetweil, 2020-08-29:
lodash won't work, require() not allowed in module.js as it runs in browser...

where is the code for configMerge()

bugsounet, 2020-08-31:
see in my repo (or look my PR)

dvbit, 2020-09-19:
@MichMich Could You please consider integrating this?
Thank You

MichMich, 2020-09-19:
If I recall correctly there is already a change in de develop branche which addresses this issue.

sdetweil, 2020-09-19:
from develop is/module.js
```
setConfig: function (config) {
		this.config = Object.assign({}, this.defaults, config);
	},
```

so, not yet

bugsounet, 2020-09-19:
Wow... Mister @MichMich is ALIVE !!!! WOW !!!!

bugsounet, 2020-09-19:
@MichMich your reponse to [this post](https://github.com/MichMich/MagicMirror/issues/2129#issuecomment-695328585) is the better response who @MichMich can do :+1: 

So don't touch to your repo !!!!!!

I will close ALL of mine :)

Good lock to MagicMirror :)

@MichMich lost 22 Modules ... and 14 NPM library
bye !

bugsounet, 2020-09-19:
too late ...

bugsounet, 2020-09-19:
I'm a Little developer!!! .... I can't comunicate and YOU aRE thE BIg Boss !!!!

MichMich, 2020-09-19:
> from develop is/module.js
> 
> ```
> 
> setConfig: function (config) {
> 
> 		this.config = Object.assign({}, this.defaults, config);
> 
> 	},
> 
> ```
> 
> 
> 
> so, not yet

It might be worth investing options like lodash merge then, indeed.

bugsounet, 2020-09-19:
don't take HEAD with this ... i says ... i close ALL... STOP !

MichMich, 2020-09-19:
@bugsounet not sure what you are trying to achieve but stop spamming the comment. Thanks.

sdetweil, 2020-09-19:
loadsh merge can't be used in browser without some webpack packaging. so his routine would probably be easier

bugsounet, 2020-09-19:
Mister @MichMich after 2 months your are alive ... so don't touch of this part of code ...

MichMich, 2020-09-19:
> loadsh merge can't be used in browser without some webpack packaging. so his routine would probably be easier

We can look for an other implementation or just copy the lodash implementation into a separate file which we include. The implementation isn't that extensive.

MichMich, 2020-09-19:
> Mister @MichMich after 2 months your are alive ... so don't touch of this part of code ...

@bugsounet keep in mind this is a side project for me. I have limited time to spent on this project and have managed to keep it maintained for the past 6 years.

I understand it can be frustrating when I won't respond to all issues. But do my utmost best to review all PR's and merge were possible. 

Keep in mind that, although I love any community contribution in the form of modules, I'm not obligated to give support. That's simply not possible. I have my day job. I have my family including two small kids, I have a house that needs a lot of work. And like everybody, I have my own struggles in this weird COVID time. 

The way you handle your frustration isn't helping anybody. It isn't helping your users. It isn't helping the community, it isn't helping the project, it isn't helping me and most of all it isn't helping you. 

Please calm down. Take some time off and return when you feel you can make a positive contribution to a community solely run by volunteers.

Thanks.

bugsounet, 2020-09-19:
[sure](https://github.com/MichMich/MagicMirror/issues/2129#issuecomment-695340974)

[look !](https://github.com/bugsounet/) Wow ...

closed 2020-09-19 by bugsounet

gelaw2, 2020-09-20:
This is sad. Updates that break the code!  Good that I keep my own backup copies.
Now I must look for alternatives to MMM-GoogleAssistant and MMM-Assistant2Display.

rejas, 2020-09-20:
@gelaw2 If you still have the latest code we could create a new module from that working with the original mm software

gelaw2, 2020-09-20:
@rejas I have backup of my systems, not of the original code.  
However, if I had the original code, I would not publish someone else's work without their consent.  
Hopefully tensions will ease and we can return to helping each other.

bugsounet, 2020-09-20:
Hey big boss of @rejas : 
if you think it's that easy to clone one of my modules ... there are NPM libraries with all that goes well behind ... So ... try ... but not works ...
I says ... STOP so .. it can't launch in MM

MichMich, 2020-09-20:
@bugsounet, correct me if I'm wrong but your messages still appear to be a bit hostile to me. 

Tell me what you want me to do. Maybe we can find a solution to your technical problem. 

Also, if you know a solution yourself, feel free to send a PR to speed up things.