MagicMirrorOrg/MagicMirror #2115: node_helper to main module payload not sended in deep

bugsounet, 2020-08-14:
Hi, I have just see this problem

I create an complex payload and it not send any array to the main module

sample:
```sh
Send this Status: {
  OS: 'Debian 10 (buster)',
  NETWORK: [
    wlx98ded00d0521: { ip: '192.168.0.193', type: 'wireless', default: true }
  ],
  MEMORY: [ total: '8Go', used: '2.56Go', percent: '33' ],
  STORAGE: [
    '/': { size: '40Go', used: '10.33Go', use: 25.84 },
    '/home': { size: '383Go', used: '53.66Go', use: 13.99 },
    '/boot/efi': { size: '96Mo', used: '31.57Mo', use: 32.89 },
    '/media/Data': { size: '541Go', used: '96.68Go', use: 17.88 }
  ],
  CPU: [ type: 'Core™ i5-3350P', temp: '44.0', usage: '7' ],
  UPTIME: '2 jours 11 heures 46 minutes',
  RECORD: '27 jours 18 heures 49 minutes',
  SCREEN_STATUS: 'OFF'
}
```
this is what is send to main module with `sendSocketNotification` command:
```
{OS: "Debian 10 (buster)", NETWORK: Array(0), MEMORY: Array(0), STORAGE: Array(0), CPU: Array(0), …}
CPU: []
MEMORY: []
NETWORK: []
OS: "Debian 10 (buster)"
RECORD: "27 jours 18 heures 49 minutes"
SCREEN_STATUS: "OFF"
STORAGE: []
UPTIME: "2 jours 12 heures 24 minutes"
__proto__: Object
```

of course, i have verified in node_helper that it will be set correcty before
```js
    console.log("Send this Status:", this.status)
    this.sendSocketNotification('STATUS', this.status)
```

I want to send one notification with all value (await for all value with async)
actually i can't send one notification so, il will do one notification for one value... I don't like that (RPI optimization)

sdetweil, 2020-08-14:
that is an invalid object 

if I change to this 
```
{
					  OS: 'Debian 10 (buster)',
					  NETWORK: [ {
					    wlx98ded00d0521: { ip: '192.168.0.193', type: 'wireless', default: true }
					  }
					  ],
					  MEMORY: [ {total: '8Go', used: '2.56Go', percent: '33'} ],
					  STORAGE: [
					    {'/': { size: '40Go', used: '10.33Go', use: 25.84 }},
					    {'/home': { size: '383Go', used: '53.66Go', use: 13.99 }},
					    {'/boot/efi': { size: '96Mo', used: '31.57Mo', use: 32.89 }},
					    {'/media/Data': { size: '541Go', used: '96.68Go', use: 17.88 }}
					  ],
					  CPU: [ {type: 'Core™ i5-3350P', temp: '44.0', usage: '7' }],
					  UPTIME: '2 jours 11 heures 46 minutes',
					  RECORD: '27 jours 18 heures 49 minutes',
					  SCREEN_STATUS: 'OFF'
					}
```
then send and receive produces this 
```
{"OS":"Debian 10 (buster)","NETWORK":[{"wlx98ded00d0521":{"ip":"192.168.0.193","type":"wireless","default":true}}],"MEMORY":[{"total":"8Go","used":"2.56Go","percent":"33"}],"STORAGE":[{"/":{"size":"40Go","used":"10.33Go","use":25.84}},{"/home":{"size":"383Go","used":"53.66Go","use":13.99}},{"/boot/efi":{"size":"96Mo","used":"31.57Mo","use":32.89}},{"/media/Data":{"size":"541Go","used":"96.68Go","use":17.88}}],"CPU":[{"type":"Core™ i5-3350P","temp":"44.0","usage":"7"}],"UPTIME":"2 jours 11 heures 46 minutes","RECORD":"27 jours 18 heures 49 minutes","SCREEN_STATUS":"OFF"}
```

send code 
```
this.item={
					  OS: 'Debian 10 (buster)',
					  NETWORK: [ {
					    wlx98ded00d0521: { ip: '192.168.0.193', type: 'wireless', default: true }
					  }
					  ],
					  MEMORY: [ {total: '8Go', used: '2.56Go', percent: '33'} ],
					  STORAGE: [
					    {'/': { size: '40Go', used: '10.33Go', use: 25.84 }},
					    {'/home': { size: '383Go', used: '53.66Go', use: 13.99 }},
					    {'/boot/efi': { size: '96Mo', used: '31.57Mo', use: 32.89 }},
					    {'/media/Data': { size: '541Go', used: '96.68Go', use: 17.88 }}
					  ],
					  CPU: [ {type: 'Core™ i5-3350P', temp: '44.0', usage: '7' }],
					  UPTIME: '2 jours 11 heures 46 minutes',
					  RECORD: '27 jours 18 heures 49 minutes',
					  SCREEN_STATUS: 'OFF'
					}
			this.sendSocketNotification('data', this.item)
```
CPU, MEMORY, NETWORK and STORAGE had invalid structure defs.. need surrounding {}

bugsounet, 2020-08-15:
mouaahah, sometime i'm so stupid :( 
thx sam

closed 2020-08-15 by bugsounet