MagicMirrorOrg/MagicMirror #2118: Tomorrow's calendar events are being displayed as 'Today'.

DivingIvan, 2020-08-26:
Hi,

I have found an issue with the MagicMirror calendar module, whereby it is displaying single full day events with tomorrow's date as being 'Today'.

So, for example, if the event is a single full day event (eg a birthday) with date '20200827', and the current date is 26th August 2020, it will show up in on the MagicMirror calendar as being 'Today' whereas it should show as 'Tomorrow'.

Events for later dates do show up correctly; for example 'In 2 days' for an event with a date '20200828'.

I am fairly certain that this is due to the way that events are added to the event list; and the way that the 'event.today' boolean property is calculated; ie., the lines of code:-

`var today = moment().startOf("day");`

and

`event.today = event.startDate >= today && event.startDate < today + 24 * 60 * 60 * 1000;`

I suspect that the 'moment().startOf("day")' is not taking account of the fact that UK time is currently set to British Summer Time, as opposed to GMT, so is returning an absolute time value that is offset by +1 hour. As a result, the code:-

`event.startDate < today + 24 * 60 * 60 * 1000`

returns a true value instead of false. Indeed, I have observed that if I manually set the system time to be 'GMT', the calendar event will show up correctly on the calendar as 'Tomorrow'.

I am running on a Raspberry Pi 3 B with the latest version of Pi OS (Rasbian Buster). Kernel version 5.4.51.

Node version is 6.14.6.

MagicMirror version is 2.12.0

magic21nrw, 2020-08-29:
Facing the same issue here. Events for tomorrow are displayed as today.

MM Version 2.12.
Node Version v10.21.0

Tried to replace ical ^0.8.0 with 0.5.0 without success. Seems that the other bug reports are referring to other issues and that yours is the same I am facing here.

rejas, 2020-08-30:
This issue I think is already fixed in the develop branch. Could you guys test if it goes away when using that branch?

DivingIvan, 2020-08-30:
Hi rejas.

Many thanks for the information. I have just checked out the develop branch, and can confirm that this does appear to have at least partially fixed the issue. I've not done a comprehensive test, but the one recurring event that I have in my calendar for tomorrow does indeed show as being 'Tomorrow'.

**However, I have also added a test event for today (I'm using Google calendar). This is an all day event that repeats annually. Unfortunately, the event does not appear at all on the MagicMirror calendar (it should appear, and show as being 'Today'). If I set the event to non-repeating (ie a one off, all day event) it does appear on the calendar, and shows as being 'Today'.**

Could someone else could please confirm this behavior.

Thanks again.

sdetweil, 2020-08-30:
@DivingIvan an 'all day' event, created on the day, some hours after the day begins, should not be shown, as the 'all day' part has already expired.

ongoing events typically don't show, as the cal code typically uses start time as the filter, not end time.

DivingIvan, 2020-08-30:
Hi @sdetweil,

Thanks for clarifying. I'll check again tomorrow, as today's 'Tomorrow' event should be tomorrow's 'Today' event. :-)

kazhokin, 2020-08-30:
Hello guys!
I'm pretty new her, so apologize me if I ask dumb questions!
My problem is, that I don't know how to use the develop branch. I tried to copy the files into my MagicMirror folder, but now I get a blackscreen after rebooting my raspberry. 

And my calendar has one problem more. It show's me events regardless of the month. So if someone has a birthday in October the 5. it's gonna show me in 6 days (starting from today). 

I hope you can help me!

Thanks

sdetweil, 2020-08-30:
to recover, git reset -- hard
that will put everything back
then
git checkout develop
now start mm like always

DivingIvan, 2020-08-31:
Hi @sdetweil,

Unfortunately, there does still appear to be an issue with the develop branch.

I have a repeating (every year on same date) event today which has always been in my Google calendar. Yesterday it was correctly showing as 'Tomorrow'. However, today it does not appear on the MagicMirror calendar at all. However, if I manually change the system timezone to be GMT-0, the event does appear, and correctly shows as 'Today'.

Setting timezone back to Europe/London, the event again disappears from the MagicMirror calendar.

Today's public holiday (which is a non-repeating event) correctly shows in the MagicMirror calendar as being 'Today' with either Europe/London timezone or GMT-0 timezone setting.

sdetweil, 2020-08-31:
@DivingIvan can u post the ics parts

the calendar def at the top and the vevent

for this recurring problem. thx

DivingIvan, 2020-08-31:
Hi @sdetweil,

As requested, the relevant sections of the calendar ics.

BEGIN:VCALENDAR
PRODID:-//Google Inc//Google Calendar 70.9054//EN
VERSION:2.0
CALSCALE:GREGORIAN
METHOD:PUBLISH
X-WR-CALNAME:Calendar
X-WR-TIMEZONE:Europe/London
BEGIN:VTIMEZONE
TZID:Europe/London
X-LIC-LOCATION:Europe/London
BEGIN:DAYLIGHT
TZOFFSETFROM:+0000
TZOFFSETTO:+0100
TZNAME:BST
DTSTART:19700329T010000
RRULE:FREQ=YEARLY;BYMONTH=3;BYDAY=-1SU
END:DAYLIGHT
BEGIN:STANDARD
TZOFFSETFROM:+0100
TZOFFSETTO:+0000
TZNAME:GMT
DTSTART:19701025T020000
RRULE:FREQ=YEARLY;BYMONTH=10;BYDAY=-1SU
END:STANDARD
END:VTIMEZONE

BEGIN:VEVENT
DTSTART;VALUE=DATE:20120831
DTEND;VALUE=DATE:20120901
RRULE:FREQ=YEARLY
DTSTAMP:20200831T133055Z
UID:xxxxxxxxxxxxxxxxxxxxxxxxxx@google.com
CREATED:20111208T151805Z
DESCRIPTION:
LAST-MODIFIED:20120105T093251Z
LOCATION:
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:Anniversary (1985)
TRANSP:TRANSPARENT
CATEGORIES:http://schemas.google.com/g/2005#event
END:VEVENT

sdetweil, 2020-08-31:
thanks.. after looking at the rfcs for ical and rrule, and the code (in develop) the problem is the code finding the range of dates in the repeating rule.   it is using TIME OF DAY even when the specified date/time ONLY includes time. 

SO,   08/31/2020.00:00:00 has passed, so the NEXT date is 08/31/2021.00:00:00, and today is NOT 08/31/2021, so it can't be displayed 

looking for solutions... the vevent doesn't have a date/time stamp..so it gets it from the rule, but the rule also doesn't have a time stamp, so it defaults to UTC.   so we still have a problem there as well... and the rule doesn't expose the actual text to see if there was time applied..    so, an all day recurring event  should NOT use time, but a recurring that DOES specify a specific time  should use it. 

i did a quick fix testing for all day and that solved it.  now gotta make sure if a start time was specified , but its an all dat event to see what needs to get fixed

sdetweil, 2020-08-31:
none of my calendar tools will let me set a time if this is a full day event

DivingIvan, 2020-09-01:
Hi @sdetweil,

Many thanks for following up on this. I've made the suggested changes to calendarfetcher.js, and it does now show today's events correctly as 'Today' and tomorrow's events as 'Tomorrow'.

As regards being able to set a time for a full day event, I have always assumed that, by definition, an 'all day event' is simply one that starts at midnight and ends at midnight (or just before), so therefore it does not require a start/end time as such. I think that if you wanted to be able to set a start time, you would have to specify the event as NOT being an all day event, and then specify the start/end dates/times.

sdetweil, 2020-09-01:
@DivingIvan thanks for the feedback I will get the change submitted today.

I agree that all day events should start and end at midnight. just wondering if I need to protect against the possibility of some sloppy provider not clearing the time field

DivingIvan, 2020-09-01:
Hi @sdetweil,

Unless I'm missing something, I don't think that it's something that you need to worry about. From what I can see of the iCalendar specification, there is no actual concept of an all day event. The fact that an event is 'all-day' seems simply to be inferred from the fact that it starts and ends at midnight. If an event has a 'start time' and/or 'end-time' other than midnight, then by definition this would (IMHO) therefore have to infer that it is NOT an all day event.

sdetweil, 2020-09-02:
@DivingIvan the fix has been added to the develop branch

do this to get a copy
```
cd ~/MagicMirror
mv modules/default/calendar/calendarfetcher.js .
git pull
```
then try mm again

if all good u can erase the calendarfetcher.js in the MagicMirror folder
or copy it back over the branch version

DivingIvan, 2020-09-03:
Hi @sdetweil,

That's great. All seems to be working as it should now. Thanks again.

closed 2020-09-18 by MichMich

dithiii, 2020-10-31:
I am still seeing this problem. Clean install of mm. Showing recurring events from a google calendar as 1 day early. I have switched to develop branch and the problem persists. 

The problem is being referenced in multiple 3rd party modules such as MMM-calendarweek and others. Develop branch has *not* fixed it. I have tried to replace calendarfetcher.js as above but that also did not help. Suggestions?

sdetweil, 2020-10-31:
there are some additional fixes coming..